open-power/hostboot #8: palmetto: Hostboot stuck in error loop at istep 21.1

jk-ozlabs, 2015-01-29:
On booting current op-build on palmetto (pass 2), I'm seeing hostboot get stuck at istep 21.1:

```
 70.54823|ISTEP 18.14
 70.58662|ISTEP 21. 1
 81.03411|================================================
 81.05850|Error reported by occc (0x2A00)
 81.05850|  <none>
 81.05850|  ModuleId   0x02 unknown
 81.05850|  ReasonCode 0x2a00 unknown
 81.05851|  UserData1  unknown : 0x0000000000000000
 81.05851|  UserData2  unknown : 0x000000000a000000
 81.05851|User Data Section 0, type UD
 81.05851|  Subsection type 0x0c
 81.05852|  ComponentId hb-trace (0x3100)
 81.05852|User Data Section 1, type UD
 81.05852|  Subsection type 0x00
 81.05852|  ComponentId occc (0x2a00)
 81.05852|User Data Section 2, type UD
 81.05853|  Subsection type 0x06
 81.05853|  ComponentId errl (0x0100)
 81.05853|  CALLOUT
 81.05853|  PROCEDURE ERROR
 81.05853|  Procedure: 85
 81.05854|User Data Section 3, type UD
 81.05854|  Subsection type 0x03
 81.05854|  ComponentId errl (0x0100)
 81.05854|User Data Section 4, type UD
 81.05854|  Subsection type 0x01
 81.05855|  ComponentId errl (0x0100)
 81.05855|  STRING
 81.05855|  Hostboot Build ID: 
 81.05855|================================================
```
- this error repeats indefinitely. 

This is a clean build, at op-build commit `5da6cb1`.

williamspatrick, 2015-01-29:
Resolved by 1fe250207a9cda3203839ec0df06030b8a627b50.  Until op-build is updated to include this, you can add it as a patch in the hostboot directory.

closed 2015-01-29 by williamspatrick

jk-ozlabs, 2015-01-29:
Super, thanks!