open-power/hostboot #10: hostboot fails to build in op-build

sharkcz, 2015-02-27:
I have a problem when building hostboot from op-build on Fedora 20

I get these errors, is it a problem in hostboot or in the build environment?
...
In file included from ../../../../obj/modules/testhwas/testhwas.C:18:0:
./hwasGardTest.H:48:28: error: ‘TargetInfo’ does not name a type
 bool compareAffinity(const TargetInfo t1, const TargetInfo t2)
                            ^
./hwasGardTest.H:48:39: error: ISO C++ forbids declaration of ‘t1’ with no type [-fpermissive]
 bool compareAffinity(const TargetInfo t1, const TargetInfo t2)
                                       ^
./hwasGardTest.H:48:49: error: ‘TargetInfo’ does not name a type
 bool compareAffinity(const TargetInfo t1, const TargetInfo t2)
                                                 ^
./hwasGardTest.H:48:60: error: ISO C++ forbids declaration of ‘t2’ with no type [-fpermissive]
 bool compareAffinity(const TargetInfo t1, const TargetInfo t2)
                                                            ^
./hwasGardTest.H: In function ‘bool compareAffinity(int, int)’:
./hwasGardTest.H:50:19: error: request for member ‘affinityPath’ in ‘t1’, which is of non-class type ‘const int’
         return t1.affinityPath < t2.affinityPath;
                   ^
./hwasGardTest.H:50:37: error: request for member ‘affinityPath’ in ‘t2’, which is of non-class type ‘const int’
         return t1.affinityPath < t2.affinityPath;
                                     ^
./hwasGardTest.H: In member function ‘void HwasGardTest::testdeconfigPresentByAssoc1()’:
./hwasGardTest.H:3388:9: error: ‘TargetInfoVector’ was not declared in this scope
         TargetInfoVector l_targets;
         ^
./hwasGardTest.H:3388:26: error: expected ‘;’ before ‘l_targets’
         TargetInfoVector l_targets;
                          ^
./hwasGardTest.H:3389:26: error: expected ‘;’ before ‘l_targToDeconfig’
         TargetInfoVector l_targToDeconfig;
                          ^
./hwasGardTest.H:3390:9: error: ‘TargetInfo’ was not declared in this scope
         TargetInfo l_TargetInfo;
         ^
./hwasGardTest.H:3390:20: error: expected ‘;’ before ‘l_TargetInfo’
         TargetInfo l_TargetInfo;
                    ^
./hwasGardTest.H:3392:9: error: ‘l_TargetInfo’ was not declared in this scope
         l_TargetInfo.pThisTarget = NULL;
         ^
./hwasGardTest.H:3400:9: error: ‘l_targets’ was not declared in this scope
         l_targets.push_back(l_TargetInfo);
         ^
./hwasGardTest.H:3402:35: error: ‘l_targToDeconfig’ was not declared in this scope
         presentByAssoc(l_targets, l_targToDeconfig);
                                   ^
./hwasGardTest.H:3402:51: error: ‘presentByAssoc’ was not declared in this scope
         presentByAssoc(l_targets, l_targToDeconfig);
                                                   ^
./hwasGardTest.H: In member function ‘void HwasGardTest::testdeconfigPresentByAssoc2()’:
./hwasGardTest.H:3470:9: error: ‘TargetInfoVector’ was not declared in this scope
         TargetInfoVector l_targets;
         ^
./hwasGardTest.H:3470:26: error: expected ‘;’ before ‘l_targets’
         TargetInfoVector l_targets;
                          ^
./hwasGardTest.H:3471:9: error: ‘TargetInfo’ was not declared in this scope
         TargetInfo l_TargInfo;
         ^
./hwasGardTest.H:3471:20: error: expected ‘;’ before ‘l_TargInfo’
         TargetInfo l_TargInfo;
                    ^
./hwasGardTest.H:3473:9: error: ‘l_TargInfo’ was not declared in this scope
         l_TargInfo.pThisTarget = NULL;
         ^
./hwasGardTest.H:3475:9: error: ‘l_targets’ was not declared in this scope
         l_targets.insert(l_targets.begin(), NUM_TARGS, l_TargInfo);
[... 195 lines cut ...]
         ^
./hwasGardTest.H:4033:9: error: ‘l_targets’ was not declared in this scope
         l_targets.insert(l_targets.begin(), NUM_TARGS, l_TargInfo);
         ^
./hwasGardTest.H:4035:26: error: expected ‘;’ before ‘l_targToDeconfig’
         TargetInfoVector l_targToDeconfig;
                          ^
./hwasGardTest.H:4087:35: error: ‘l_targToDeconfig’ was not declared in this scope
         presentByAssoc(l_targets, l_targToDeconfig);
                                   ^
./hwasGardTest.H:4087:51: error: ‘presentByAssoc’ was not declared in this scope
         presentByAssoc(l_targets, l_targToDeconfig);
                                                   ^
    MAKE       test CODE
./hwasGardTest.H: In function ‘bool compareAffinity(int, int)’:
./hwasGardTest.H:51:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
    TESTGEN    testintr.C
cc1plus: all warnings being treated as errors
make[10]: *** [../../../../obj/modules/testhwas/testhwas.o] Error 1
make[9]: *** [_BUILD/PASSES/CODE/BODY] Error 2
make[8]: *** [_BUILD/SUBDIR/CODE/test] Error 2
make[7]: *** [_BUILD/PASSES/CODE/POST] Error 2
make[6]: *** [_BUILD/SUBDIR/CODE/hwas] Error 2

sharkcz, 2015-02-27:
command used to build:
op-build palmetto_defconfig && op-build

geissonator, 2015-02-27:
Was Fedora 20 working before with op-build compiles or is this the first time trying?

sharkcz, 2015-02-27:
I'm getting the same error for couple weeks, but F-20 is mentioned in op-build README as "supported" build environment. The only thing I haven't made yet is downgrading make from 3.82 to 3.81, but it sounds unlikely as the cause of the failure.

geissonator, 2015-02-27:
@scprek Can you take a look?  Nothing jumps out at me other then it's not getting the targeting include for some reason.

causten, 2015-02-27:
It appears the steps to have all the packages isn't whole for Fedora.  I needed to add these...

perl(XML::LibXML)
perl(Digest::SHA1)

yum install libxml2-devel 

needed dlevel i86_64 not the 686 stuff that is listed (that may just because of my handy dandy W530)

yum install ""

sharkcz, 2015-02-28:
the workaround is to edit the generated file obj/modules/testhwas/testhwas.C with the snipped bellow after it fails and restart "op-build"

--- testhwas.C.orig	2015-02-28 21:59:08.476191769 +0100
+++ testhwas.C	2015-02-28 21:57:07.557141946 +0100
@@ -14,6 +14,7 @@
 #include <trace/interface.H>
 #include <limits.h>
 #include <cxxtest/TestSuite.H>
+#include <hwas/common/hwas.H>
 
 #include "hwasGardTest.H"

williamspatrick, 2015-02-28:
What is the order of the includes in the generated 'testhwas.C'?

Mine has:
#include "hwas1test.H"
#include "hwasGardTest.H"
#include "hwasSysAvailSvcTest.H"

hwas1test.H includes hwas/common/hwas.H but hwasGardTest.H does not.  From the diff you gave, it seems like maybe Fedora is generating the includes in a different order, which is exposing a missing include in hwasGardTest.H.  This would appear to be a valid bug in that file because we should be able set TESTS=hwasGardTest.H in that makefile and have it successfully compile.

I will commit this to our repository:
> diff --git a/src/usr/hwas/test/hwasGardTest.H b/src/usr/hwas/test/hwasGardTest.H
> index 8237c7f..5576a28 100644
> --- a/src/usr/hwas/test/hwasGardTest.H
> +++ b/src/usr/hwas/test/hwasGardTest.H
> @@ -39,6 +39,7 @@
>  #include \<cxxtest/TestSuite.H>
>  #include \<errl/errlmanager.H>
>  #include \<targeting/common/targetservice.H>
> +#include \<hwas/common/hwas.H>
>  #include \<hwas/common/deconfigGard.H>
>  #include \<hwas/common/hwas_reasoncodes.H>

sharkcz, 2015-02-28:
I see

#include "hwasGardTest.H"
#include "hwasSysAvailSvcTest.H"
#include "hwas1test.H"

williamspatrick, 2015-02-28:
Interesting.  Must be something different about how Fedora 20 (or you mentioned updated 'gmake') does wildcards that changes the order of the files.  We'll get it fixed.  Thanks for reporting.

sharkcz, 2015-02-28:
Thanks for fixing :-)

closed 2015-02-28 by williamspatrick

causten, 2015-02-28:
There is actually more then just this problem.  As you continue to build using the basic instructions you will hit xxd and xml problems.  I've got the rest of the issues in an edit.  Just need to get a pull request going