open-power/hostboot #11: Document attributeOverride tool.

williamspatrick, 2015-03-02:
Attribute override text file format is not documented in a publically available document.  Please create a readme under Hostboot and/or open-power/docs.

aalugore, 2015-03-03:
I have added a README file to the top level of open-power/hostboot. Someone please review to ensure it has the information we want.

aalugore, 2015-03-03:
Second draft is up

aalugore, 2015-03-03:
Third draft

williamspatrick, 2015-03-03:
We should leave issue open until code shows up.  Please put 'Resolves open-power/hostboot#11' in your commit message and github will automatically close this issue once the commit has been merged and mirrored to github.

aalugore, 2015-03-03:
What code? isnt this issue just for the README?

williamspatrick, 2015-03-03:
You said you had a commit you were working on.  This is what I meant.  I guess if it is just documentation there is no code but there is still a commit.

williamspatrick, 2015-03-04:
I see the confusion.  You modified hostboot directly in github, which is not allowed.  You need to make a commit in gerrit for review.

closed 2015-04-10 by aalugore