open-power/hostboot #13: Add "IPMI shutdown requested" to non-debug console

shenki, 2015-03-03:
My Palmetto was being told by the BMC over IPMI to shut down, but until I put a debug build of hostboot on there I had no idea that was the case.

We should display something so the non-debug console knows what's going on.

```
  5.38149|IPMI|rp: queuing sync 18:6
  6.67555|IPMI|rp: timeout: 18:36
  6.67555|IPMI|rp: get_capabilities not ok 195, using defaults
  6.67563|IPMI|dd: I>write ok 18:35 seq 2 len 0
  6.69803|IPMI|dd: I>read b2h ok 1c:35 seq 2 len 10 cc 0
  6.69804|IPMI|rp: queuing event 3a:4 for handler
  6.69807|IPMI|rp: Graceful shutdown request recieved
  6.69809|INITSVC|doShutdown(i_status=0000000001230000)
  6.69809|INITSVC|doShutdown> status=0000000001230000
  6.69810|IPMI|dd: I>write ok 18:6 seq 3 len 2
  6.69836|IPMI|rp: I>MSG_STATE_GRACEFUL_SHUTDOWN: send power off command to BMC
  6.79797|IPMI|dd: I>read b2h ok 1c:6 seq 3 len 0 cc 0
  6.79804|
```

williamspatrick, 2015-03-03:
We should add a console output in the case that CONSOLE_OUTPUT_TRACE is not set to indicate the reason for the shutdown.  We added that recently for SBE update reboots.

Also, it doesn't seem like the BMC actually gave us enough time to gracefully shutdown.  Is this a problem with the way we are responding to the IPMI request, is it a problem with our "graceful shutdown" path, or is it a problem with the BMC code?

rjknight, 2015-03-09:
Added code to print shutdown message to non-debug console, currently being reviewed.  I also opened an RTC issue to investigate the shutdown path.

closed 2015-03-11 by rjknight