open-power/hostboot #17: Hostboot build fails with ENABLE_CHECKSTOP_ANALYSIS set in Palmetto config

sbroylesibm, 2015-03-07:
When I set ENABLE_CHECKSTOP_ANALYSIS in the hostboot palmetto config I get the following build failure:
exception caught: Could not find symbol _ZNK4PRDF17PnorFirDataReader7getScomEPN9TARGETING6TargetEmRm
make[4]: *** [../img/hbirt.bin] Error 255
make[4]: *** Waiting for unfinished jobs....
exception caught: Could not find symbol _ZNK4PRDF17PnorFirDataReader7getScomEPN9TARGETING6TargetEmRm
make[4]: *** [../img/hbirt_test.bin] Error 255
make[3]: *** [_BUILD/PASSES/IMAGE/BODY] Error 2
make[2]: *** [_BUILD/SUBDIR/IMAGE/src] Error 2
make[1]: *** [_BUILD/PASSES/IMAGE/POST] Error 2

My op build is pointing to HB commit 3593853a4323d9ec7de1ad6502e22b3e66af03ae

williamspatrick, 2015-03-07:
I'm pretty sure they currently also need HBRT_PRD with ENABLE_CHECKSTOP_ANALYSIS.  It isn't currently marked in the HBconfig file.

williamspatrick, 2015-03-07:
@dcrowell77 - Can you mention this issue to Zane and have him look into it?  He doesn't have a github account right now that I know of.

zane131, 2015-03-09:
@sbroylesibm You are missing 1b488dc and daf669b, which are required for the checkstop analysis to work. However, this should not prevent it from compiling. Continuing investigation.

closed 2015-03-16 by williamspatrick