open-power/hostboot #20: Uncorrectable ECC error after user initiated shutdown

bofferdn, 2015-03-31:
Reference https://github.com/open-power/tyan-openpower/issues/49 for history

I was able to reproduce this pretty easily

** Booted, waited until I saw "PNOR_PN/SN" traces in the console, then ipmi powered off
target 00040001
  6.43127|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 0003000A
  6.46956|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00040001
  6.47407|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 0003000B
  for target 00040001
  6.55746|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 0003000D
   6.59989|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 0003000E
IsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 0003000F
  6.65167|VPD|>IpVpdFacade::findRecordOffsetPnor: No matching Record (VINI) found in TOC!
  6.65168|ERRL|>>addHwCallout(0x00040002 0x6 0x1 0x0)
  6.65168|ERRL|>>addProcedureCallout(0x4, 0x5)
  6.65169|ERRL|>>addProcedureCallout(0x55, 0x1)
  6.66640|VPD|E>IpVpdFacade::findRecordOffsetPnor: No matching Record (VINI) found in TOC!
  6.66640|ERRL|>>addHwCallout(0x00040002 0x6 0x1 0x0)
  6.66641|ERRL|>>addProcedureCallout(0x4, 0x5)
  6.66641|ERRL|>>addProcedureCallout(0x55, 0x1)
  6.66643|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN != SEEPROM_PN/SN, Loading PNOR from SEEPROM for target 00040002
  6.67729|PNOR|>>PnorDD::writeFlash(i_address=0x172000)>
  6.79018|PNOR|<<PnorDD::
  6.79864|VPD|E>IpVpdFacade::findKeywordAddr: No matching PT keyword found within VTOC record!
  6.79865|ERRL|>>addHwCallout(0x00040002 0x6 0x1 0x0)
  6.79865|ERRL|>>addProcedureCallout(0x4, 0x5)
  6.79866|ERRL|>>addProcedureCallout(0x55, 0x1)
0NR|>>PnorDD::writeFlashss01  7.1176ess=0x173200)> io_buflen=00001200
7.14732|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00030010
 00 for target 00030011
  7.23656|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00040002
  7.24078|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00030012
  7.27902|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00040002
  7.28192|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN  7.3209nsureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00030014
  7.36327|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00040002
  7.36622|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00030015
::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00030016
  7.44855|VPDreCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00040002
  7.45157|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN = SEEPROM_PN/SN for target 00030017
  7.46169|VPD|E>IpVpdFacade::findRecordOffsetPnor: No matching Record (VINI) found in TOC!
  7.46170|ERRL|>>addHwCallout(0x00040003 0x6 0x1 0x0)
  7.46171|ERRL|>>addProcedureCallout(0x4, 0x5)
  7.46171|ERRL|>>addProcedureCallout(0x55, 0x1)
  7.47164|VPD|E>IpVpdFacade::findRecordOffsetPnor: No matching Record (VINI) found in TOC!
  7.47165|ERRL|>>addHwCallout(0x00040003 0x6 0x1 0x0)
  7.47165|ERRL|>>addProcedureCallout(0x4PROM for target 00040003
  7.48210|PNOR|>>PnorDD::writeFlash(i_address=0x174400)>
  7.59941|PNOR|<<PnorDD::writeFlash(i_a> io_buflen=00001200
  7.61239|VPD|E>IpVpdFacade::findKeywordAddrut(0x4, 0x5)
  7.61241|ERRL|>>addProcedureCallout(0x55, 0x1)
n::d44

Then
[bofferdn@gfw161 ~]$ $IPMI chassis power off
[bofferdn@gfw161 ~]$   $IPMI raw 0x04 0x30 0x50 0x01 0x00 0x02 0 0 0 0 0   // (make sure boot count 2)
[bofferdn@gfw161 ~]$ $IPMI chassis power on

After booting, it came back, fails with signature we're familiar with:
 7.29852|PNOR|PnorRP::readFromDevice> Uncorrectable ECC error : chip=0,offset=0x173200
The failing address is somewhat near the addresses being played with during the initial power interruption.  I think that help bolster our theory that pnor pages are being wiped without ECC, and power off request is intervening before ECC can be re-written.  
It's more concerning because this was a user initiated action

Patrick noted "The close time proximity and the fact that there are no traces between the 'doshutdown' call and the IPMI-RP trace indicate to me that there is a bug in this shutdown path where the IPMI-RP is sending the shutdown message to the BMC in the first callback instead of the second.  

There is one shutdown callback to the IPMI-RP that is to tell it "stop accepting new messages and flush your current ones".  Then we do the memory flush.  Then there is a second callback to tell the IPMI-RP "you are ok to ack the BMC now for that previous shutdown request."

bofferdn, 2015-04-01:
One question is why do we 
  6.66643|VPD|VPD::ensureCacheIsInSync: PNOR_PN/SN != SEEPROM_PN/SN, Loading PNOR from SEEPROM for target 00040002

That is what causes the reload to PNOR; I don't think this was the first boot, so why did anything change?

dcrowell77, 2015-04-01:
We should not be seeing a mismatch unless the hardware really changed or the pnor cache was cleared.  The pnor cache is cleared on a reflash or if we boot from the golden side.

dcrowell77, 2015-05-13:
I think that the issues mentioned in here have been addressed with If792fae2964011c1ca80e8cc08235477dc8a7a3b and I1d882e8596a0218fa499d947f3c6387e7ecd4ee8

closed 2015-05-13 by dcrowell77