open-power/hostboot #21: Two socket systems fail with a deconfigured slave processor.

williamspatrick, 2015-04-02:
8.50476|ISTEPS_TRACE|PRESENT> Proc[05]=C000000000000000
...
  9.19614|HWAS_I|I>hwasPlatDeconfigGard.C: Get returning 1 GARD Records
  9.21550|HWAS_I|I>deconfigGard.C: Deconfiguring Target 00050001, errlEid 0x90000015
...
  9.23584|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc BUS Peer: 000F0000
  9.23585|HWAS_I|I>deconfigGard.C: Deconfiguring Target 000F0000, errlEid 0x90000015
...
 19.51245|ISTEPS_TRACE|Running proc_a_x_pci_dmi_pll_initf HWP on target HUID 00050000
 19.53177|FAPI|proc_a_x_pci_dmi_pll_initf.C: 
   Parameter1, start_XBUS=true
   Parameter2, start_ABUS=false
   Parameter3, start_PCIE=true
   Parameter4, start_DMI=true 
...
 19.93265|FAPI_I|proc_start_clocks_chiplets.C: proc_start_clocks_chiplets: Entering ...
 19.95199|XSCOM|E>XSCOM status error HMER: 00c0050000000000 ,XSComStatus = 5, Addr=8030006

Looks like if the Abus unit is deconfigured (by association) we do not call PLL init for the Abus logic on the master processor.  The very next istep tries to touch the logic still.

We do not see this problem on our multi-node system because we keep all the Abus units "configured" until much later in the IPL.

williamspatrick, 2015-08-21:
@dcrowell77 , @sannerd - Any update?

dcrowell77, 2015-08-21:
@rjknight actually has a fix in flight

williamspatrick, 2015-09-12:
Resolved now?

dcrowell77, 2015-09-12:
I think that 102ee037ba001da14eaa754766130558c1e8a0e0 has the fix.

closed 2015-09-14 by williamspatrick