open-power/hostboot #22: Error data from IPMI is not parsed correctly.

williamspatrick, 2015-04-10:
If we have an error in IPMI, the console displays as 'unknown' for all the fields that should be derived from the error log tags in the code (/*@ ... */ field).  Specifically I saw this problem on this error in ipmibt.C:

                /* @errorlog tag
                 * @errortype       ERRL_SEV_INFORMATIONAL
                 * @moduleid        IPMI::MOD_IPMISRV_REPLY
                 * @reasoncode      IPMI::RC_READ_EVENT_FAILURE
                 * @userdata1       command of message
                 * @userdata2       completion code
                 * @devdesc         an async completion code was not CC_OK
                 * @custdesc        Unexpected IPMI completion code from the BMC
                 */

I suspect the reason for this is that the error comments have the stray "tag" in them after "errorlog" and it is throwing off one of the parsers we have.

We should either remove the stray tags or make sure that our parser code and handle any ascii garbage after the 'errorlog' tag.

brihh, 2015-04-10:
part of the problem (and a source of future issues..) is that our internal
error parsers are different from this code that displays to the console. we
would probably be better off if that code were common..

.bri.

williamspatrick, 2015-08-21:
Dan fixed via 69262b80495d705baf5e9759beab6f113481405b.

closed 2015-08-21 by williamspatrick