open-power/hostboot #25: Machine doesn't boot after software error

apopple, 2015-05-26:
After getting the below xstop the machine stops booting. Admittedly the xstop is due to a bug in system software, but it shouldn't permanately stop the machine booting:

CPU Summary:

== Recoverables ==

== Checkstops ==
p8      k0:n0:s0:p00         EX9 | 0x19040000 (00) | XFIR_IN0
p8      k0:n0:s0:p00         EX9 | 0x19010C00 (01) | TLBIE_SW_ERR

== FIR Lookup ==
                                    TLBIE_SW_ERR |   TLBIE received illegal AP/LP field from core.; 
                                        XFIR_IN0 |   summary bit(any xstop); 

I get the following errors from Hostboot on reboot:

 3.64150|ISTEP  6. 3
  3.99286|ISTEP  6. 4
  3.99388|ISTEP  6. 5
  4.44992|HWAS|PRESENT> DIMM[03]=00AA000000000000
  4.44993|HWAS|PRESENT> Membuf[04]=4000000000000000
  4.44993|HWAS|PRESENT> Proc[05]=8000000000000000
  7.61691|ISTEP  6. 6
  9.18410|================================================
  9.18410|Error reported by unknown (0xE500)
  9.18410|  <none>
  9.18410|  ModuleId   0x0b unknown
  9.18411|  ReasonCode 0xe540 unknown
  9.18411|  UserData1  unknown : 0x0006000600000101
  9.18411|  UserData2  unknown : 0x7e0d000100000000
  9.18412|User Data Section 0, type UD
  9.18412|  Subsection type 0x04
  9.18412|  ComponentId errl (0x0100)
  9.18412|User Data Section 1, type UD
  9.18412|  Subsection type 0x06
  9.18413|  ComponentId errl (0x0100)
  9.18413|  CALLOUT
  9.18413|  HW CALLOUT
  9.18413|  Reporting CPU ID: 75
  9.18413|  Called out entity:
  9.18414|User Data Section 2, type UD
  9.18414|  Subsection type 0x06
  9.18414|  ComponentId errl (0x0100)
  9.18414|  CALLOUT
  9.18414|  PROCEDURE ERROR
  9.18415|  Procedure: 16
  9.18415|User Data Section 3, type UD
  9.18415|  Subsection type 0x33
  9.18415|  ComponentId unknown (0xe500)
  9.18415|User Data Section 4, type UD
  9.18416|  Subsection type 0x01
  9.18416|  ComponentId unknown (0xe500)
  9.18416|  STRING
  9.18416|  
  9.18416|User Data Section 5, type UD
  9.18417|  Subsection type 0x0c
  9.18417|  ComponentId hb-trace (0x3100)
  9.18417|User Data Section 6, type UD
  9.18417|  Subsection type 0x03
  9.18418|  ComponentId errl (0x0100)
  9.18418|User Data Section 7, type UD
  9.18418|  Subsection type 0x01
  9.18418|  ComponentId errl (0x0100)
  9.18418|  STRING
  9.18419|  Hostboot Build ID: hostboot-70b5e31-opdirty-71b88d7/hbicore.bin/cyril
  9.18419|================================================
  9.54200|================================================
  9.54201|Error reported by initservice (0x0500)
  9.55198|  Deconfigured occurred during an istep. The reconfig loop was not performed by Hostboot because either the Istep is outside the reconfig loop (desired steps 0), too many reconfig loops were attempted, in manufacturing mode or in istep mode.
  9.55198|  ModuleId   0x03 ISTEP_INITSVC_MOD_ID
  9.55198|  ReasonCode 0x050a ISTEP_FAILED_DUE_TO_DECONFIG
  9.55199|  UserData1  Istep that failed : 0x0000000600000006
  9.55199|  UserData2  SubStep that failed : 0x0000000000000000
  9.55199|User Data Section 0, type UD
  9.55199|  Subsection type 0x0c
  9.55200|  ComponentId hb-trace (0x3100)
  9.55200|User Data Section 1, type UD
  9.55200|  Subsection type 0x0c
  9.55200|  ComponentId hb-trace (0x3100)
  9.55201|User Data Section 2, type UD
  9.55201|  Subsection type 0x06
  9.55201|  ComponentId errl (0x0100)
  9.55201|  CALLOUT
  9.55201|  PROCEDURE ERROR
  9.55202|  Procedure: 1
  9.55202|User Data Section 3, type UD
  9.55202|  Subsection type 0x01
  9.55202|  ComponentId errl (0x0100)
  9.55202|  STRING
  9.55203|  libistepdisp.so
  9.55203|User Data Section 4, type UD
  9.55203|  Subsection type 0x03
  9.55203|  ComponentId errl (0x0100)
  9.55203|User Data Section 5, type UD
  9.55204|  Subsection type 0x01
  9.55204|  ComponentId errl (0x0100)
  9.55204|  STRING
  9.55205|  Hostboot Build ID: hostboot-70b5e31-opdirty-71b88d7/hbicore.bin/cyril
  9.55205|================================================

And on subsequent reboots:
  1.91500|ERRL|Dumping errors reported prior to registration
  2.28838|IPMI: power cycle requested
  2.69033|IPMI: shutdown complete
  1.91555|ERRL|Dumping errors reported prior to registration
  3.75812|ISTEP  6. 3
  4.10948|ISTEP  6. 4
  4.11048|ISTEP  6. 5
  4.56734|HWAS|PRESENT> DIMM[03]=00AA000000000000
  4.56734|HWAS|PRESENT> Membuf[04]=4000000000000000
  4.56734|HWAS|PRESENT> Proc[05]=8000000000000000
  7.93459|ISTEP  6. 6
  8.17292|================================================
  8.17292|Error reported by hwas (0x0C00)
  8.18265|  HWAS host_gard: no masterCore found
  8.18265|  ModuleId   0x81 MOD_HOST_GARD
  8.18265|  ReasonCode 0x0c84 RC_MASTER_CORE_NULL
  8.18265|  UserData1  0 : 0x0000000000000000
  8.18266|  UserData2  0 : 0x0000000000000000
  8.18266|User Data Section 0, type UD
  8.18266|  Subsection type 0x06
  8.18266|  ComponentId errl (0x0100)
  8.18267|  CALLOUT
  8.18267|  PROCEDURE ERROR
  8.18267|  Procedure: 85
  8.18267|User Data Section 1, type UD
  8.18268|  Subsection type 0x01
  8.18268|  ComponentId errl (0x0100)
  8.18268|  STRING
  8.18268|  host_gard
  8.18269|User Data Section 2, type UD
  8.18269|  Subsection type 0x03
  8.18269|  ComponentId errl (0x0100)
  8.18269|User Data Section 3, type UD
  8.18270|  Subsection type 0x01
  8.18270|  ComponentId errl (0x0100)
  8.18270|  STRING
  8.18271|  Hostboot Build ID: hostboot-70b5e31-opdirty-71b88d7/hbicore.bin/cyril
  8.18271|================================================

williamspatrick, 2015-05-26:
@zane131 - is the callout correct on this?

ozbenh, 2015-05-26:
> After getting the below xstop the machine stops booting. Admittedly
> the xstop is due to a bug in system software, but it shouldn't
> permanately stop the machine booting:

This is yet another case of gard'ing as a result of a SW/HV bug. This
needs to stop :-)

Ben.

williamspatrick, 2015-05-27:
https://github.com/open-power/hostboot/blob/master/src/usr/diag/prdf/common/plat/pegasus/Ex.rule#L1818

Callout of the core, with gard, is by design in this case.  If you want to change that behavior we need to get it discussed with the core design and RAS teams.  They are the ones that dictate these requirements.

apopple, 2015-05-28:
Ok. We've taken this issue up with the RAS teams as preventing a system boot due to a software bug doesn't really seem like the right thing to do, especially as there is no obvious user friendly way to make the machine bootable again (at least that I'm aware of).

williamspatrick, 2015-08-21:
I believe this has been resolved.

closed 2015-08-21 by williamspatrick