open-power/hostboot #26: op-build failure at building hostboot

weiwangudel, 2015-06-29:
Dear Hostboot Team,

I am trying to build op-build, but failed at building hostboot. I tried on two machines, one is op-build v1.3, the other is v1.2. 

They all have errors similar to the following: 
make[8]: *** [../../../obj/modules/errl/errludcallout.o] Segmentation fault (core dumped)
make[8]: *** [../../../obj/modules/errl/errludcallout.o] Segmentation fault (core dumped)
make[8]: *** Deleting file `../../../obj/modules/errl/errludcallout.o'
make[8]: *** Waiting for unfinished jobs....
    LD         libdevicefw_rt.so
    MAKE       test CODE
    TESTGEN    testdevicefw_rt.C
    DEP        testdevicefw_rt.C
    CXX        testdevicefw_rt.C
    LD         libtestdevicefw_rt.so
make[7]: *** [_BUILD/PASSES/CODE/BODY] Error 2
make[6]: *** [_BUILD/SUBDIR/CODE/errl] Error 2
make[5]: *** [_BUILD/PASSES/CODE/POST] Error 2
make[4]: *** [_BUILD/SUBDIR/CODE/usr] Error 2
make[3]: *** [_BUILD/PASSES/CODE/POST] Error 2
make[2]: *** [_BUILD/SUBDIR/CODE/src] Error 2
make[1]: *** [_BUILD/PASSES/CODE/POST] Error 2
make[1]: Leaving directory `/home/weiwang/git/op-build/output/build/hostboot-70b5e31d74487d51e69a0e0a390adea6b4f32dc5'

What is interesting is, the machine that failed initially with v1.3, later was successfully built with palmetto.pnor!  I did not do anything special other than retype "op-build". 

However, on the other machine op-build v1.2, I still fail at this step. 

By the way, could you also be so kind to explain how palmetto.pnor is supposed to be used? Does it need to be combined with occ package as occ readme mention  
"
op-build occ-rebuild openpower-pnor-rebuild
"? 

Thank you!

Wei Wang

williamspatrick, 2015-06-29:
Are you sure this was built on a 64 bit machine?  I've seen this exact failure on a 32 bit system.

weiwangudel, 2015-06-29:
Hi Patrick,

Thanks for the quick reply! 
You are right, the one that still fails is not a 64 bit linux. 
The one that successfully built .pnor is 64-bit.  

Is the next step "pflash" with the palmetto.pnor, or should I do something more? 

Thank you!

closed 2015-06-29 by weiwangudel

williamspatrick, 2015-06-29:
Yes, you should be able to pflash the resulting palmetto.pnor on.  Especially if you've done that before with v1.2.

There have been compatibility problems between different levels of the BMC code.  If you get a failure during the boot sequence related to IPMI you might need a BMC update for your machine.

Another option would be to edit this file to disable IPMI but that also requires disabling the OCC:
https://github.com/open-power/op-build/blob/master/openpower/configs/hostboot/palmetto.config