open-power/hostboot #30: Hostboot is setting pa-features incorrectly

sannerd, 2015-08-18:
> > 
> > The other thing to check is whether you have MMU_FTR_CI_LARGE_PAGE:
> > 
> > $ dmesg|grep mmu_features
> > [    0.000000] mmu_features      = 0x7c000003
> > 
> > 
> > You can see 0x20000000 is set.
> 
> They have this:
> root@ubuntu:/var/log# grep mmu_features dmesg* 
> dmesg:[ 0.000000] mmu_features = 0x5c000001
> 
> So it's not set.
> 
> Looks like the bug is in hostboot.  We have this in the device tree in
> linux.
> 
> root@palm6-p1:~# lsprop  /proc/device-tree/cpus/PowerPC\,POWER8@20/ibm
> \,pa-features 
> /proc/device-tree/cpus/PowerPC,POWER8@20/ibm,pa-features
>  00000006 00000000 000000f6 0000003f
>  000000c7 00000000 00000080 000000c0
> 
> That's 32bit properties there but should be 8 bit.
> 
> For openpower, that property is generated from hostboot and skiboot
> just
> passes it through.  
> 
> from:
> https://github.com/open-power/hostboot/blob/master/src/usr/devtree/bld_devtree.C#L763
> 
> It looks like this will fix it in hostboot (totally untested)
> 
> diff --git a/src/usr/devtree/bld_devtree.C
> b/src/usr/devtree/bld_devtree.C
> index ff5fce0..957f1e8 100644
> --- a/src/usr/devtree/bld_devtree.C
> +++ b/src/usr/devtree/bld_devtree.C
> @@ -760,7 +760,7 @@ uint32_t bld_cpu_node(devTree * i_dt, dtOffset_t &
> i_parentNode,
>      * of thread 0 of that core.
>      */
> 
> -    uint32_t paFeatures[8] = { 0x6, 0x0, 0xf6, 0x3f, 0xc7, 0x0, 0x80,
> 0xc0 };
> +    uint32_t paFeatures[2] = { 0x0600f63f, 0xc70080c0 };
>     uint32_t pageSizes[4] = { 0xc, 0x10, 0x18, 0x22 };
>     uint32_t segmentSizes[4] = { 0x1c, 0x28, 0xffffffff, 0xffffffff };
>     uint32_t segmentPageSizes[] =
> 
> Dean, Patrick, this seem like the right fix?  There might be others
> features with the same problem, so that code might need an audit.
> 
> Mikey
> 
>

dcrowell77, 2015-08-18:
I'm confused, the current code already has 8 entries, are you saying that we need to add a 8-bit cell instead of 32-bit?
    uint32_t paFeatures[8] = { 0x6, 0x0, 0xf6, 0x3f, 0xc7, 0x0, 0x80, 0xc0 };
    i_dt->addPropertyCells32(cpuNode, "ibm,pa-features",
                             paFeatures,
                             sizeof(paFeatures)/sizeof(uint32_t));
becomes
    uint8_t paFeatures[8] = { 0x6, 0x0, 0xf6, 0x3f, 0xc7, 0x0, 0x80, 0xc0 };
    i_dt->addPropertyCells8(cpuNode, "ibm,pa-features",
                             paFeatures,
                             sizeof(paFeatures)/sizeof(uint8_t));
?

Kind of shocked this just got noticed after 1.5 years...  I don't even know where these values are documented.

mikey, 2015-08-18:
Yeah, I was shocked too.  The Linux kernel keys a lot of MMU features off the PVR but some come from the device tree.  Cache inhibited large pages is on that comes from the device tree, hence this bug.  
There doesn't seem to be a addPropertyCells8() function in devtree.C, hence I suggested continuing to use addPropertyCells32().  But I don't really care how you implement it.

ozbenh, 2015-08-18:
> I'm confused, the current code already has 8 entries, are you saying
> that we need to add a 8-bit cell instead of 32-bit?
> uint32_t paFeatures[8] = { 0x6, 0x0, 0xf6, 0x3f, 0xc7, 0x0, 0x80,
> 0xc0 }; [...]

They are documented in PAPR, most of them aren't going to be a problem
or event relevant to us, this one is pretty much the only one. We don't
set it by default in our PVR-based table because historically, phyp
didn't support CI large pages even on CPUs that did. We also need to
keep it disabled under KVM when the host page size is smaller than the
guest page size.

Cheers,
Ben.

closed 2015-09-11 by williamspatrick