open-power/hostboot #31: Doesn't build with binutils 2.25

stewart-ibm, 2015-08-20:
In my working tree for updating buildroot, we have the option of going to binutils 2.25 rather than a patched 2.24, but if we enable binutils 2.25, hostboot no longer builds.

I'll update this issue with more information as my tree stabilises.

williamspatrick, 2015-08-21:
Can you copy the build log into a gist output?  I wouldn't be surprised
if something we're using in libelf changed on us.

williamspatrick, 2015-08-21:
Might be caused by #24 too?

stewart-ibm, 2015-08-24:
https://gist.github.com/stewart-ibm/c01c0cf3e947823c5fb9

stewart-ibm, 2015-08-24:
The key bit seems to be:

    MAKE       runtime GEN
/home/stewart/op-build/output/build/host-binutils-2.25/bfd/libbfd.a(plugin.o): In function `try_load_plugin':
plugin.c:(.text+0x2a2): undefined reference to `dlopen'
plugin.c:(.text+0x2b8): undefined reference to `dlsym'
plugin.c:(.text+0x3f8): undefined reference to `dlerror'
collect2: error: ld returned 1 exit status
makefile:41: recipe for target 'tracehash' failed
make[8]: *** [tracehash] Error 1
../../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES
/GEN/BODY' failed

williamspatrick, 2015-08-25:
This looks like a pretty minor miss of -ldl here:

https://github.com/open-power/hostboot/blob/master/src/build/linker/makefile#L29

Can you add as a patch to op-build and see if that resolves?

stewart-ibm, 2015-08-25:
That didn't actually help.. which is odd, but what did help was adding it to
https://github.com/open-power/hostboot/blob/master/src/build/trace/makefile#L32

and that made it build (well, at least Hostboot - haven't tried to see if anything boots yet)

stewart-ibm, 2015-08-25:
I tell a lie, it was actually needed in both places. Pull request coming.

closed 2015-08-26 by williamspatrick