open-power/hostboot #34: System always shows "dumping errors" message

shenki, 2015-09-08:
I have a Plametto machine and it appears that every time I boot I see this on the console:

```
  2.16642|ERRL|Dumping errors reported prior to registration
```

Why does this happen?

Is there something I can do to fix it?

Thanks!

ozbenh, 2015-09-08:
> I have a Plametto machine and it appears that every time I boot I see
> this on the console:
>   2.16642|ERRL|Dumping errors reported prior to registration
> Why does this happen?
> Is there something I can do to fix it?
Afaik it always did that ... even when there's nothing to dump.

dcrowell77, 2015-09-08:
Ben is right, that is normal (though I would admit a bit odd.  If there were errors you would see them there.  This seems like something we could remove, need to poke at the code to see if we know about the existence of logs at the point where we print that.

ploetzma, 2015-10-08:
This would be a good thing for Manali to look into with her error log fix.
We display this trace once the error log handler is enabled - we can just move it into the loop where we find if there are previous error logs.

mingfengwuye, 2015-11-24:
If errors are dumped here, what does that mean? there is an error in last IPL or now?

dcrowell77, 2015-11-24:
Any errors logged at this point are leftover from a previous boot.  There 
are situations where we are not able to send the error log out to the BMC 
during the initial error log.  In those cases we save the data to PNOR to 
report on the next boot as extra information.

message (#34)

shenki, 2015-11-24:
We seem to see these messages on every boot, even after doing a clean shutdown. Can we identify what the error is, and check that everything is working as it should?

dcrowell77, 2015-11-24:
You shouldn't be seeing errors on every boot, just the message "Dumping errors reported prior to registration".  The message alone means nothing.

shenki, 2015-11-24:
Okay. So it will print "Dumping errors" even when it's not dumping any errors?

dcrowell77, 2015-11-24:
Yes, I agree that is less than ideal.  It's in our queue of random things to do.

mingfengwuye, 2015-11-24:
Is there a tool to parsing Errlog in PNOR?

dcrowell77, 2015-11-24:
> Is there a tool to parsing Errlog in PNOR?
Not at this point.  An error log parser is planned to be released to github at some point.  That tool will be able to take in a flattened binary version of an error log and decode the SRC and other details.

mingfengwuye, 2015-11-25:
Is there a schedule for that tool?  
Furthermore, now we analyse IPL errors based on BMC's sel. Is that enough and are there any other errors stored in pnor but not sent to BMC?

dcrowell77, 2015-11-25:
The data in PNOR is only necessary if for some reason we can't send the log out to the BMC.  For most errors you shouldn't need anything except the sel data to replace the bad part.  The full error log is sent to the BMC as an OEM eSEL (0xdf).  That data is useful for more complicated failure analysis.

There is no schedule for release yet.

closed 2015-12-06 by williamspatrick