open-power/hostboot #38: Unused variable errors

shenki, 2015-11-30:
When building with GCC 5, hostboot errors out due to unused variable warnings:

```
In file included from ../../../src/include/usr/targeting/common/predicates/predicateisfunctional.H:58:0,
                 from ../../../src/include/usr/targeting/common/predicates/predicates.H:34,
                 from ../../../src/include/usr/targeting/common/commontargeting.H:39,
                 from compdesc.C:31:
../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
```

A workaround is to build with `-Wno-unused-variable`.

shenki, 2015-12-08:
I suspect the `targetservice.H` error is a real one, due to GCC trying harder to know what is a constant expression and what isn't. Unfortunately it considers the `sizeof()` call to not be constant, which seems a little surprising.

The compiler defines `__SIZEOF_POINTER__`, so an option would be to use that instead.

shenki, 2015-12-08:
Another error. This one looks real:

```
In file included from errlud_i2c.C:33:0:
i2c.H:79:23: error: ‘I2C::g_I2C_NEST_FREQ_MHZ’ defined but not used [-Werror=unused-variable]
 static uint64_t g_I2C_NEST_FREQ_MHZ = i2cGetNestFreq();
                       ^
cc1plus: all warnings being treated as errors
../../../src/build/mkrules/cc.rules.mk:33: recipe for target '../../../obj/modules/i2c/errlud_i2c.o' failed
```

pragupta, 2018-08-17:
Current effort is to able to compile of GCC7, so, closing this one.

closed 2018-08-17 by pragupta