open-power/hostboot #42: Cannot build with -Os

stewart-ibm, 2015-12-09:
Once you convert -Werror to -Wno-error (as various legitimate warnings are emitted), later on:

exception caught: Duplicate weak symbol with contained value member detected: _ZN9SingletonI6PnorDDE8instanceEv with member: _ZGVZN9SingletonI6PnorDDE8instanceEvE8instance

williamspatrick, 2016-01-09:
We can look into why -Os doesn't compile in general, but we don't intend
to compile it that way.  The particular error message you are seeing is
pointing to a real valid problem, so there must be something earlier on
causing this.

$ c++filt _ZGVZN9SingletonI6PnorDDE8instanceEvE8instance
guard variable for Singleton<PnorDD>::instance()::instance

This means there are two symbols in the generated code that are both a
guard variable for the same symbol.  This means there are actually two
copies of the symbol, which is suppose to be a singleton.  Very bad
things could happen if you ran the code like this so we catch it in the
linker and stop the compile.

stewart-ibm, 2016-01-12:
In a parallel discussion, it seems that in some build configurations for debug HB gets too big, so -Os may be something worth investigating.