open-power/hostboot #43: FSPCT: No centaur memory detected after cold reset

saqibkh, 2015-12-15:
No centaur memory was detected after doing a cold reset of the BMC:

System Info:
BMC ip address= 9.41.164.80 name=habcap10.aus.stglabs.ibm.com
OS ip address=9.41.164.12 name=habcap10p1.aus.stglabs.ibm.com
----------------------------------------------------------------------------------------------------------
Following is a list of commands that were executed and their outcome:

HOST OS: $ sudo /home/tyan/skiboot/external/xscom-utils/getscom -l
[sudo] password for tyan:
Chip ID | Rev | Chip type
---------|-------|--------
80000005 | DD2.0 | Centaur memory buffer
80000004 | DD2.0 | Centaur memory buffer
80000001 | DD2.0 | Centaur memory buffer
80000000 | DD2.0 | Centaur memory buffer
00000000 | DD2.0 | P8 (Venice) processor
ipmitool -H habcap10 -I lanplus -U ${Userid} -P ${Password} mc cold reset

HOST OS: $ sudo /home/tyan/skiboot/external/xscom-utils/getscom -l
[sudo] password for tyan:
Chip ID | Rev | Chip type
---------|-------|--------
00000000 | DD2.0 | P8 (Venice) processor <=========== No Centaur Memory

Also attached is an output obtained from the serial port, which shows that the BMC was unable to write/detect these memory.
[image]

------------------------------------------------------------------------------------------------------------
Following is the response from skiboot team:
This isn't a skiboot problem. HostBoot didn't forward the information about the Centaurs to us.

dcrowell77, 2015-12-15:
We'll need to see the devtree to confirm the above statements.  

I'm a little confused by the order.  Are you saying you did a mc reset and the output from getscom -l changed without a box reboot in the middle?

saqibkh, 2015-12-15:
devtree is stored in /esw/san2/khansa/files/data.txt.
If you can't access this file, then let me know a location and I'll move the file there.

Yes I did a cold reset and the output from "getscom -l" changed without any sort of reboot (or any other command).

dcrowell77, 2015-12-15:
Since the CEC didn't reboot, Hostboot can't be at fault.  We only run during the boot.

The devtree file shows the centaur entries I expect to see.
	memory-buffer@3e02000000000 {
		reg = <0x3e020 0x0 0x8 0x0>;
		ecid = "019A008300106718041D0000006403F1";
		compatible = "ibm,centaur", "ibm,centaur-v20";
		ibm,fsi-master-chip-id = <0x0>;
		ibm,chip-id = <0x80000001>;
	memory-buffer@3e08000000000 {
		reg = <0x3e080 0x0 0x8 0x0>;
		ecid = "019A0083011017180D1B000000640391";
		compatible = "ibm,centaur", "ibm,centaur-v20";
		ibm,fsi-master-chip-id = <0x0>;
		ibm,chip-id = <0x80000004>;
	memory-buffer@3e00000000000 {
		reg = <0x3e000 0x0 0x8 0x0>;
		ecid = "019A0083011017180B1B0000006203C1";
		compatible = "ibm,centaur", "ibm,centaur-v20";
		ibm,fsi-master-chip-id = <0x0>;
		ibm,chip-id = <0x80000000>;
	memory-buffer@3e0a000000000 {
		reg = <0x3e0a0 0x0 0x8 0x0>;
		ecid = "019A0083011037181318000000B402B1";
		compatible = "ibm,centaur", "ibm,centaur-v20";
		ibm,fsi-master-chip-id = <0x0>;
		ibm,chip-id = <0x80000005>;

I saved the file here as well.
[data.txt](https://github.com/open-power/hostboot/files/63018/data.txt)

ozbenh, 2015-12-15:
> Since the CEC didn't reboot, Hostboot can't be at fault. We only run
> during the boot.
> The devtree file shows the centaur entries I expect to see.

Ok, my fault, ignore my previous statement. So the problem would
then be that I'm failing to contact them via FSI.

Can you post the skiboot log after the attempt ?
(/sys/firmware/opal/msglog)

Cheers
Ben.

saqibkh, 2015-12-16:
skiboot log is stored in /esw/san2/khansa/files/msglog.txt

saqibkh, 2015-12-16:
[msglog.txt](https://github.com/open-power/hostboot/files/64477/msglog.txt)

ozbenh, 2015-12-16:
On Wed, 2015-12-16 at 11:33 -0800, saqibkh wrote:
> skiboot log is stored in /esw/san2/khansa/files/msglog.txt

Sorry I don't know how to get to these files, can you email it to me ?

Cheers,
Ben.

ozbenh, 2015-12-16:
Is this actually a Habanero machine or some other machine ?

If this is a habanero, please update to a more recent skiboot. This is
ancient, I have fixed a number of issues with FSI error recovery since
then.

If it's an other machine, why does it advertize being a Habanero ?

Cheers,
Ben.

saqibkh, 2015-12-16:
Hi Ben!

I attached a msglog above for you to access.

This is a Habanero system with the latest pnor code available to me:
 Product Name          : OpenPOWER Firmware
 Product Version       : IBM-habanero-ibm-OP8_v1.7_1.18
 Product Extra         : 	hostboot-bc98d0b-ac044ba
 Product Extra         : 	occ-0362706-6130f92
 Product Extra         : 	skiboot-5.1.9-8555b6b
 Product Extra         : 	hostboot-binaries-43d5a59
 Product Extra         : 	habanero-xml-a71550e-e2bb81a
 Product Extra         : 	capp-ucode-105cb8f

Thanks,
Saqib Khan

ozbenh, 2015-12-16:
On Wed, 2015-12-16 at 13:25 -0800, saqibkh wrote:
> I attached a msglog above for you to access.
> This is a Habanero system with the latest pnor code available to me:
> Product Name : OpenPOWER Firmware
> Product Version : IBM-habanero-ibm-OP8_v1.7_1.18 [...]

Hrm, the skiboot claims to be a 5.0-rc3 in the log... not sure what's
going on here.

Cheers,
Ben.

saqibkh, 2015-12-17:
Hi Ben!

Most probably cause seems to be that the system booted from the golden side of PNOR (since I was doing error injection earlier), thus having an old skiboot version.

However I have verified this issue again being on the primary side and having the latest PNOR code.

Attached is the skiboot/msglog.

Thanks,
Saqib Khan
[msglog.txt](https://github.com/open-power/hostboot/files/65868/msglog.txt)

ozbenh, 2015-12-17:
> Hi Ben!
> Most probably cause seems to be that the system booted from the
> golden side of PNOR (since I was doing error injection earlier), thus
> having an old skiboot version.
> However I have verified [...]

I'm getting XCSOM errors trying to talk to the centaur over FSI. I
don't know what's up, it seems to be a HW problem.

Ben.

saqibkh, 2016-01-05:
Hi Ben,

I verified the same defect on another habanero system.
---------------------------------------------------------------------------------------
Following are the system details:
hablc2.aus.stglabs.ibm.com ---> 9.3.181.38
OS# hablc2p1.aus.stglabs.ibm.com ----> 9.3.181.101

khansa@habcap11p1:~$ ipmitool -H hablc2 -I lanplus -U ADMIN -P admin fru list 43 Product Name          : OpenPOWER Firmware
 Product Version       : IBM-habanero-ibm-OP8_v1.7_1.17
 Product Extra         :        hostboot-bc98d0b-5463ef4
 Product Extra         :        occ-0362706-9e016d2
 Product Extra         :        skiboot-5.1.9-ada75ff
 Product Extra         :        hostboot-binaries-43d5a59
 Product Extra         :        habanero-xml-a71550e-85cdcb7
 Product Extra         :        capp-ucode-105cb8f

Thanks,
Saqib Khan

saqibkh, 2016-01-05:
Attached is the msglog from hablc2
[hablc2p1.txt](https://github.com/open-power/hostboot/files/78712/hablc2p1.txt)

ozbenh, 2016-01-05:
On Tue, 2016-01-05 at 09:12 -0800, saqibkh wrote:
> Hi Ben,
> I verified the same defect on another habanero system.
> Following are the system details:
> hablc2.aus.stglabs.ibm.com ---> 9.3.181.38
> OS# hablc2p1.aus.stglabs.ibm.com ----> 9.3.181.101 [...]

I don't know what is happening here. This seems to be a HW issue,
basically the Centaur stops responding on the FSI. I have no idea how a
BMC reset can cause that.

Cheers,
Ben.

saqibkh, 2016-01-05:
Hi Ben!
Do you know whom I should contact for this issue?

ozbenh, 2016-01-06:
On Tue, 2016-01-05 at 15:55 -0800, saqibkh wrote:
> Hi Ben!
> Do you know whom I should contact for this issue?

Not sure no, Dan do you know ?

Cheers,
Ben.

dcrowell77, 2016-01-06:
I'll poke around and see if I can find someone.

dcrowell77, 2016-01-06:
One possibility came up in discussions today.  The BMC controls a pin to the processor that tells it where to get its FSI clocks.  If this pin toggles during the reset, the processor would lose the clocks and everything downstream would go wacky.  @amiviswa may be able to provide an answer.

ozbenh, 2016-01-06:
On Wed, 2016-01-06 at 13:19 -0800, Dan wrote:
> One possibility came up in discussions today. The BMC controls a pin
> to the processor that tells it where to get its FSI clocks. If this
> pin toggles during the reset, the processor would lose the clocks and
> everything downstream would go wacky. @amiviswa may be able to
> provide an answer.

I have seen that toggling these tends to crash the CFAM yes, this also
tends to crash the SBE, which means OPAL loses the timer functionality,
which is eventually going to prevent proper operations of a number of
things.

Cheers,
Ben.

stewart-ibm, 2018-08-07:
Realistically, nothing is ever going to change to even possibly fix it with an AMI BMC and P8. It unlikely makes any sense to keep this issue open unless it's reproduced with an OpenBMC P8 platform or a P9 platform.

closed 2018-08-07 by stewart-ibm