open-power/hostboot #44: Hostboot broke op-build

shenki, 2015-12-16:
This commit from November breaks building the Palmetto pnor https://github.com/open-power/hostboot/commit/02353bc77297c319cdb02b6f83473c4f533560b3

Reverting it fixes the build.

```
###############################/scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//buildpnor.pl --pnorOutBin /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/pnor/palmetto.pnor --pnorLayout /scratch/joel/op-build/output/build/openpower-pnor-40e407735b317d0174645b7543e0a72019709ce2/defaultPnorLayoutSingleSide.xml --binFile_HBD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//PALMETTO_HB.targeting.bin.ecc --binFile_SBE /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//venice_sbe.img.ecc --binFile_SBEC /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//centaur_sbec_pad.img.ecc --binFile_WINK /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//p8.ref_image.hdr.bin.ecc --binFile_HBB /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hostboot.header.bin.ecc --binFile_HBI /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hostboot_extended.header.bin.ecc --binFile_HBRT /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hostboot_runtime.header.bin.ecc --binFile_HBEL /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hbel.bin.ecc --binFile_GUARD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//guard.bin.ecc --binFile_PAYLOAD /scratch/joel/op-build/output/images/skiboot.lid --binFile_BOOTKERNEL /scratch/joel/op-build/output/images/zImage.epapr --binFile_NVRAM /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//nvram.bin.ecc --binFile_MVPD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//mvpd_fill.bin.ecc --binFile_DJVPD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//djvpd_fill.bin.ecc --binFile_CVPD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//cvpd.bin.ecc --binFile_ATTR_TMP /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//attr_tmp.bin.ecc --binFile_ATTR_PERM /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//attr_perm.bin.ecc --binFile_OCC /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/occ/occ.bin.ecc --binFile_FIRDATA /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//firdata.bin.ecc --binFile_CAPP /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//cappucode.bin.ecc --binFile_VERSION /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/openpower-pnor.version.txt --fpartCmd "fpart" --fcpCmd "fcp"
TRACE:  main::loadPnorLayout: metadata: imageSize = 33554432, blockSize=4096, arrangement = A-D-B, numOfSides: 1, sideSize: 33554432, tocSize: 32768
TRACE: A-D-B: side:A HBB:32927744, primaryTOC:0, backupTOC:33521664, golden: no
TRACE: Done checkSpaceConstraints
TRACE: createPnorPartition:: 33521664
TRACE: createPnorImg:: 33521664
TRACE: createPnorPartition:: 0
TRACE: createPnorImg:: 0
TRACE: fpart --target /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/pnor/palmetto.pnor --partition-offset 33521664 --add --offset 33521664 --size 32768 --name BACKUP_PART --flags 0x0
fpart: unexpected : ../src/libffs.c(968) : (code=-1) 'BACKUP_PART' at offset 33521664 and size 32768 overlaps 'part' at offset 33521664 and size 4096
ERROR: Call to add partition BACKUP_PART failed. Aborting! at /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//buildpnor.pl line 534.
Error running command: /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//buildpnor.pl --pnorOutBin /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/pnor/palmetto.pnor --pnorLayout /scratch/joel/op-build/output/build/openpower-pnor-40e407735b317d0174645b7543e0a72019709ce2/defaultPnorLayoutSingleSide.xml --binFile_HBD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//PALMETTO_HB.targeting.bin.ecc --binFile_SBE /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//venice_sbe.img.ecc --binFile_SBEC /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//centaur_sbec_pad.img.ecc --binFile_WINK /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//p8.ref_image.hdr.bin.ecc --binFile_HBB /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hostboot.header.bin.ecc --binFile_HBI /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hostboot_extended.header.bin.ecc --binFile_HBRT /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hostboot_runtime.header.bin.ecc --binFile_HBEL /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hbel.bin.ecc --binFile_GUARD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//guard.bin.ecc --binFile_PAYLOAD /scratch/joel/op-build/output/images/skiboot.lid --binFile_BOOTKERNEL /scratch/joel/op-build/output/images/zImage.epapr --binFile_NVRAM /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//nvram.bin.ecc --binFile_MVPD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//mvpd_fill.bin.ecc --binFile_DJVPD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//djvpd_fill.bin.ecc --binFile_CVPD /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//cvpd.bin.ecc --binFile_ATTR_TMP /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//attr_tmp.bin.ecc --binFile_ATTR_PERM /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//attr_perm.bin.ecc --binFile_OCC /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/occ/occ.bin.ecc --binFile_FIRDATA /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//firdata.bin.ecc --binFile_CAPP /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//cappucode.bin.ecc --binFile_VERSION /scratch/joel/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/openpower-pnor.version.txt --fpartCmd "fpart" --fcpCmd "fcp". Nonzero return code of (256) returned.
package/pkg-generic.mk:267: recipe for target '/scratch/joel/op-build/output/build/openpower-pnor-40e407735b317d0174645b7543e0a72019709ce2/.stamp_images_installed' failed
make: *** [/scratch/joel/op-build/output/build/openpower-pnor-40e407735b317d0174645b7543e0a72019709ce2/.stamp_images_installed] Error 1
make: Leaving directory '/scratch/joel/op-build/buildroot'
```

stewart-ibm, 2015-12-16:
The script and ffs itself are generally awful.

I'm working on replacing it all with simple calls to pflash. I may add 100 LoC to pflash, but otherwise remove 20,000 of ffs garbage and 994 lines of perl.

Also, the process is non deterministic in some other ways too. Not fun times.

shenki, 2015-12-16:
Thanks @stewart-ibm.

How did this change get merged into op-build?

stewart-ibm, 2015-12-16:
Good question. You'd expect the jenkins to catch it. But....

dcrowell77, 2015-12-16:
We compile op-build master and master-next for all 4 platforms almost every day so is it just a timing issue?  @scprek and @jazurin please take a look.

williamspatrick, 2015-12-16:
It seems to be a bug in the sorting of some map in the Perl script. Depending on the system you have a different probability of success. We haven't seen it fail on RHEL but do see it fail on ppc64le Ubuntu. I had a proposed fix but it didn't seem to resolve everything.  

Patrick

williamspatrick, 2015-12-16:
Stewart,

Before you spend a lot of time moving the script over to pflash, we need to have a discussion with regards to FSP image building. The reason why we are using ffs (and this ugly Perl script) to begin with is because the FSP build system already relies on having ffs for its own partition building.

Before we deprecate this script we have to get the fsp team to support building with pflash. 

Patrick

scprek, 2015-12-16:
My comments are all in the pull request #45

jazurin, 2015-12-16:
@shenki could you please let us know what operating system you are using to build op-build?  Is it RHEL, ppc64le Ubuntu, or something else?   As @williamspatrick said, we saw similar failures on ppc64le Ubuntu.  The failures were caught by a nightly Jenkins job that runs a master-next build test on ppc64le Ubuntu.  But the job is not part of CI, though we might consider adding it to the CI pipeline.  Thanks.

shenki, 2015-12-16:
Hey @jazurin, I'm seeing the failure on our public Jenkins instance at https://openpower.xyz. 
This runs the jobs inside an Ubuntu 15.10 amd64 Docker container.

 https://openpower.xyz/job/openpower-op-build/distro=ubuntu,target=palmetto/lastBuild/console

I also see the failure on my work laptop, which is has an Ubuntu 16.04 toolchian (gcc 5.3, binutils2.25.90.20151211-0ubuntu2, perl v5.20.2).

I think that adding an Ubuntu builder to your CI pipeline would be very useful. Those of us on the LTC side of the fence are using modern Ubuntu and Fedora installs for our development machines as well as our own CI.

Looking at bug reports from OpenPower developers in the community, it appears Ubuntu is the most popular option, so we should ensure that they have the best chance possible of building the firmware.

Ping me, @stewart-ibm  or @csmart on IRC if you would like to reuse any of the infrastructure we have created for doing builds.

stewart-ibm, 2015-12-17:
I have also seen this on Fedora 23 x86-64, Debian stable x86-64 and various Ubuntu docker containers in a PowerKVM guest on ppc64el.

stewart-ibm, 2015-12-17:
@williamspatrick We could leave the script there for FSP so they don't have to switch at the same time and can do so if they wish and on their own schedule.

closed 2016-01-09 by williamspatrick