open-power/hostboot #46: about XSCOM

cooktail, 2016-01-05:
Hi williamspatrick,
I boot op-build1.7, And there is an error message on IPL.
Can you give me some advise?thanks.
What's the userdata1 mean?Can you give me some matirial of xscom? I don't the function and theory of xscom.
UserData1 HMER value (piberr in bits 21:23) : 0x00c0040000000000

In spec bits 21:23:
HMER0_XSCOM_STATUS: 000 = no error
001 = XSCOM blocked due to pending error
010 = chiplet offline
011 = partial good
100 = invalid address / address error
101 = clock error
110 = parity err
111 = time out

110.60816|================================================ 
110.64730|Error reported by xscom (0x0400) 
110.65466| XSCom access error 
110.65467| ModuleId 0x07 XSCOM_DO_OP 
110.65467| ReasonCode 0x0401 XSCOM_STATUS_ERR 
110.65467| UserData1 HMER value (piberr in bits 21:23) : 0x00c0040000000000 
110.65468| UserData2 XSCom address : 0x00000000006b0328 
110.65468|User Data Section 0, type UD 
110.65468| Subsection type 0x05 
110.65468| ComponentId errl (0x0100) 
110.65469|User Data Section 1, type UD 
110.65469| Subsection type 0x15 
110.65469| ComponentId hb-trace (0x3100) 
110.65469|User Data Section 2, type UD 
110.65470| Subsection type 0x04 
110.65470| ComponentId errl (0x0100) 
110.65470|User Data Section 3, type UD 
110.65470| Subsection type 0x06 
110.65471| ComponentId errl (0x0100) 
110.65471| CALLOUT 
110.65471| HW CALLOUT 
110.65471| Reporting CPU ID: 8 
110.65471| Called out entity: 
110.65472|User Data Section 4, type UD 
110.65472| Subsection type 0x06 
110.65472| ComponentId errl (0x0100) 
110.65472| CALLOUT 
110.65472| PROCEDURE ERROR 
110.65473| Procedure: 85 
110.65473|User Data Section 5, type UD 
110.65473| Subsection type 0x15 
110.65473| ComponentId hb-trace (0x3100) 
110.65474|User Data Section 6, type UD 
110.65474| Subsection type 0x15 
110.65474| ComponentId hb-trace (0x3100) 
110.65474|User Data Section 7, type UD 
110.65475| Subsection type 0x15 
110.65475| ComponentId hb-trace (0x3100) 
110.65475|User Data Section 8, type UD 
110.65475| Subsection type 0x03 
110.65476| ComponentId errl (0x0100) 
110.65476|User Data Section 9, type UD 
110.65476| Subsection type 0x01 
110.65476| ComponentId errl (0x0100) 
110.65476| STRING 
110.65477| Hostboot Build ID: 
110.65477|User Data Section 10, type UD 
110.65477| Subsection type 0x04 
110.65477| ComponentId errl (0x0100) 
110.65478|================================================

dcrowell77, 2016-01-05:
XSCOM is the host-based method to access the SCOM logic on the chip.  SCOMs are a method on POWER chips to access internal chip logic while the system is running.  If you are familiar with boundary scan rings, SCOM is a way to access some of the bits within those rings while running functionally with clocks on.

The error is being logged here - https://github.com/open-power/hostboot/blob/master/src/usr/xscom/xscom.C#L516.  So what you have is an error to scom address 0x6b035, which is part of the OCC logic.  Historically we've seen this error when the OCC didn't set some of the hardware up correctly.  What platform are you running on?

williamspatrick, 2016-01-05:
I haven't seen this XSCOM error before.  Dan, this is the register for
the OCB channel 1.  I think we enabled some of the code for IPL-time
checkstop analysis in v1.7.  Anything you can think of there that might
be causing it?

dcrowell77, 2016-01-05:
I don't think we ever saw this error during bringup of that function, but there definitely is related code.  @cooktail Are there any other messages?  Can you copy in the entire console output from a clean run (reflash the code)?  Also, clear the sel list and send the results after the boot fails.

touchqvfb, 2016-01-06:
Thanks for reply,The issue occur probability,maybe it need 100 times reboot,But I can open trace,then reboot,when it happen again,I send you the log,And PLS help me to Analysis. 
what's 0x00c0040000000000 mean? Is XSCOM blocked due to pending error?
UserData1 HMER value (piberr in bits 21:23) : 0x00c0040000000000
In spec bits 21:23:
HMER0_XSCOM_STATUS: 
000 = no error
001 = XSCOM blocked due to pending error
010 = chiplet offline
011 = partial good
100 = invalid address / address error
101 = clock error
110 = parity err
111 = time out
thanks again!

dcrowell77, 2016-01-06:
The error is '100' actually (remember, we are big-endian so we count bit zero on the left).  However that decode is misleading.  As mentioned above, we've seen this error before in the lab and it has been related to something amiss on the OCC side of things.  @cjcain Do you have any suggestions on what else to look at?

touchqvfb, 2016-01-07:
op-build v1.6 also have this issue.

cooktail, 2016-03-22:
171.22345|FAPI|p8_ocb_init.C: Executing p8_ocb_init ....
171.22345|FAPI|p8_ocb_init.C:  *** Setup OCB Indirect Channel 0
171.22347|FAPI_I|p8_ocb_init.C: -----------------------------------------------------
171.22347|FAPI_I|p8_ocb_init.C: OCB Channel Configuration
171.22348|FAPI_I|p8_ocb_init.C: -----------------------------------------------------
171.22348|FAPI_I|p8_ocb_init.C:   channel number             => 0
171.22349|FAPI_I|p8_ocb_init.C:   channel type               => 2
171.22349|FAPI_I|p8_ocb_init.C:   channel base address       => 0xFFFF7000
171.22350|FAPI_I|p8_ocb_init.C: -----------------------------------------------------
171.22350|FAPI|p8_ocb_init.C: Completing p8_ocb_init ....
171.22351|FAPI|p8_ocb_indir_access.C: Executing p8_ocb_indir_access op 1 channel 0 of length (in 8B) 1....
171.32353|HTMGT|E>waitForOccCheckpoint OCC0 still NOT ready!
171.32353|HTMGT|Send initial poll to all OCCs to establish comm
171.31333|HTMGT|sendOccPoll(flush=n)
171.31334|HTMGT|sendOccPoll: Polling OCC0
171.31334|HTMGT|Send OCC0 cmd=0x00 (POLL), length=0x0001
171.31336|HTMGT|cmd data (up to 16 bytes)
~[0x0000] 10                                     *.               *
171.31337|HTMGT|buildOccCmdBuffer: OCC command (up to 300 bytes)
~[0x0000] 01000001 100012                        *.......         *
171.31339|FAPI|p8_ocb_indir_access.C: Executing p8_ocb_indir_access op 2 channel 1 of length (in 8B) 1....
171.31340|XSCOM|E>XSCOM status error HMER: 00c0040000000000 ,XSComStatus = 4, Addr=6b035
171.31342|XSCOM|collectXscomFFDC: XSCOM COLLECT FFDC STARTED
171.31356|XSCOM|resetScomEngine: XSCOM RESET INTIATED
171.30626|XSCOM|xscomPerformOp - RESOURCE OCCUPIED LOOP Cntr = 128: OpType 0x0000000000000001, Address 0xB0020, MMIO Address 0xB00200, HMER=0000000000000000
171.31365|ERRL|>>addHwCallout(0x00050000 0x6 0x2 0x0)
171.31366|TARG|[TARG] E> Number of Parent chip is not 1, but 0
171.31367|ERRL|>>addProcedureCallout(0x55, 0x5)
171.31368|FAPI_I|fapiReturnCode.C: setPlatError: Creating PLAT error 0x2000001
171.31369|FAPI_I|fapiHwAccess.C: fapiPutScom failed - Target p8 k0:n0:s0:p00       , Addr 000000000006B035
171.31370|FAPI_I|p8_ocb_indir_access.C: Put SCOM error for OCB indirect access
171.31370|FAPI_I|fapiPlatHwpInvoker.C: fapiRcToErrl: PLAT error: 0x02000001
171.31373|FAPI_I|E>accessOCBIndirectChannel: Error [0x401] in call to FAPI_INVOKE_HWP(p8_ocb_indir_access)
171.31374|HTMGT|E>writeOccCmd: Error writing to OCC Circular Buffer, rc=0x0401
171.31375|ERRL|commitErrLog() called by 2600 for eid=9000000c, Reasoncode=0401
171.33786|HWAS_I|I>deconfigGard.C: registerDeferredDeconfigure Target 00050000, errlEid 0x9000000C
171.33787|HWAS_I|I>deconfigGard.C: Creating a Deconfigure Record
171.33787|ERRL|>>ErrlManager::sendErrlogToMessageQueue
171.33788|ERRL|I>Send an error log to message queue to commit. eid=9000000C
171.33788|ERRL|<<ErrlManager::sendErrlogToMessageQueue
171.33791|ERRL|I>Got an error log Msg - Type: 0x00000030
171.37161|================================================
171.37162|Error reported by xscom (0x0400)
171.40800|  XSCom access error
171.40800|  ModuleId   0x07 XSCOM_DO_OP
171.40801|  ReasonCode 0x0401 XSCOM_STATUS_ERR
171.40801|  UserData1  HMER value (piberr in bits 21:23) : 0x00c0040000000000
171.40802|  UserData2  XSCom address : 0x00000000006b0328

williamspatrick, 2016-03-23:
@dcrowell77 / @cjcain - I got an email with this information but I don't see
it in the issue proper.

dcrowell77, 2016-03-23:
I thought I was going crazy, I figured it was just a glitch repeating the first entry or something.

That xscom error is what we get when the OCC has not properly configured the circular buffer and/or it is not pulling things out of the FIFO.  This typically means the OCC is dead or never started.

(Side note, I have a commit out there to change the callouts for this specific address range, but it doesn't really clear up the issue if you don't look at the callouts in the error log and/or the comments in the code.)

cjcain, 2016-03-23:
Is the full ipl trace available?  If HTMGT was not able to communicate with the OCC we will still try to read SRAM and see if there are any OCC exceptions.  (Traces would be after the xscom error above)

dcrowell77, 2017-05-01:
No response for a year, closing

closed 2017-05-01 by dcrowell77