open-power/hostboot #47: Non-present FRUs are being reported as faulted.

williamspatrick, 2016-01-09:
In our IPMI code for OpenBMC, we observed that the hostboot code is setting the fault bit in the FRU sensor for every non-present device.  For DIMM slots that are not populated and for processors with partial-good records, this is probably not the best option.  Can this be changed in the host firmware without breaking other BMC implementations?

See openbmc/openbmc#92.

dcrowell77, 2016-01-11:
I'm a little confused here.  What is meant by "fault bit" here?  We do send down information for every part, even the non-present ones, but that data should indicate "not present".  We need to send down something to handle the case where the previous boot had detected the part and we need to clear the old status out.  (Adding @wghoffa and @rjknight to the discussion)

williamspatrick, 2016-01-12:
The ipmi fru status sensor has multiple "offsets" asserted. Non-present and non-functional. Hostboot has a simple translation from hwas bits to ipmi bits. 1 to 1 mapping. Instead we should map non-present (plus non-functional) hwas to just non-present ipmi. 

Patrick

rjknight, 2016-01-12:
The current mappings are

 States  | IPMI bits
------------ | -------------
NOT_PRESENT | turn off the present bit, turn on the disabled bit
PRESENT | turn on the present bit
FUNCTIONAL | turn off the disabled bit
PRESENT_FUNCTIONAL | turn on the present bit, turn off the disabled bit
PRESENT_NONFUNCTIONAL | turn on the present bit, turn on the disabled bit
NON_FUNCTIONAL  | turn on the disabled bit

Sounds like the suggestion is to update the NOT_PRESENT to turn off the present bit and the disabled bit?

williamspatrick, 2016-01-12:
Yes, exactly. 

Patrick

dcrowell77, 2016-01-13:
I'm checking with our service team to see if there are any legacy support issues, so far this request seems reasonable.

dcrowell77, 2016-01-15:
There were no objections to this change so have at it.

closed 2016-02-18 by ploetzma