open-power/hostboot #49: VPD Caching Issue with CVPD when SN/PN not programmed

wghoffa, 2016-01-25:
On a recent system bringup, I was encountering several VPD errors related to CVPD. All the reads were resulting in errors stating the entire record/keyword was missing. This was a configuration wiith combined centaur + planar VPD on one chip. The issue was that when CVPD PNOR Caching was enabled, the code determined the cache was up to date and did not correctly populate the PNOR partition. I believe this was due to no PN/SN being programmed in the VPD at this point. I think we need to add an additional check to the logic determining whether to write the VPD cache or not taking into account an empty PN/SN scenario.

dcrowell77, 2016-01-25:
Adding @cvswen

wghoffa, 2016-01-25:
Note - we should be getting a board this week so this scenario could be recreated/verified as well.

pragupta, 2018-08-17:
Code should be there.. just double check..

cvswen, 2018-08-17:
Currently if SEEPROM VPD part number and serial number are 0 and values in our VPD cache are 0 then we do not write the VPD cache.  Need to always write the VPD cache in this case.