open-power/hostboot #52: [FATAL!] Could not find enumeration with ID of PCI_BASE_ADDRS_64

naynajain, 2016-02-17:
Hi,
I am doing op-build for habanero on Fedora 23 distro
After fixing various perl missing packages like XML::Simple, XML::Parser, XML::SAX, type could make hostboot build to proceed.
After that as it started building at openpower-mrw, it failed with following error

reating XML: /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/HABANERO_hb.mrw.xml
MRW created successfully!
# merge in any system specific attributes, hostboot attributes
/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/mergexml.sh /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"HABANERO_hb.system.xml" /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types.xml /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types_hb.xml /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_merged.xml /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_hb.xml /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"HABANERO_hb.mrw.xml" > /home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml;
# creating the targeting binary
/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl --hb-xml-file=/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml --fapi-attributes-xml-file=/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/fapiattrs.xml --src-output-dir=none --img-output-dir=/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/ --vmm-consts-file=/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/vmmconst.h --noshort-enums --bios-xml-file=/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"HABANERO_bios.xml" --bios-schema-file=/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/bios.xsd --bios-output-file=/home/nayna/Project/Code/16022016next/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/"habanero"_bios_metadata.xml
Warning: XML::LibXML compiled against libxml2 20903, but runtime libxml2 is older 20902
[FATAL!] Could not find enumeration with ID of PCI_BASE_ADDRS_64

     1: main::getEnumerationType(1222)
     2: main::packSingleSimpleTypeAttribute(4673)
     3: main::packAttribute(4746)
     4: main::generateTargetingImage(5453)
     5: (369)

naynajain, 2016-02-17:
Forgot to add the commit level on which my branch is

[nayna@swastik hostboot-390b72b029ddee979a1e0350db8fa521bf15fb52]$ git show
commit 156807c06f43f162018875ac7e8748e098894f05
Merge: 2a71078 d9669fe
Author: Patrick Williams <patrick@stwcx.xyz>
Date:   Mon Feb 8 14:57:49 2016 -0600

    Merge pull request #339 from shenki/kernel-4.4
    
    Release 4.4-openpower2 kernel

williamspatrick, 2016-02-17:
Can you confirm this is a 64 bit Linux machine?  The build requires 64 bit and this looks similar to one of the error symptoms when running on a 32 bit machine. 

Patrick

naynajain, 2016-03-15:
[nayna@swastik logs]$ uname -a
Linux swastik 4.4.4-301.fc23.x86_64 #1 SMP Fri Mar 4 17:42:42 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

pragupta, 2018-08-17:
Seems like a specific compile issue, no activity closing..

closed 2018-08-17 by pragupta