open-power/hostboot #53: Hostboot hangs without VUART support

anoo1, 2016-02-19:
Reference openbmc issue: https://github.com/openbmc/openbmc/issues/132

Starting with hostboot commit https://github.com/open-power/hostboot/commit/1fbe5e7bf5042ee9e7353a169d13eec54c270f04 , hostboot hangs on systems where the BMC firmware does not have VUART support and relies on the HW only.

anoo1, 2016-02-19:
The hostboot printk stops at:
....
        Initing module libvfs.so...done.
        Initing module libsio.so...done.
Modules initialized.
init_main: Starting Initialization Service...
InitService entry.
ExtInitSvc entry.
ast2400: SUART config in process
Unhandled msg rc 4294967291 for key 0x8093b000 on task 10 @ 0x15d88
Unhandled msg rc 4294967291 for key 0x80016000 on task 17 @ 0x15c60
Unhandled msg rc 4294967291 for key 0x40015000 on task 32 @ 0x666dc

williamspatrick, 2016-02-22:
@anoo1 - There is a scratch register that needs to be populated by the BMC to tell Hostboot to use the VUART instead of the SUART.  Is that being done?  The printk here implies not.

anoo1, 2016-02-23:
Closing, hostboot support already in place.

anoo1, 2016-02-26:
What is the scratch register that needs to be populated? Need to verify that physical uart would work.

williamspatrick, 2016-02-26:
See this code:
https://github.com/open-power/hostboot/blob/master-p8/src/usr/console/ast2400.C#L281

Register 0x28 and 0x2d of the SUART appear to be the scratch registers we use.  I think @sannerd owns the spec on these.

dcrowell77, 2016-02-26:
Note that all of the scratch regs must contain valid data in order to use any of them.

mdmillerii, 2016-07-11:
The problem is when `VUART` is **disabled** not **enabled**.  Or more precisely, when **`SUART`** is enabled.

The system stopped hanging with the new hostboot when the `VUART` support was added to OpenBMC and the `VUART` was enabled in the scratch register.  The earlier hostboot worked with the old code and setup and continued to work with the new `VUART` setup.

I think this is the openbmc kernel commit that changed the contents of the scratch register via the ast2400 side address: openbmc/linux@271462b84385ba270690a40bc9ebc32dea6db020

    -       writel(0x00004000, AST_IO(AST_BASE_LPC | 0x174));
    +       writel(0x00008000, AST_IO(AST_BASE_LPC | 0x174));

williamspatrick, 2016-07-11:
@shenki - Two questions.

1. Why is this code in the kernel at all?  The scratch registers (or at least their interpretation) is openpower-specific.
2. Did you pick this patch up into the 4.7 tree for Witherspoon?

mdmillerii, 2016-07-11:
Responding to @williamspatrick questions to @shenki :

1) The setup is with the "we don't have pinmux" workarounds and is platform specific (keyed off of the device tree `compatible` property).  Or better stated "don't use devmem from userspace to setup the machine" section of the kernel.

The scratch registers eventually need to be exported to user space for set up.  Since they don't seem to have any direct operation on the kernel we need to find the proper abstraction for them.  I have suggested the `nvmem` subsystem as a candidate.  Although this expands existing users, which are exporting hardware fuses with thermal calibration data and mac addresses, it seems to be in the scope of the subsystem statement.

2) The change to request the vuart is in the `dev-4.4` and `dev-4.6` branches.
[dev-4.6:arch/arm/mach-aspeed/aspeed.c](https://github.com/openbmc/linux/blob/dev-4.6/arch/arm/mach-aspeed/aspeed.c#L=122)

        /* Setup scratch registers */
        writel(0x00000042, AST_IO(AST_BASE_LPC | 0x170));
        writel(0x00004000, AST_IO(AST_BASE_LPC | 0x174));

Again, this hostboot defect is to address the problem selecting the `SUART` in the scratch register does not result in a working system, even if the `VUART` is disabled.  When the openbmc/openbmc#132 was written @openbmc did not have support for `VUART` and all builds after 1fbe5e7bf5042ee9e7353a169d13eec54c270f04 did not work until we added our `VUART` support.

williamspatrick, 2016-07-11:
I'm not working on the Hostboot team anymore, so I removed my ownership of this defect.  Per Milton, there is some suspicion that the SUART code no longer works in Hostboot but the VUART (used in most cases) does.

dcrowell77, 2016-07-12:
Can someone ( @mdmillerii or @anoo1 ) confirm that when the SUART attempt was made that all of the scratch regs were set to valid values?  The use of the SIO bits is an all-or-nothing thing.

mdmillerii, 2016-07-12:
I can confirm the only change made to the scratch registers from todays code is the one bit change shown above, which seems to match the comment in the code.  I'll let your team look at a current boot to determine the validity of the registers.

madscientist159, 2016-10-14:
At least on Firestone it looks like this commit broke both the SUART _and_ the VUART.  Investigation continues.

wghoffa, 2016-10-25:
@causten - I'm attempting to get a system setup in our lab so I can debug/resolve this. Do you have a recommendation on which system I should use?

adamliyi, 2016-11-02:
@wghoffa  , @madscientist159 and I observed that hostboot cannot boot on Firestone. So could you debug on a Firestone?

madscientist159, 2016-11-28:
@wghoffa Has there been any progress made on this issue?

madscientist159, 2016-11-30:
@causten , @wghoffa any word on a fix?

pragupta, 2018-08-17:
Changes may have already been merged.. But double check.

dcrowell77, 2018-08-17:
A change went up recently under de06d0f09c74f1bcf777716fb33db356b320dcf6 that is related to UART handling.  I'm going to assume this addresses any lingering issues unless told otherwise.

closed 2018-08-17 by dcrowell77