open-power/hostboot #54: Intermittent failures with duplicated enum elements in fapiHwpReturnCodes.H

deece, 2016-04-01:
I rarely get the following type of failures when compiling within Ozlab's CI:
05:06:47 In file included from ../../../../src/include/usr/hwpf/fapi/fapiReturnCode.H:78:0,
05:06:47 from ../../../../src/include/usr/hwpf/fapi/fapiUtil.H:63,
05:06:47 from fapiReturnCodeDataRef.C:47:
05:06:47 ../../../../obj/genfiles/fapiHwpReturnCodes.H:323:5: error: redefinition of 'RC_PROCPM_CHKSTOP'
05:06:47 RC_PROCPM_CHKSTOP = 0xd2d3e9,
05:06:47 ^
05:06:47 ../../../../obj/genfiles/fapiHwpReturnCodes.H:288:5: note: 'RC_PROCPM_CHKSTOP' previously defined here
05:06:47 RC_PROCPM_CHKSTOP = 0xd2d3e9,
05:06:47 ^
05:06:47 ../../../../obj/genfiles/fapiHwpReturnCodes.H:324:5: error: redefinition of 'RC_MSS_EFF_CONFIG_TERMINATION_INVALID_DIMM_RCD_IBT'
05:06:47 RC_MSS_EFF_CONFIG_TERMINATION_INVALID_DIMM_RCD_IBT = 0x0829ae,
05:06:47 ^
05:06:47 ../../../../obj/genfiles/fapiHwpReturnCodes.H:42:5: note: 'RC_MSS_EFF_CONFIG_TERMINATION_INVALID_DIMM_RCD_IBT' previously defined here
05:06:47 RC_MSS_EFF_CONFIG_TERMINATION_INVALID_DIMM_RCD_IBT = 0x0829ae,
....(many more)

I suspect that the code generation is triggering an issues in Perl (we are using Perl 5.20.2).

Sorting the output of keys() before emitting the enums seems to avoid triggering the bug, and also makes the generated output stable (the ordering of keys() is not guaranteed).
[console.txt.zip](https://github.com/open-power/hostboot/files/198900/console.txt.zip)
[fapiHwpReturnCodes.H.zip](https://github.com/open-power/hostboot/files/198902/fapiHwpReturnCodes.H.zip)
[fapiHwpReturnCodes.H-sorted.zip](https://github.com/open-power/hostboot/files/198901/fapiHwpReturnCodes.H-sorted.zip)

stewart-ibm, 2016-04-01:
This worries me... as there *should* be no difference to the resulting code (apart from things being sorted, which is arguably a good idea anyway).

Odds are, this could be a bug in perl on whatever distro you're running on....

Might be worth running the script in a tight loop for a while to see if you can trigger it and submit it as a bug to the distro too.

williamspatrick, 2016-04-01:
I suspect it could also be caused by a male file bug that is causing the script to be invoked twice at the same time. 

Patrick

deece, 2016-04-02:
I'm leaning more towards a perl bug, if 2 versions of the file were interspersed, we would see random syntax errors as statements & functions get broken in weird places. I've only ever seen duplicates in the enums.

jazurin, 2016-04-08:
We are hitting a very similar error in hostboot  open power CI.  This has been seen a few times when we build on our Ubuntu 14.10 ppc le machine, but not on our RHEL 6.4 pool machines.

In file included from ../../../src/include/usr/hwpf/fapi/fapiReturnCode.H:76:0,
                 from ../../../src/include/usr/hwpf/fapi/fapiUtil.H:63,
                 from ecmdDataBufferBase.C:41:
../../../obj/genfiles/fapiHwpErrorInfo.H:22161:5: error: redefinition of ‘RC_GET_PROC_ABUS_CUPLL_REFCLKSEL_OFFSET_BAD_CHIP_NAME_EC_CHIP_EC’
     RC_GET_PROC_ABUS_CUPLL_REFCLKSEL_OFFSET_BAD_CHIP_NAME_EC_CHIP_EC = 0x5ba426ef,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:21387:5: note: ‘RC_GET_PROC_ABUS_CUPLL_REFCLKSEL_OFFSET_BAD_CHIP_NAME_EC_CHIP_EC’ previously defined here
     RC_GET_PROC_ABUS_CUPLL_REFCLKSEL_OFFSET_BAD_CHIP_NAME_EC_CHIP_EC = 0x5ba426ef,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:22162:5: error: redefinition of ‘RC_MSS_DRAMINIT_RTT_WR_IMP_INPUT_ERROR_IMP’
     RC_MSS_DRAMINIT_RTT_WR_IMP_INPUT_ERROR_IMP = 0x7da6cf9c,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:22042:5: note: ‘RC_MSS_DRAMINIT_RTT_WR_IMP_INPUT_ERROR_IMP’ previously defined here
     RC_MSS_DRAMINIT_RTT_WR_IMP_INPUT_ERROR_IMP = 0x7da6cf9c,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:22164:5: error: redefinition of ‘RC_MSS_PDA_DRAM_DNE_DIMM_VALUE’
     RC_MSS_PDA_DRAM_DNE_DIMM_VALUE = 0xc6d11217,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:21481:5: note: ‘RC_MSS_PDA_DRAM_DNE_DIMM_VALUE’ previously defined here
     RC_MSS_PDA_DRAM_DNE_DIMM_VALUE = 0xc6d11217,
     ^

closed 2016-06-14 by dcrowell77