open-power/hostboot #56: Hostboot sends incomplete FRU data for CPU1

anoo1, 2016-04-22:
Seen in Barreleye, have not check other platforms.
Issue: Hostboot only sends the first chunk for VPD data for CPU1 (FRU_ID 2). Because the VPD data is incomplete, the data is not written to the inventory.

BMC Traces:
** CPU0 (FRU_ID 1), complete VPD data is sent in 2 chunks:
```
IPMI Incoming: Seq 0x0b, NetFn 0x0a, CMD: 0x12
0x000000: 01 00 00 01 00 00 01 00 00 00 fe 01 07 00 00 00 ................
0x000010: 00 c3 49 42 4d d0 50 52 4f 43 45 53 53 4f 52 20 ..IBM.PROCESSOR
0x000020: 4d 4f 44 55 4c 45 cc 59 41 33 39 33 32 36 34 37 MODULE.YA3932647
0x000030: 35 36 35 c7 30 30 55 4c 38 36 34 02             565.00UL864.
ipmi_storage_write_fru_data(): Start
IPMI WRITE-FRU-DATA for [/tmp/ipmifru01]  Offset = [0] Length = [57]
...
IPMI Incoming: Seq 0x0d, NetFn 0x0a, CMD: 0x12
0x000000: 01 39 00 20 20 00 00 c1 00 02                   .9.  .....
ipmi_storage_write_fru_data(): Start
IPMI WRITE-FRU-DATA for [/tmp/ipmifru01]  Offset = [57] Length = [7]
```

** CPU1 (FRU_ID 2) only receives the first chunk:
```
IPMI Incoming: Seq 0x0f, NetFn 0x0a, CMD: 0x12
0x000000: 02 00 00 01 00 00 01 00 00 00 fe 01 07 00 00 00 ................
0x000010: 00 c3 49 42 4d d0 50 52 4f 43 45 53 53 4f 52 20 ..IBM.PROCESSOR
0x000020: 4d 4f 44 55 4c 45 cc 59 41 33 39 33 32 36 34 37 MODULE.YA3932647
0x000030: 35 36 39 c7 30 30 55 4c 38 36 34 02             569.00UL864.
ipmi_storage_write_fru_data(): Start
IPMI WRITE-FRU-DATA for [/tmp/ipmifru02]  Offset = [0] Length = [57]
```

dcrowell77, 2016-04-23:
I just verified that 'ipmitool fru list' returns unique data on a random Firestone system running pretty recent code.

Looking at the data above it seems like the CPU1 data is actually the same as CPU0 (the serial numbers are the same).  My first suspect would be something in the xml data that is throwing HB off.  You should compare the final barreleye attribute xml with what we have from Firestone, maybe a FRU_ID or something is off somewhere.

williamspatrick, 2016-04-23:
The barreleye xml is also here on GitHub.

dcrowell77, 2016-04-25:
I was suggesting that someone needs to look at the generated xml, not the source.  There could be another issue related to how attributes flow from one to the other.

anoo1, 2016-04-25:
Looking at firestone.xml vs barreleye.xml, the CPU0 entry for barreleye has an extra field (INSTANCE_ID). Are these the output attribute xml files we're interested on?

From output/build/firestone-xml-15e371874a8dec1d8ef279aef2d0a0fed3ded8dd/firestone.xml
```
<globalSetting>
    <id>/sys-0/node-0/ponderosa-0/proc_socket-0/module-0/proc</id>
    <property>
    <id>IO_CONFIG_SELECT</id>
    <value></value>
    </property>
    <property>
    <id>FRU_NAME</id>
    <value>CPU0</value>
    </property>
    <property>
    <id>FRU_ID</id>
    <value>1</value>
    </property>
    <property>
    <id>IPMI_INSTANCE</id>
    <value>0</value>
    </property>
</globalSetting>

<globalSetting>
    <id>/sys-0/node-0/ponderosa-0/proc_socket-1/module-1/proc</id>
    <property>
    <id>IO_CONFIG_SELECT</id>
    <value></value>
    </property>
    <property>
    <id>FRU_NAME</id>
    <value>CPU1</value>
    </property>
    <property>
    <id>FRU_ID</id>
    <value>2</value>
    </property>
    <property>
    <id>IPMI_INSTANCE</id>
    <value>1</value>
    </property>
</globalSetting>
```

From output/build/barreleye-xml-81ac3ff3c4ccd16d2174f455c1cb17976daca948/barreleye.xml
```
<globalSetting>
    <id>/sys-0/node-0/motherboard-0/proc_socket-0/module-0/proc</id>
    <property>
    <id>FRU_ID</id>
    <value>1</value>
    </property>
    <property>
    <id>FRU_NAME</id>
    <value>CPU0</value>
    </property>
    <property>
    <id>INSTANCE_ID</id>
    <value>proc</value>
    </property>
    <property>
    <id>IO_CONFIG_SELECT</id>
    <value></value>
    </property>
    <property>
    <id>IPMI_INSTANCE</id>
    <value>0</value>
    </property>
</globalSetting>

<globalSetting>
    <id>/sys-0/node-0/motherboard-0/proc_socket-1/module-0/proc</id>
    <property>
    <id>FRU_ID</id>
    <value>2</value>
    </property>
    <property>
    <id>FRU_NAME</id>
    <value>CPU1</value>
    </property>
    <property>
    <id>IO_CONFIG_SELECT</id>
    <value></value>
    </property>
    <property>
    <id>IPMI_INSTANCE</id>
    <value>1</value>
    </property>
</globalSetting>
```

anoo1, 2016-04-25:
I just noticed also that barreleye has the same module-0 in the path for both CPUs, vs firestone has module-1 for CPU1:
```
<id>/sys-0/node-0/motherboard-0/proc_socket-X/module-0/proc</id>
```

dcrowell77, 2016-04-25:
The final output files that are used to generate the binary data used by Hostboot are in:
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/

Your last comment looks like it could be important...

anoo1, 2016-04-25:
From output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/*rpt files, yeah it appears the only difference is the path (Target).

firestone.rpt:
|                   Sensor Name |   FRU Name | Ent ID | Type |   ID | Inst |  FRU |       HUID | Target|
| ------------------------------ | ---------- | ------ | ---- | ---- | ---- | ---- | ---------- | ----------|
|                      CPU0_Temp |       CPU0 |  0x03  | 0x01 | 0x0B |    0 |    1 | 0x00050000 | /sys-0/node-0/ponderosa-0/proc_socket-0/module-0/proc|
|                      CPU0_Func |       CPU0 |  0x03  | 0x07 | 0x0C |    0 |    1 | 0x00050000 | /sys-0/node-0/ponderosa-0/proc_socket-0/module-0/proc|
|                      CPU1_Temp |       CPU1 |  0x03  | 0x01 | 0x0D |    1 |    2 | 0x00050001 | /sys-0/node-0/ponderosa-0/proc_socket-1/module-1/proc|
|                      CPU1_Func |       CPU1 |  0x03  | 0x07 | 0x0E |    1 |    2 | 0x00050001 | /sys-0/node-0/ponderosa-0/proc_socket-1/module-1/proc|

barreleye.rpt
|                   Sensor Name |   FRU Name | Ent ID | Type |   ID | Inst |  FRU |       HUID | Target|
| ------------------------------ | ---------- | ------ | ---- | ---- | ---- | ---- | ---------- | ----------|
|                      CPU0_Temp |       CPU0 |  0x03  | 0x01 | 0x0B |    0 |    1 | 0x00050000 | /sys-0/node-0/ponderosa-0/proc_socket-0/module-0/proc|
|                      CPU0_Func |       CPU0 |  0x03  | 0x07 | 0x0C |    0 |    1 | 0x00050000 | /sys-0/node-0/ponderosa-0/proc_socket-0/module-0/proc|
|                      CPU1_Temp |       CPU1 |  0x03  | 0x01 | 0x0D |    1 |    2 | 0x00050001 | /sys-0/node-0/ponderosa-0/proc_socket-1/module-1/proc|
|                      CPU1_Func |       CPU1 |  0x03  | 0x07 | 0x0E |    1 |    2 | 0x00050001 | /sys-0/node-0/ponderosa-0/proc_socket-1/module-1/proc|

johnhihi, 2016-04-26:
Hi, 
This is Foxconn BIOS team member.

I had tried to split module to two modules, module-0 for CPU0 and module-1 for CPU1.
Seems the DBUS still not showing the CPU1 inventory information.

Here is the xml I trying. [xml_commit_240c985@johnhihi](https://github.com/johnhihi/barreleye-xml/commit/240c985323ebd3941934236b5aa06111b3d510d2)

alvintpwang, 2016-04-27:
From the log, the serial numbers of CPU0 and CPU1 are different

CPU0 Serial number: YA3932647565
CPU1 Serial number:YA3932647569

johnhihi, 2016-04-27:
I add some debug code in ipmifruinv.C , which is in charge of collecting and sending fru data.

Below image shows the location of debug code. 
[image]

I found the code will be execute in ISTEP 6.5 and 11.1

P8 Traces:

ISTEP 6.5   CPU0
```
 26.08815|Print sendFruData....
 26.08815|01 00 00 01 00 00 00 FE 01 08 00 00 00 00 C3 49
 26.08815|42 4D D0 50 52 4F 43 45 53 53 4F 52 20 4D 4F 44
 26.08816|55 4C 45 CC 59 41 33 39 33 32 36 34 37 36 38 37
 26.08816|C7 30 30 55 4C 38 36 34 02 20 20 00 C5 45 43 3A
 26.08816|32 30 C1 00 00 00 00 13 
```

ISTEP 6.5   CPU1
```  
 26.09553|Print sendFruData....
 26.09554|01 00 00 01 00 00 00 FE 01 08 00 00 00 00 C3 49
 26.09554|42 4D D0 50 52 4F 43 45 53 53 4F 52 20 4D 4F 44
 26.09554|55 4C 45 CC 59 41 31 39 33 32 38 30 30 34 36 30
 26.09555|C7 30 30 55 4C 38 36 34 02 20 20 00 C5 45 43 3A
 26.09555|32 30 C1 00 00 00 00 29
```

ISTEP 11.1   CPU0
```
 40.37058|Print sendFruData....
 40.37058|01 00 00 01 00 00 00 FE 01 0D 00 00 00 00 C3 49
 40.37058|42 4D D0 50 52 4F 43 45 53 53 4F 52 20 4D 4F 44
 40.37059|55 4C 45 CC 59 41 33 39 33 32 36 34 37 36 38 37
 40.37059|C7 30 30 55 4C 38 36 34 02 20 20 00 E5 45 43 49
 40.37060|44 3A 30 31 39 36 31 38 33 31 34 32 30 30 34 35
 40.37060|39 36 30 42 31 30 30 30 30 30 30 30 34 35 32 43
 40.37060|38 37 C5 45 43 3A 32 30 00 C1 00 00 00 00 00 5F
```
ISTEP 11.1   CPU1
```
 40.38049|Print sendFruData....
 40.38050|01 00 00 01 00 00 00 FE 01 0D 00 00 00 00 C3 49
 40.38050|42 4D D0 50 52 4F 43 45 53 53 4F 52 20 4D 4F 44
 40.38051|55 4C 45 CC 59 41 31 39 33 32 38 30 30 34 36 30
 40.38051|C7 30 30 55 4C 38 36 34 02 20 20 00 E5 45 43 49
 40.38051|44 3A 30 31 38 41 30 30 32 36 43 30 31 30 32 37
 40.38052|31 38 30 34 30 34 30 30 30 30 30 30 44 45 33 38
 40.38052|37 30 C5 45 43 3A 32 30 00 C1 00 00 00 00 00 65
```

Seems the fru data sending from P8 is slightly different for CPU0 and CPU1, but not likely missing some data for CPU1.

alvintpwang, 2016-04-27:
Looks the fru data is good and I suspect the "write fru data" ipmi command failed. Please add the trace in https://github.com/open-power/hostboot/blob/master-p8/src/usr/ipmi/ipmifru.C#L157 to check.

anoo1, 2016-04-27:
Adding trace to ipmifru as suggested, and as suspected the ipmi command is failing. It appears in file /usr/sbin/Barreleye.py, under the "FRU_STR", this line is causing an issue:

        'CHASSIS_2'  : '<inventory_root>/system/chassis',

The CPU1 FRU ID is 2, and the VPD data has a section for CHASSIS, so it's trying to read VPD data but hostboot just sends the BOARD info, not CHASSIS.

Changing the number 2 on that line to something like 200 seems to fix the issue and CPU1 data appears now in the inventory: 

        'CHASSIS_200'  : '<inventory_root>/system/chassis',

I'll find out more where this definition comes from and how to permanently change it to a correct value in Barreleye.py.

anoo1, 2016-04-28:
Fixed by removing the CHASSIS_2 line with pull request https://github.com/openbmc/skeleton/pull/68 .

closed 2016-04-28 by dcrowell77