open-power/hostboot #57: Hostboot not respecting BT capabilities

stewart-ibm, 2016-05-20:
According to @rjknight , Hostboot has hardcoded values for BT capabilities  rather than interrogating the BMC and acting accordingly within those capabilities.

williamspatrick, 2016-05-20:
The capability that is hard-coded is the number of outstanding (parallel) IPMI commands. 

The origin of this is buggy AMI code that lies to us. This code is "in the wild" and it is impossible for us to identify truthful and untruthful BMCs.

The side effect of this is a [arguably] slightly slower boot. Considering the risk of breaking machines with existing untruthful BMC code, I do not see the benefit of removing this for P8 machines.

Can you give a rationale on why this should be fixed in the near term?

stewart-ibm, 2016-05-23:
We hardcode the outstanding requests for the same reason in skiboot (and as a bonus, it makes the code much simpler).

If HB takes the queue sizes from the capabilities, then we should likely just close this issue.

dcrowell77, 2016-05-23:
If we got really motivated we could add a config flag to either ignore or believe the call so that partners could theoretically improve things if they know their BMC is trustworthy, but this doesn't seem worth the effort at this time.

closed 2016-05-23 by dcrowell77