open-power/hostboot #58: BIOS Hang in a loop on Barreleye

agangidi53, 2016-06-01:
I've seen couple of Barreleye servers running that failed the boot up process as they were stuck in loop. 
​
Hconsole logs show that they go back to the start of loop when they hit Step 14. And this keeps repeating on and on.

In general is there a guide to decoding what the step info means ? Full log is attached.
[BIOS hang.pdf](https://github.com/open-power/hostboot/files/293946/BIOS.hang.pdf)

22.35508|ISTEP 13.11
 22.46796|ISTEP 13.12
 22.46877|ISTEP 14. 1
 22.54008|ISTEP 14. 2
 22.56077|ISTEP 14. 3
  2.10687|ERRL|Dumping errors reported prior to registration
  3.75598|ISTEP  6. 3
  4.09279|ISTEP  6. 4
  4.09330|ISTEP  6. 5
  8.99220|HWAS|PRESENT> DIMM[03]=A0A0A0A0A0A0A0A0
  8.99221|HWAS|PRESENT> Membuf[04]=CCCC000000000000
  8.99221|HWAS|PRESENT> Proc[05]=C000000000000000

dcrowell77, 2016-06-01:
I see no indications from Hostboot that we are requesting a reboot so it must be coming from the BMC directly.  Do you have SELs?  Any logs from the BMC that show what is going on?

agangidi53, 2016-06-01:
BMC event logs (Barreleye runs openBMC) are here:
Don't see anything fishy there

agangidi53, 2016-06-03:
Forgot to add these 
[BIOS_hang_loop_event_logs_BMC.pdf](https://github.com/open-power/hostboot/files/298043/BIOS_hang_loop_event_logs_BMC.pdf)

Kenthliu, 2016-06-06:
@agangidi53 The SEL is just show the HDD present status. There is no message that BMC send reboot request. Can you please give me the journalctl log to clarify this issue? I did not see the symptom before. Thanks.

agangidi53, 2016-06-13:
hi @Kenthliu 

Here is the journal log. It looks like a watch dog error to me.

Apr 27 10:04:09 barreleye system_manager.py[943]: Turn on power and boot
Apr 27 10:04:09 barreleye system_manager.py[943]: Running System State: HOST_POWERING_ON
Apr 27 10:04:09 barreleye system_manager.py[943]: Setting watchdog poll interval to: 30000
Apr 27 10:04:09 barreleye system_manager.py[943]: Starting watchdog with poll interval: 30000
Apr 27 10:04:12 barreleye system_manager.py[943]: Running System State: HOST_POWERED_ON
Apr 27 10:04:12 barreleye system_manager.py[943]: Starting process: /usr/sbin/pcie_slot_present.exe: pcie_present
Apr 27 10:04:12 barreleye system_manager.py[943]: Booting host
Apr 27 10:04:13 barreleye system_manager.py[943]: Checking Presence: SLOT0_RISER_PRESENT
Apr 27 10:04:13 barreleye system_manager.py[943]: GPIO Lookup:  SLOT0_RISER_PRESENT = 424,in
Apr 27 10:04:13 barreleye system_manager.py[943]: Checking Presence: SLOT1_RISER_PRESENT
Apr 27 10:04:13 barreleye system_manager.py[943]: GPIO Lookup:  SLOT1_RISER_PRESENT = 425,in
Apr 27 10:04:13 barreleye system_manager.py[943]: Checking Presence: SLOT2_RISER_PRESENT
Apr 27 10:04:13 barreleye system_manager.py[943]: GPIO Lookup:  SLOT2_RISER_PRESENT = 426,in
Apr 27 10:04:14 barreleye system_manager.py[943]: Checking Presence: SLOT0_PRESENT
Apr 27 10:04:14 barreleye system_manager.py[943]: GPIO Lookup:  SLOT0_PRESENT = 427,in
Apr 27 10:04:14 barreleye system_manager.py[943]: Checking Presence: SLOT1_PRESENT
Apr 27 10:04:14 barreleye system_manager.py[943]: Using primary side of the bios flash
Apr 27 10:04:14 barreleye system_manager.py[943]: GPIO Lookup:  SLOT1_PRESENT = 428,in
Apr 27 10:04:14 barreleye system_manager.py[943]: Checking Presence: SLOT2_PRESENT
Apr 27 10:04:14 barreleye system_manager.py[943]: GPIO Lookup:  SLOT2_PRESENT = 429,in
Apr 27 10:04:14 barreleye system_manager.py[943]: Checking Presence: MEZZ0_PRESENT
Apr 27 10:04:14 barreleye system_manager.py[943]: Running System State: HOST_BOOTING
Apr 27 10:04:14 barreleye system_manager.py[943]: GPIO Lookup:  MEZZ0_PRESENT = 432,in
Apr 27 10:04:15 barreleye system_manager.py[943]: Checking Presence: MEZZ1_PRESENT
Apr 27 10:04:15 barreleye system_manager.py[943]: GPIO Lookup:  MEZZ1_PRESENT = 433,in
Apr 27 10:04:35 barreleye systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
Apr 27 10:04:39 barreleye system_manager.py[943]: Polling watchdog: 0
Apr 27 10:04:39 barreleye system_manager.py[943]: Watchdog Error, Hard Rebooting
Apr 27 10:04:39 barreleye system_manager.py[943]: Turn off power
Apr 27 10:04:39 barreleye system_manager.py[943]: Stopping watchdog
Apr 27 10:04:39 barreleye system_manager.py[943]: Running System State: HOST_POWERING_OFF
Apr 27 10:04:39 barreleye system_manager.py[943]: Deleting OCC device
Apr 27 10:04:39 barreleye system_manager.py[943]: Running System State: HOST_POWERED_OFF
Apr 27 10:04:39 barreleye system_manager.py[943]: Turn on power and boot
Apr 27 10:04:39 barreleye system_manager.py[943]: Deleting OCC device
Apr 27 10:04:39 barreleye system_manager.py[943]: Running System State: HOST_POWERING_ON
Apr 27 10:04:39 barreleye system_manager.py[943]: Setting watchdog poll interval to: 30000
Apr 27 10:04:39 barreleye system_manager.py[943]: Starting watchdog with poll interval: 30000
Apr 27 10:04:42 barreleye system_manager.py[943]: Running System State: HOST_POWERED_ON
Apr 27 10:04:42 barreleye system_manager.py[943]: Starting process: /usr/sbin/pcie_slot_present.exe: pcie_present
Apr 27 10:04:42 barreleye system_manager.py[943]: Booting host
Apr 27 10:04:43 barreleye system_manager.py[943]: Checking Presence: SLOT0_RISER_PRESENT
Apr 27 10:04:43 barreleye system_manager.py[943]: GPIO Lookup:  SLOT0_RISER_PRESENT = 424,in
Apr 27 10:04:43 barreleye system_manager.py[943]: Checking Presence: SLOT1_RISER_PRESENT
Apr 27 10:04:43 barreleye system_manager.py[943]: GPIO Lookup:  SLOT1_RISER_PRESENT = 425,in
Apr 27 10:04:43 barreleye system_manager.py[943]: Checking Presence: SLOT2_RISER_PRESENT
Apr 27 10:04:43 barreleye system_manager.py[943]: GPIO Lookup:  SLOT2_RISER_PRESENT = 426,in
Apr 27 10:04:44 barreleye system_manager.py[943]: Checking Presence: SLOT0_PRESENT
Apr 27 10:04:44 barreleye system_manager.py[943]: GPIO Lookup:  SLOT0_PRESENT = 427,in
Apr 27 10:04:44 barreleye system_manager.py[943]: Checking Presence: SLOT1_PRESENT
Apr 27 10:04:44 barreleye system_manager.py[943]: GPIO Lookup:  SLOT1_PRESENT = 428,in
Apr 27 10:04:44 barreleye system_manager.py[943]: Using primary side of the bios flash
Apr 27 10:04:44 barreleye system_manager.py[943]: Checking Presence: SLOT2_PRESENT
Apr 27 10:04:44 barreleye system_manager.py[943]: GPIO Lookup:  SLOT2_PRESENT = 429,in
Apr 27 10:04:44 barreleye system_manager.py[943]: Running System State: HOST_BOOTING
Apr 27 10:04:44 barreleye system_manager.py[943]: Checking Presence: MEZZ0_PRESENT
Apr 27 10:04:45 barreleye system_manager.py[943]: GPIO Lookup:  MEZZ0_PRESENT = 432,in
Apr 27 10:04:45 barreleye system_manager.py[943]: Checking Presence: MEZZ1_PRESENT
Apr 27 10:04:45 barreleye system_manager.py[943]: GPIO Lookup:  MEZZ1_PRESENT = 433,in
Apr 27 10:05:05 barreleye systemd[1]: systemd-hwdb-update.service: Cannot add dependency job, ignoring: Unit systemd-hwdb-update.service is masked.
Apr 27 10:05:05 barreleye systemd[1]: Started SSH Per-Connection Server (10.127.90.74:47104).
[... cut ...]
Apr 27 10:05:05 barreleye dropbear[14966]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Apr 27 10:05:05 barreleye dropbear[14966]: Child connection from 10.127.90.74:47104
Apr 27 10:05:05 barreleye dropbear[14966]: Exit before auth: Exited normally
Apr 27 10:05:09 barreleye system_manager.py[943]: Polling watchdog: 0
Apr 27 10:05:10 barreleye system_manager.py[943]: Watchdog Error, Hard Rebooting
Apr 27 10:05:10 barreleye system_manager.py[943]: Turn off power
Apr 27 10:05:10 barreleye system_manager.py[943]: Stopping watchdog

Kenthliu, 2016-06-14:
OK it seems like the BMC waiting pgood and watchdog timeout. Our EE release a new CPLD to fixed this. Can you please help to confirm the CPLD version? Is that CPLD 0.2.1_0429? Thanks.

agangidi53, 2016-07-05:
@Kenthliu  CPLD Version according to i2c commands via BMC is 011

williamspatrick, 2016-07-20:
Please re-open if this is still an issue.  Sounds like it was a CPLD code level that fixed this?

closed 2016-07-20 by williamspatrick

agangidi, 2016-07-20:
Yes. This doesn't happen with new CPLD version