open-power/hostboot #59: System Halt during Barreleye bootup process

agangidi53, 2016-06-01:
We have Barreleye server for which:

a) Host bootup process hangs and then restarts around 50 seconds or so after we power on. 
b) After restarting (by itself in step above) bootup process  gets further but finally hangs around where it's trying to pick up the kernel:
[8082470429,5] INIT: Starting kernel at 0x20010000, fdt at 0x305efb18 (size 0x1b72d)
[host hangs loading skiboot kernel.pdf](https://github.com/open-power/hostboot/files/294495/host.hangs.loading.skiboot.kernel.pdf)

Attaching the log, that should give you an idea so as to what is happening.

dcrowell77, 2016-06-01:
Where is the top of the console (the first 50 seconds)?  We should always see a message like this when we start.
> 1.94888|ERRL|Dumping  errors  reported  prior  to  registraNon

The error log that is displayed (e540) indicates that we analyzed a system checkstop, that should be investigated.  That would have deconfigured some piece(s) of hardware.

There is then a secondary issue of why the kernel hung later on.  This could be related to the deconfigured hardware or it could be another symptom of the same system problem that triggered the checkstop originally.  What happens after this point?  Does the system reboot again?

agangidi53, 2016-06-01:
I'll gather initial part of bootup logs and post here . 

Finally the system powers off after displaying kernel hang message.

deathscytheh28, 2016-06-06:
UserData1 unknown : 0x**00050001**00000101 indicates the HUID, and according to the HUID and AFFINITY_PATH mapping, it indicates affinity:sys-0/node-0/proc-1. 

Maybe CPU1 got some problems, could you try to remove CPU1 and power on again to see if your system can boot successfully?

stewart-ibm, 2016-06-10:
Hrm... could you dump out the BOOTKERNEL pnor partition and then compare it to what it's meant to be.

One possible cause of boot stopping at "Starting Kernel" is a corrupted BOOTKERNEL partition.

agangidi53, 2016-06-13:
@deathscytheh28 @dcrowell77 

Here are the first 50 seconds of the log.

  1.94873|ERRL|Dumping errors reported prior to registration
  3.88464|ISTEP  6. 3
  4.33553|ISTEP  6. 4
  4.33634|ISTEP  6. 5
 30.67827|HWAS|PRESENT> DIMM[03]=A0A0A0A0A0A0A0A0
 30.67827|HWAS|PRESENT> Membuf[04]=CCCC000000000000
 30.67828|HWAS|PRESENT> Proc[05]=C000000000000000
 54.58987|ISTEP  6. 6
 55.50168|================================================
 55.50169|Error reported by unknown (0xE500)
 55.50169|  <none>
 55.50169|  ModuleId   0x0b unknown
 55.50169|  ReasonCode 0xe540 unknown
 55.50169|  UserData1  unknown : 0x0005000000000101
 55.50170|  UserData2  unknown : 0x4241000500000000

agangidi53, 2016-06-13:
hi @stewart-ibm   preliminarily the BOOTKERNEL looks different than the one running the same version of BMC but not running into this issue. But that said, I don't know what's different in them, yet. I'll check.

dcrowell77, 2016-06-13:
The top of the console looks like I expected it to, this is definitely the result of analyzing a system checkstop.

agangidi53, 2016-06-13:
[journal_logs.txt](https://github.com/open-power/hostboot/files/312849/journal_logs.txt)

Here are corresponding journal logs.

stewart-ibm, 2016-06-14:
@dcrowell77 if we have a corrupted BOOTKERNEL, I'd suspect that a checkstop is quite likely - random data executed as PPC instructions generally doesn't go too well :)

@agangidi53 - "pflash -r bootkernel.dump -P BOOTKERNEL -F pnor_dump" will extract into a file named bootkernel.dump the BOOTKERNEL partition  from the file pnor_dump which is a full dump of the host pnor. From there, you can run the same thing on the expected PNOR content and use diff or similar

agangidi53, 2016-06-14:
@stewart-ibm  Thanks for the info.  I was diffing bin files. What you suggested makes more sense.

stewart-ibm, 2016-06-14:
Yeah, if you diff the whole pnor image you're diffing VPD cache et al and they may be different on different systems (or, conceivably, different boots)

Kenthliu, 2016-06-20:
I saw "Emergency Shutdown!" in the journalctl log. It may cause by CPU or DIMM over warning threshold. I think the environment temperature should be fine. I guess the DIMM has some problem. The log looks similar that I reported before. https://github.com/openbmc/skeleton/issues/71

pragupta, 2018-08-17:
No activity in 2 years..seems to be a specific problem.. closing.

closed 2018-08-17 by pragupta