open-power/hostboot #61: Doesn't build with GCC5

stewart-ibm, 2016-06-22:
I'm seeing build failures on GCC5.3 when building hostboot, several unused variable warnings. This seems to be because of how Hostboot code defines some constants.

The below patch is a quick stab at solving it, but this just gets hostboot to compile, booting does **not** work (checkstop) - this is probably due to me being very rusty on C++ linkage for constants and not entirely sure how hostboot links itself at runtime that may cause problems.

<pre>
--- a/src/include/usr/targeting/attrPlatOverride.H
+++ b/src/include/usr/targeting/attrPlatOverride.H
@@ -52,13 +52,7 @@ struct AttrOverrideSection
  *        enums and make it obvious which layers map to what PNOR section.
  *        Currrently the pair is only used in a test case to keep it in order
  */
-const std::pair<AttributeTank::TankLayer, PNOR::SectionId>
-    tankLayerToPnor[AttributeTank::TANK_LAYER_LAST] =
-    {
-        std::make_pair(AttributeTank::TANK_LAYER_FAPI, PNOR::ATTR_TMP),
-        std::make_pair(AttributeTank::TANK_LAYER_TARG, PNOR::ATTR_TMP),
-        std::make_pair(AttributeTank::TANK_LAYER_PERM, PNOR::ATTR_PERM)
-    };
+extern const std::pair<AttributeTank::TankLayer, PNOR::SectionId> tankLayerToPn
or[AttributeTank::TANK_LAYER_LAST];
  
 /**
  * @brief This function gets any Attribute Overrides in PNOR
diff --git a/src/include/usr/targeting/common/targetservice.H b/src/include/usr/
targeting/common/targetservice.H
index 828fe86f3e51..8603f5174743 100644
--- a/src/include/usr/targeting/common/targetservice.H
+++ b/src/include/usr/targeting/common/targetservice.H
@@ -134,10 +134,7 @@ namespace TARGETING
  *      DD framework to bring PNOR device driver online.  Note this target
  *      cannot be used as input to any target service APIs.
  */
-static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
-    = (sizeof(void*) == 4) ?
-        reinterpret_cast<TARGETING::Target* const>(0xFFFFFFFF)
-      : reinterpret_cast<TARGETING::Target* const>(0xFFFFFFFFFFFFFFFFULL);
+extern Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL;
 
 /**
  *  @brief TargetService class
diff --git a/src/usr/i2c/i2c.C b/src/usr/i2c/i2c.C
index 2afd7730c4ba..0261dda6db9b 100755
--- a/src/usr/i2c/i2c.C
+++ b/src/usr/i2c/i2c.C
@@ -89,6 +89,7 @@ const TARGETING::ATTR_I2C_BUS_SPEED_ARRAY_type g_var = {{NULL}};
 
 namespace I2C
 {
+uint64_t g_I2C_NEST_FREQ_MHZ = i2cGetNestFreq();
 
 // Register the generic I2C perform Op with the routing code for Procs.
 DEVICE_REGISTER_ROUTE( DeviceFW::WILDCARD,
diff --git a/src/usr/i2c/i2c.H b/src/usr/i2c/i2c.H
index 05f11d5514d6..0b15c5a096d7 100755
--- a/src/usr/i2c/i2c.H
+++ b/src/usr/i2c/i2c.H
@@ -88,7 +88,7 @@ ALWAYS_INLINE inline uint64_t i2cGetNestFreq()
 
   return sysTarget->getAttr<TARGETING::ATTR_NEST_FREQ_MHZ>();
 };
-static uint64_t g_I2C_NEST_FREQ_MHZ = i2cGetNestFreq();
+extern uint64_t g_I2C_NEST_FREQ_MHZ;
 
 
 /**
diff --git a/src/usr/targeting/attrPlatOverride.C b/src/usr/targeting/attrPlatOverride.C
index 9368c4af1378..4edac1fa339a 100644
--- a/src/usr/targeting/attrPlatOverride.C
+++ b/src/usr/targeting/attrPlatOverride.C
@@ -30,6 +30,13 @@
 
 namespace TARGETING
 {
+const std::pair<AttributeTank::TankLayer, PNOR::SectionId>
+    tankLayerToPnor[AttributeTank::TANK_LAYER_LAST] =
+    {
+        std::make_pair(AttributeTank::TANK_LAYER_FAPI, PNOR::ATTR_TMP),
+        std::make_pair(AttributeTank::TANK_LAYER_TARG, PNOR::ATTR_TMP),
+        std::make_pair(AttributeTank::TANK_LAYER_PERM, PNOR::ATTR_PERM)
+    };
 
 errlHndl_t getAttrOverrides(PNOR::SectionInfo_t &i_sectionInfo,
                       AttributeTank* io_tanks[AttributeTank::TANK_LAYER_LAST])
diff --git a/src/usr/targeting/common/targetservice.C b/src/usr/targeting/common/targetservice.C
index cd3870dedced..2822b47080e3 100644
--- a/src/usr/targeting/common/targetservice.C
+++ b/src/usr/targeting/common/targetservice.C
@@ -61,7 +61,7 @@
 
 namespace TARGETING
 {
-
+Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL = reinterpret_cast<TARGETING::Target* const>(~0);
 #define TARG_NAMESPACE "TARGETING::"
 
 #define TARG_CLASS "targetService"

</pre>

dcrowell77, 2016-06-22:
Can paste in the compile errors that you saw as well?  We compile with unused vars as fails so it is a little weird that a different GCC exposes new fails.

stewart-ibm, 2016-06-22:
On (maybe 1 or 2 revisions behind what's in master-next):
<pre>
In file included from ../../../src/include/usr/targeting/common/predicates/predicateisfunctional.H:58:0,
                 from ../../../src/include/usr/targeting/common/predicates/predicates.H:34,
                 from ../../../src/include/usr/targeting/common/commontargeting.H:39,
                 from compdesc.C:31:
../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from ../../../src/include/usr/devicefw/userif.H:41:0,
                 from userif.C:27:
../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from ../../../src/include/usr/devicefw/userif.H:41:0,
                 from ../../../src/include/usr/devicefw/driverif.H:37,
                 from driverif.C:26:
../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from ./misctest.H:28:0,
                 from ../../../../obj/modules/testkernel/testkernel.C:18:
../../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors
../../../../src/build/mkrules/cc.rules.mk:33: recipe for target '../../../../obj/modules/testkernel/testkernel.o' failed

In file included from ../../../src/include/usr/devicefw/userif.H:41:0,
                 from ../../../src/include/usr/devicefw/driverif.H:37,
                 from scomtrans.C:38:
../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from ../../../../src/include/usr/targeting/common/predicates/predicateisfunctional.H:58:0,
                 from ../../../../src/include/usr/targeting/common/predicates/predicates.H:34,
                 from ../../../../src/include/usr/targeting/common/commontargeting.H:39,
                 from ../../../../src/include/usr/initservice/initserviceif.H:31,
                 from ../baseinitsvc/initservice.H:53,
                 from extinitsvc.H:47,
                 from extinitsvc.C:44:
../../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from ../../../../src/include/usr/devicefw/userif.H:41:0,
                 from ./xscomtest.H:35,
                 from ../../../../obj/modules/testxscom/testxscom.C:16:
../../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from ../../../../src/include/usr/targeting/common/predicates/predicateisfunctional.H:58:0,
                 from ../../../../src/include/usr/targeting/common/predicates/predicates.H:34,
                 from ../../../../src/include/usr/targeting/common/commontargeting.H:39,
                 from ../../../../src/include/usr/initservice/initserviceif.H:31,
                 from ./../baseinitsvc/initservice.H:53,
                 from ./initservicetest.H:40,
                 from ../../../../obj/modules/testinitservice/testinitservice.C:17:
../../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
[... 174 lines cut ...]
                 from ../../../../src/include/usr/targeting/common/predicates/predicates.H:34,
                 from ../../../../src/include/usr/targeting/common/commontargeting.H:39,
                 from ../../../../src/include/usr/initservice/initserviceif.H:31,
                 from fapiPlatUtil.C:42:
../../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from ../../../../src/include/usr/devicefw/userif.H:41:0,
                 from fapiPlatMvpdAccess.C:42:
[... cut ...]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

In file included from fapiPlatAttributeService.C:38:0:
../../../../src/include/usr/targeting/common/targetservice.H:137:22: error: ‘TARGETING::MASTER_PROCESSOR_CHIP_TARGET_SENTINEL’ defined but not used [-Werror=unused-variable]
 static Target* const MASTER_PROCESSOR_CHIP_TARGET_SENTINEL
                      ^
cc1plus: all warnings being treated as errors

</pre>

(and there's a few more of those specific ones).

shenki, 2016-06-23:
See https://github.com/open-power/hostboot/issues/38 for a different method of fixing this.

williamspatrick, 2016-06-23:
We could add Joel's warning removal to the op-build recipe for the time being. 

Stewart, you mentioned that Hostboot "checkstoped" with your changes. Is this due to your changes or due to something else with GCC5?

stewart-ibm, 2016-06-24:
@williamspatrick It was pretty early in boot (I didn't even get any SOL output) and didn't have time to hook up any debugging tools (FSP2 or pdebug) - so I'm completely unsure as to what it was checkstopping on... it's just what the BMC said and I thought I'd file this before running off to deal with the other fire..

stewart-ibm, 2016-07-05:
@williamspatrick I think the checkstop was actually due to some oddity in how I was building HB... it seems that HOSTBOOT_SITE_METHOD=local gives you a not bootable system, but HOSTBOOT_SITE_METHOD=git and cloning from my repo works....

stewart-ibm, 2016-08-26:
This pull request https://github.com/open-power/hostboot/pull/62 fixes master-p8 up to build and boot with GCC6.

closed 2016-12-26 by williamspatrick