open-power/hostboot #68: IPL failure with corrupt HBEL partition

stewart-ibm, 2016-10-21:
(from https://github.com/open-power/hostboot/issues/67 )

Erase the HBEL partition, but *not* setting ECC:

```pflash -P HBEL -e```

and then, when trying to IPL, you end up failing:
```
  0.61819|ECC error in PNOR flash in section offset 0x00008000

  0.62322|System shutting down with error status 0x60F
  3.22583|Ignoring boot flags, incorrect version 0x0
  3.30757|ISTEP  6. 3
  1.09243|ECC error in PNOR flash in section offset 0x00008000

  1.09246|System shutting down with error status 0x60F
  3.68140|Ignoring boot flags, incorrect version 0x0
  3.76426|ISTEP  6. 3
```

dcrowell77, 2016-10-21:
Yes, that is expected behavior.  The ECC algorithm in P8 means that an erased (all FF) partition is invalid ECC.  You must explicitly populate it with an ECC good 'blank' image.  Because the HBEL partition is vital to our operation we shut down if it corrupted.

closed 2016-10-21 by dcrowell77

stewart-ibm, 2016-10-21:
but should it fail rather than, say, erase the HBEL partition with valid ECC data and then continue? Is there anything that *must* be stored there in order to boot (or at least attempt to boot)?

dcrowell77, 2016-10-21:
The data in HBEL may contain error logs that we were unable to send to the BMC.  Just erasing that loses data that could potentially be vital to doing FA on a system failure.  I do understand that there are some RAS requirement differences that need to be resolved on how to handle things like this, but this is the intended design as of right now.

Secondarily, the way we have the PNOR mapped into our VMM makes recovery for some of the partitions non-trivial so we'd need to look into it more to see how painful a change would even be.

leftyfb, 2017-02-06:
How cab I resolve this on a Power S822LC where it gets stuck on:

38.32244|ECC error in PNOR flash in section offset 0x01E73200
38.32249|System shutting down with error status 0x60F
38.32265|System shutting down with error status 0x90000021

This is only via serial console. I get nothing over the java remote console or ipmi. I can login to the web management interface but don't see any place to erase or reset the HBEL partition.

stewart-ibm, 2017-02-06:
leftyfb <notifications@github.com> writes:

> How cab I resolve this on a Power S822LC where it gets stuck on:
>
> 38.32244|ECC error in PNOR flash in section offset 0x01E73200
> 38.32249|System shutting down with error status 0x60F
> 38.32265|System [...]

There's no UI for it unfortunately.

The easiest way to remedy this situation is to re-flash your firmware
which you can do either over IPMI or from the web UI to the BMC.

If I'm looking at the right thing, your error is actually in the FIRDATA
partition.

I'd certainly say that an ECC error there should *not* be *that* fatal,
and that it's a bug. Now, if that's something that Hostboot can address
in POWER8 releases is another question :)

-- 
Stewart Smith
OPAL Architect, IBM.

leftyfb, 2017-02-07:
Thank you @stewart-ibm. I reflashed the firmware(v1.11_2.1)  and am now getting past the error.