open-power/hostboot #70: Hostboot silently fails if STB Container header sw_key_count==0

stewart-ibm, 2016-11-17:
Construct a secure/trusted boot header with sw_key_count of 0, so the ContainerHeader::validate() fails, and hostboot will just hang at istep 20.1 for a while and then reboot (in an infinite loop). No error displayed, no error logs, nothing.

nasastry, 2016-11-17:
@stewart-ibm 
How to "Construct a secure/trusted boot header with sw_key_count of 0"?
Thanks in advance.

stewart-ibm, 2016-11-17:
@nasastry https://patchwork.ozlabs.org/patch/695995/

A separate bug is that with trusted boot disabled, hostboot still requires a container.

stewart-ibm, 2016-11-17:
It's likely more than just this, I had to hack on things a bit to get things going.
FWIW, this worked: https://patchwork.ozlabs.org/patch/696025/

It seems that the Hostboot code to parse the STB container isn't too informative when parsing fails.

Considering this is *before* we validate the content, the result of an attack could be zero errors but instead a reboot loop, which isn't too informative.

dcrowell77, 2016-11-17:
@bofferdn - Take a look

bofferdn, 2016-11-18:
Going on vacation, might have to have Steve Cprek comment while I'm gone

Some background
- Which release/defconfig is being used?
- Secure mode enabled or disabled?

Per the prefix documentation Tim Block provided, it's mandatory to have SW key count of 0 at minimum when building a container.  The ROM code burned into P8 has a special branch below which it executes if it sees 0 SW key count, and is not really a supported path:

 if(!prefix->sw_key_count) {
        // finish processing special prefix header
        // test for machine specfic (sanity check)
        if(prefix->ecid_count == 0)
          FAILED(SPECIAL_NO_ECID_TEST,"special prefix with ecid_count == 0");
        // test for signing of keys only (sanity check)
        if(size != 0)
          FAILED(SPECIAL_SIZE_0_TEST,"special prefix with payload_size != 0");
        // return hrmor relative code start address
        params->entry_point = GET64(prefix->code_start_offset);
        //check if the entry is HRMOR-relative and aligned
        if(params->entry_point & ~(ENTRY_MASK))
          FAILED(ENTRY_VALID_TEST,"entry is not HRMOR relative or not aligned");
        params->log=CONTEXT|COMPLETED;
        return ROM_DONE;
      }

At initial glance I didn't find a path (other than testcase) where the ContainerHeader::validate() actually leads to any sort of failure on its own; perhaps there is some other downstream assumption that later fails, but we'd need to recreate that with better traces, since it sounds like you're not seeing much of anything.

scprek, 2016-11-18:
Don't you mean a min SW key count of 1? 1 is required.

scprek, 2016-11-18:
I can add an assert in the validate() function, but ContainerHeader::validate() failing shouldn't cause a hang, as it doesn't do anything but set an internal variable of the class. So either the parsing of it header itself is causing the issue, sw key count 0 would mess up reading the header as 1 sw key is required and we get a memory exception of some sort? Or there's an error after parsing it.

scprek, 2016-11-18:
Looks like ContainerHeader is used before verification in 2 places
1. `SPnorRP::verifySections`
```c++        
SECUREBOOT::ContainerHeader l_conHdr(l_tempAddr);
size_t l_totalContainerSize = l_conHdr.totalContainerSize();
```
2. PnorRP::getSectionInfo
```c++
// Do an existence check on the container to see if it's non-empty
// and has valid beginning bytes. For optional Secure PNOR sections.
if (PNOR::cmpSecurebootMagicNumber(l_vaddr))
{
    SECUREBOOT::ContainerHeader l_conHdr(l_vaddr);
    payloadTextSize = l_conHdr.payloadTextSize();
    assert(payloadTextSize > 0,"Non-zero payload text size expected.");
}
```

If `CONFIG_SECUREBOOT` is set this operations will run regardless if in securemode or not, so a proper secureboot header is expected.

scprek, 2016-11-18:
This patch will assert and print a console message if the weak ContainerHeader validate() function finds an error https://github.com/bofferdn/op-build/pull/4

stewart-ibm, 2016-11-20:
@scprek that patch would have certainly helped :)

@scprek where is the format documented? All I've seen is several sets of header files, all differing slightly and not standing alone as a definition of what is/is not valid.

The ultimate documenation seems to be the ROM, but its implementation is likely different to that in Hostboot and skiboot.

stewart-ibm, 2016-11-20:
Oh, I was using a SuperMicro build with Trusted Boot built, but without a TPM.

bofferdn, 2018-05-23:
Can this be closed?  Steve's fix is in.  Tim Block had the original documentation which described the restrictions.

stewart-ibm, 2018-05-23:
Yeah, I'll close it as I've seen much better errors spewing out now. If I can reproduce something with the same symptoms, I'll re-open.

closed 2018-05-23 by stewart-ibm