open-power/hostboot #73: Duplicate i2c-bus@2 nodes in device tree

stewart-ibm, 2016-11-24:
On habanero (with TPM): hostboot-09cfacb-opdirty
On Stratton (with tpm): hostboot-7fdfb37

skiboot throws the following error on boot:
```
[   47.442378943,3] DT: dt_attach_root failed, duplicate i2c-bus@2
```

Which is because of a duplicate entry in the device tree, specifically for  `ibm,port-name = "p8_00000000_e1p2";` - which is where the TPM is behind.

If I hack skiboot with a patch to add the node anyway, but under a different name, I see these two:

```
                        DUPLICATE-i2c-bus@2 {
                                reg = <0x2>;
                                bus-frequency = <0x61a80>;
                                compatible = "ibm,power8-i2c-port", "ibm,opal-i2
c";
                                ibm,opal-id = <0x3>;
                                ibm,port-name = "p8_00000000_e1p2";
                                #address-cells = <0x1>;
                                phandle = <0x10000014>;
                                #size-cells = <0x0>;

                                eeprom@50 {
                                        reg = <0x50>;
                                        label = "system-vpd";
                                        compatible = "atmel,24c128";
                                        phandle = <0x10000015>;
                                };
                        };

                        i2c-bus@2 {
                                reg = <0x2>;
                                bus-frequency = <0x61a80>;
                                compatible = "ibm,power8-i2c-port", "ibm,opal-i2
c";
                                ibm,opal-id = <0x4>;
                                ibm,port-name = "p8_00000000_e1p2";
                                #address-cells = <0x1>;
                                phandle = <0x10000016>;
                                #size-cells = <0x0>;

                                tpm@57 {
                                        reg = <0x57>;
                                        label = "tpm";
                                        compatible = "nuvoton,npct650", "nuvoton
,npct601";
                                        linux,sml-base = <0x3f 0xfd450000>;
                                        linux,sml-size = <0x10000>;
                                        status = "okay";
                                        phandle = <0x10000017>;
                                };
                        };
```

Note that down to pure luck (as in, the order in which these are in the FDT) you end up with the TPM in the device tree that skiboot and Linux use. What is missing is the system-vpd entry.

dcrowell77, 2016-11-28:
@mabaiocchi - Please take a look

mabaiocchi, 2016-11-29:
(Sorry I thought I posted this yesterday, but it somehow didn't go through)

I think the system-vpd is in the first "**DUPLICATE-i2c-bus@2**" entry above in this section: **_eeprom@50 { }_** .  
And then the second "**i2c-bus@2**" entry has the TPM entry: **tpm@57**

So is the issue here that both of the system_vpd and tpm entries should be in the same i2c-bus@2 entry?

I'm assuming so and am looking at the code for how this could have possibly happened.

mabaiocchi, 2016-11-29:
Yes, hostboot code does not try to combine EEPROM and TPM devices onto the same I2C bus.  It was only trying to keep the EEPROM and TPMs on the same I2C masters and was creating unique buses for EEPROMs versus TPMs.

I will coordinate with @dcrowell77 and @mzipse to prioritize and assign this fix.

bofferdn, 2016-12-05:
@mabaiocchi Does this affect p9 + p8 ... or p8 only?  @mzipse If this is p8 only, are we going to need to come up with a fix?

mabaiocchi, 2016-12-05:
@bofferdn This affects p8 and p9.

dcrowell77, 2016-12-05:
Hostboot does not populate the devtree in p9 (Opal will consume HDAT), so this issue is p8 only.