open-power/hostboot #74: ISTEP 12.4 Error reported by hwpf

tniedermeier, 2016-12-07:
On Habanero we encountered a issue in ISTEP 12.4.

The system has been powered on and after a few seconds shut off again, after the fans began to spin-down from full speed, to show the loudness. It runs official Tyan firmware.

Back at our test lab, the server does not boot up any more. It runs into
an error in ISTEP 12.4, see the attached log. We have
already tried to flash the firmware again, but the situation stays the
same. We think that the error might be caused by the unclean shutdown,
but we also think that an unclean shutdown should not break the system or
bring it into a state where it cannot boot anymore.

Are there any similar experiences out there, Tyan Support told us they never encountered such a problem.

[ISTEP-12.4-Error-reported-by-hwpf.txt](https://github.com/open-power/hostboot/files/636144/ISTEP-12.4-Error-reported-by-hwpf.txt)

dcrowell77, 2016-12-07:
52.85031|Error reported by hwpf (0x0900)
 52.85940|  HW Procedure generated error. See User Data.
 52.85941|  ModuleId   0x02 MOD_HWP_RC_TO_ERRL
 52.85941|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
 52.85942|  UserData1  RC value from HWP : 0x00000000001a73c1
 52.85942|  UserData2  <unused> : 0x0000000000000000

obj/genfiles/fapiHwpReturnCodes.H:    RC_MSS_EFF_CONFIG_MISMATCH_SIDE = 0x1a73c1,

This is a plug-rule violation with the dimms.  I suspect you've got different sizes installed incorrectly.  The fault sensors should show you which dimm(s) we think are invalid.  A full parsed error log should also provide more detailed failure info.

tk-wfischer, 2016-12-09:
Thank you for your hint. It seems to me that the error code 0x00000000001a73c1 may be wrong in our case. The system is a Habanero with 8x 16 GB DIMMs (the [BP012...-T3 model](http://www.tyan.com/Barebones_TN71-BP012_BSP012T71V14HR-4T-3) with its original 8 x R-DDR3L 16GB DIMMs (we have not changed the memory configuration of the system)).

We have tested to AC cycle the machine, upload the [firmware of TYAN](http://www.tyan.com/support_download_bios.aspx?model=B.TN71-BP012) (TN71-BP012V101.hpm) via the BMC web interface again, but nothing has changed. The server has only run TYAN's firmware (we have not yet tested a self-compiled firmware on these servers). We have also tested to remove some DIMMs, but the error has remained.

My questions:
1. Do you know any other possible reasons which could lead to this error?
2. How can we get a full parsed error log? (you mentioned "_A full parsed error log should also provide more detailed failure info._")
3. Could it be that the hardware is somehow damaged? According to https://github.com/open-power/docs/blob/master/hostboot/HostBoot_PG.md HWPF means: _HWPF Hardware procedure framework. A code layer that executes black box hardware procedures for different platforms and environments. Also refers to a hostboot component that integrates the HWPF along with additional platform-unique implementations._ So in other words: Could it be that the black box hardware procedures encounter an error and report a wrong error code?

dcrowell77, 2016-12-11:
If you really have identical dimms installed in the right ports, then you might be seeing some kind of a hw failure.  The error indicates a plugging error only though, so it would be something fundamentally broken about the slot.  However, I doubt that based on this output:
 19.72785|HWAS|PRESENT> DIMM[03]=C0C00000C0C00000
That shows symmetrical dimms were detected.

I'm not familiar with any known issues in this error.  Has the system ever booted?  You might try just swapping out all of the dimms with some other known ones, just in case.

tk-wfischer, 2016-12-12:
Thanks for your answer. We have identical dimms installed. We have also tried to swap out all dimms and test with other ones, but the error remained.

And yes, the system has booted before in the past. Luckily I have also a log of such an successful startup:
[Successful-Bootlog-before-ISTEP-12.4-Error-occured.txt](https://github.com/open-power/hostboot/files/646297/Successful-Bootlog-before-ISTEP-12.4-Error-occured.txt)

I only just noticed a difference at the beginning of the log. In the successful log I see at the beginning:
 10.40495|HWAS|PRESENT> DIMM[03]=A0A00000A0A00000
 10.40497|HWAS|PRESENT> Membuf[04]=CC00000000000000
 10.40498|HWAS|PRESENT> Proc[05]=8000000000000000

In the current output (with the error) the "HWAS|PRESENT> DIMM" line shows "C" characters, while in the output where no error comes, are "A" characters above:
 19.72785|HWAS|PRESENT> DIMM[03]=C0C00000C0C00000
 19.72785|HWAS|PRESENT> Membuf[04]=CC00000000000000
 19.72786|HWAS|PRESENT> Proc[05]=8000000000000000

Can you tell us, how this line "HWAS|PRESENT> DIMM" can be read/interpreted? (and what the difference "A" vs. "C" means)?

dcrowell77, 2016-12-12:
Those messages are basically bitstrings that show which DIMMs we detected in a logical way.  The 'A' line shows that we see logical dimms 0,2,8,10,etc.  The 'C' shows dimms 0,1,8,9.  That tells me that the dimms are not in the same configuration now as what they were in originally.  I don't have a Tyan board so I can't tell you any more details on the physical locations that correspond to those positions.  If you have another box that is working, I would suggest comparing the slots.

tk-wfischer, 2016-12-12:
Thank you for your feedback. Tomorrow I'll be myself personally in the test lab, where is machine is placed. I'll look into it and compare the DIMM population and give you an update then.

tk-wfischer, 2016-12-13:
I have checked the DIMM population and indeed, the placing of the DIMMs was wrong.

We have placed the 8 DIMMs in the ivory-colored slots (according to [TN71-BP012_UG_v1.0h_for_Channel](http://www.tyan.com/manuals/TN71-BP012_UG_v1.0h_for_Channel.pdf) page 37) and now the systems boots up again. It seems that the DIMM population has been changed after one of the former tests and we did not notice it.

So now everything is working again :-) Sorry the work that we have caused from your side on paying attention to our issue. But we have learned that obj/genfiles/fapiHwpReturnCodes.H has a very valuable list of the error codes, which would be otherwise hard to interpret.

So thank you very much for your valuable help. You can close this issue.

dcrowell77, 2016-12-13:
You shouldn't really need to look at the code (though it can be handy).  The IPMI sensor values should have told you which dimms were misplaced.  The fault sensors are the way parts are identified for replacement.

closed 2016-12-13 by dcrowell77