open-power/hostboot #80: P9 Hostboot should handle bad DIMMs without using persistent GARD

rlippert, 2017-02-09:
Filing issue discussed separately with @dsanner.

Current hostboot support for handling bad DIMMs requires writing a GARD record for the bad DIMM and then rebooting in order to offline the DIMM.  This does not work well with our use case as we always run with GARD functionality disabled (and do not want to enable it).

Dean is going to look at modifying the code to offline a failed DIMM during training or MBIST and re-IPL similar to how P8 does it, or come up with some way of doing a temporary or 1-time type of GARD operation that will not persist on the flash.

rlippert, 2017-06-02:
@sannerd can you update this issue with the latest status?

JacobHarvey, 2017-09-06:
Update: DIMMs should not be GARDed or disabled during training anymore. Can't talk about MCBIST.... This issue probably still needs an update from @sannerd

dcrowell77, 2017-09-06:
I think this is all covered by the new HB_VOLATILE and pseudo-gard that Dean added support for recently.