open-power/hostboot #83: op-build failure at building hostboot-eb217e512338e8d2182762cbe97f55d4759e799e

deathscytheh28, 2017-02-12:
Dear Hostboot team,

I was trying to build op-build branch p9_bu_17_02_06 but failed at building hostboot.

Here is the error log:
    CXX        p9_mss_draminit_mc.C
tracehash: tracehash.c:539: copy_relocs: Assertion `format_node != NULL' failed.
../../../../src/build/mkrules/cc.rules.mk:33: recipe for target '../../../../obj/modules/istep13/p9_throttle_sync.o' failed
make[11]: *** [../../../../obj/modules/istep13/p9_throttle_sync.o] Aborted (core dumped)
make[11]: *** Deleting file '../../../../obj/modules/istep13/p9_throttle_sync.o'
make[11]: *** Waiting for unfinished jobs....
    MAKE       istep14 CODE
    CXX        p9_throttle_sync.C
    MAKE       istep15 CODE
    MAKE       istep16 CODE
    MAKE       istep18 CODE
    DEP        p9_tod_init.C
    DEP        p9_tod_setup.C
    DEP        p9_tod_utils.C
    DEP        tod_init.C
    CXX        tod_init.C
tracehash: tracehash.c:539: copy_relocs: Assertion `format_node != NULL' failed.
../../../../src/build/mkrules/cc.rules.mk:33: recipe for target '../../../../obj/modules/istep14/p9_throttle_sync.o' failed
make[11]: *** [../../../../obj/modules/istep14/p9_throttle_sync.o] Aborted (core dumped)
make[11]: *** Deleting file '../../../../obj/modules/istep14/p9_throttle_sync.o'
../../../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/CODE/BODY' failed
make[10]: *** [_BUILD/PASSES/CODE/BODY] Error 2
../../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/CODE/istep14' failed
make[9]: *** [_BUILD/SUBDIR/CODE/istep14] Error 2
make[9]: *** Waiting for unfinished jobs....
    CXX        p9_tod_utils.C
../../../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/CODE/BODY' failed
make[10]: *** [_BUILD/PASSES/CODE/BODY] Error 2
../../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/CODE/istep13' failed
make[9]: *** [_BUILD/SUBDIR/CODE/istep13] Error 2
    CXX        p9_tod_setup.C
    CXX        p9_tod_init.C
    LD         libistep18.so
../../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/CODE/POST' failed
make[8]: *** [_BUILD/PASSES/CODE/POST] Error 2
../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/CODE/isteps' failed
make[7]: *** [_BUILD/SUBDIR/CODE/isteps] Error 2
../../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/CODE/POST' failed
make[6]: *** [_BUILD/PASSES/CODE/POST] Error 2
../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/CODE/usr' failed
make[5]: *** [_BUILD/SUBDIR/CODE/usr] Error 2
../src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/CODE/POST' failed
make[4]: *** [_BUILD/PASSES/CODE/POST] Error 2
src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/SUBDIR/CODE/src' failed
make[3]: *** [_BUILD/SUBDIR/CODE/src] Error 2
src/build/mkrules/passes.rules.mk:105: recipe for target '_BUILD/PASSES/CODE/POST' failed
make[2]: *** [_BUILD/PASSES/CODE/POST] Error 2
package/pkg-generic.mk:216: recipe for target '/home/fxn/Zaius/op-build/output/build/hostboot-eb217e512338e8d2182762cbe97f55d4759e799e/.stamp_built' failed
make[1]: *** [/home/fxn/Zaius/op-build/output/build/hostboot-eb217e512338e8d2182762cbe97f55d4759e799e/.stamp_built] Error 2
Makefile:76: recipe for target '_all' failed
make: *** [_all] Error 2

Does anyone know how to solve this issue?

By the way, I can build op-build v1.14 without any problem.

Thanks a lot.

dcrowell77, 2017-02-12:
Not sure who else has tried that branch, maybe @sannerd .

None of those errors ring a bell to me.  Have you tried doing a hostboot-dirclean and rebuiilding?

deathscytheh28, 2017-02-13:
Hi @dcrowell77 ,

Yes, I've tried doing a hostboot-dirclean and then doing op-build again, and still see this error.

williamspatrick, 2017-02-13:
@ozbenh - I think you ran into this too?

e-liner, 2017-02-13:
I was able to pull out a clean branch from that tag and build successfully.  So there's either something extra in your directory, or something left over from a previous build that dirclean couldn't get to.  You could also run an op-build dirclean ?

ozbenh, 2017-02-13:
@e-liner I hit something similar (fails on a different file but same error). This is a clean clone from Prachi's witherspoon_bup_02_08 branch

e-liner, 2017-02-13:
Okay.  Did you also run an op-build dirclean, or just a hostboot dirclean?  There might be an MRW mismatch, or something else leftover in op-build/output/build that is messing up the process.

ozbenh, 2017-02-13:
It was a fresh clone

ozbenh, 2017-02-13:
I can try a clean & rebuild later today

ozbenh, 2017-02-14:
Try this patch
`
diff --git a/src/build/trace/tracehash.c b/src/build/trace/tracehash.c
index 076d3e1..8aa6a40 100644
--- a/src/build/trace/tracehash.c
+++ b/src/build/trace/tracehash.c
@@ -665,7 +665,7 @@ char* create_format_string(const char* string)
     size_t r_pos = 0; // Current position in result string.
 
     // Iterate through source string looking for format tags.
-    for(size_t pos = 0; pos < length; pos++)
+    for(size_t pos = 0; pos < (length-2); pos++)
     {
         // Skip if not %.
         if (string[pos] != '%') continue;
@@ -882,7 +882,7 @@ void parse_traceinfo(bfd* inFile, asection* s)
             static const char filesep[] = ": ";
 
             size_t len_begin = replace_pos - (char*)&contents[pos];
-            size_t len_end = strlen(&contents[pos]) -
+            size_t len_end = strlen(&contents[pos + len_begin]) -
                              strlen(TRACEPP_REPLACE_WITH_FILENAME);
             size_t length = len_begin + strlen(filename) + len_end +
                             strlen(filesep) + 1;

```

ozbenh, 2017-02-14:
Try this patch:
```
diff --git a/src/build/trace/tracehash.c b/src/build/trace/tracehash.c
index 076d3e1..8aa6a40 100644
--- a/src/build/trace/tracehash.c
+++ b/src/build/trace/tracehash.c
@@ -665,7 +665,7 @@ char* create_format_string(const char* string)
     size_t r_pos = 0; // Current position in result string.
 
     // Iterate through source string looking for format tags.
-    for(size_t pos = 0; pos < length; pos++)
+    for(size_t pos = 0; pos < (length-2); pos++)
     {
         // Skip if not %.
         if (string[pos] != '%') continue;
@@ -882,7 +882,7 @@ void parse_traceinfo(bfd* inFile, asection* s)
             static const char filesep[] = ": ";
 
             size_t len_begin = replace_pos - (char*)&contents[pos];
-            size_t len_end = strlen(&contents[pos]) -
+            size_t len_end = strlen(&contents[pos + len_begin]) -
                              strlen(TRACEPP_REPLACE_WITH_FILENAME);
             size_t length = len_begin + strlen(filename) + len_end +
                             strlen(filesep) + 1;
```

ozbenh, 2017-02-14:
More details: the bug I tracked down in tracehash is caused by buggy format strings that terminate with a % sign. The one tripping me today was:
```
diff --git a/src/import/chips/p9/procedures/hwp/nest/p9_throttle_sync.C b/src/import/chips/p9/procedures/hwp/nest/p9_throttle_sync.C
index 0e20e7e..3c3195c 100644
--- a/src/import/chips/p9/procedures/hwp/nest/p9_throttle_sync.C
+++ b/src/import/chips/p9/procedures/hwp/nest/p9_throttle_sync.C
@@ -203,7 +203,7 @@ fapi2::ReturnCode throttleSync(
             // this MCS as master
             if (l_mcSide[l_sideNum].masterMcsFound == false)
             {
-                FAPI_INF("Mark MCS %u as master for MC side %",
+                FAPI_INF("Mark MCS %u as master for MC side %u",
                          l_pos, l_sideNum);
                 l_mcSide[l_sideNum].masterMcsFound = true;
                 l_mcSide[l_sideNum].masterMcs = l_mcs;

```

dcrowell77, 2017-02-14:
Odd that we aren't seeing the failures, it must depend on some environmental differences in the build environment.  Regardless, the bottom issue is definitely a bug to fix.

ozbenh, 2017-02-14:
Yes it depends on what happens to be in memory beyond the end of the malloc'ed string, so very environment sensitive

ozbenh, 2017-02-14:
BTW. Please double check my patch ! I used length - 2 because "length" already includes the 0 terminal byte and we want to make sure we have at least one char beyond the "%" ... that way we have two which should deal with most cases
However it might be a better fix to check for the 0 terminator in the parsing itself to deal with all corner cases

deathscytheh28, 2017-02-17:
@ozbenh , I tried your patch and it worked for me, thanks.

BTW, I noticed that I could build PNOR without patch on Ubuntu 14.04 successfully, but need patch on Ubuntu 16.04.

Thank you all.

wghoffa, 2017-03-22:
The bad trace was fixed here: https://github.com/open-power/hostboot/commit/eec586620b5f5970520f8def196403d2b4f71857

wghoffa, 2017-03-24:
Base fix is here: https://github.com/open-power/hostboot/commit/82c7782756df58c99a0979777250c5f0987980fa

closed 2017-03-24 by wghoffa