open-power/hostboot #91: [P9] Couldn't leave the PEC1's phb2 bus link empty as the Romulus HW design

HankChang, 2017-03-03:
Hi,
Based on the Romulus HW design, its 2nd CPU's PEC1 phb2_x8_pci doesn't connect to any PCIE slot or device. So, once we leave the specific bus link as empty in MRW, hostboot will detect following errors:
==>
Code bug! PEC PCIE IOP configuration not found.
PEC chip target number 0x00000001, HUID = 0x002D0004.

```
# Full trace:
 30.21276|ISTEP  8.17
 30.21543|INITSVC|I>Progress Code 8.17 Sent
 30.21549|MBOX|W>MSGSEND - mailboxsp is disabled. Message dropped! msgQ=0x80000008 type=0x40000011
 30.22808|ISTEPS_TRACE|PROC 00050000 PEC0 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x01
 30.22810|ISTEPS_TRACE|PROC 00050000 PEC1 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x03
 30.22813|ISTEPS_TRACE|PROC 00050000 PEC2 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x07
 30.22814|ISTEPS_TRACE|Running p9_pcie_scominit HWP on target HUID 00050000
 30.23950|FAPI|p9_pcie_scominit.C: Start
 30.27901|FAPI|p9_pcie_scominit.C: End
 30.27902|ISTEPS_TRACE|SUCCESS :  proc_pcie_scominit HWP

 **30.27916|ISTEPS_TRACE|E>computeProcPcieConfigAttrs> Code bug! PEC PCIE IOP configuration not found.
Continuing with no PHBs active.
IOP0 Lane set 0: Lane mask = 0xFF00
IOP0 Lane set 1: Lane mask = 0x0000
IOP0 Lane set 2: Lane mask = 0x0000**

 30.27918|ISTEPS_TRACE|PEC chip target number 0x00000001, HUID = 0x002D0004.
 30.27919|ISTEPS_TRACE|Proc chip target HUID = 0x00050001.
 30.27921|ERRL|>>addProcedureCallout(0x55, 0x6)
 30.27924|ERRL|E>ErrlEntry::collectTrace(): getBuffer(istep) rets zero.
 30.27926|ERRL|commitErrLog() called by 1700 for eid=90000008, Reasoncode=170E
 30.27927|ERRL|>>ErrlManager::sendErrlogToMessageQueue
 30.27929|ERRL|I>Send an error log to message queue to commit. eid=90000008
 30.27930|ERRL|<<ErrlManager::sendErrlogToMessageQueue
 30.27935|ERRL|I>Got an error log Msg - Type: 0x00000030
 30.27971|ISTEPS_TRACE|PHB 1 on PEC 1 has been found INACTIVE. Deconfiguring PHB 1.
 30.32480|HWAS_I|I>hwasPlat.C: HostBoot is running so system is not at runtime.
 30.32481|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E0007, errlEid 0xFF12
 30.32484|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 30.35902|================================================
 30.35903|Error reported by istep (0x1700) PLID 0x90000008
 30.37033|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E0007 (i_deconfigRule 0)
 30.37037|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E0008, errlEid 0xFF12
 30.37039|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 30.37041|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E0008 (i_deconfigRule 0)
 30.37045|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002D0004, errlEid 0xFF12
 30.37047|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 30.37049|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002D0004 (i_deconfigRule 0)
 30.37052|ISTEPS_TRACE|PHB 2 on PEC 1 has been found INACTIVE. Deconfiguring PHB 2.
 30.37053|HWAS_I|I>hwasPlat.C: HostBoot is running so system is not at runtime.
 30.37055|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E0008, errlEid 0xFF12
 30.37057|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E0008 (i_deconfigRule 0)
 30.37062|ISTEPS_TRACE|PHB 4 on PEC 2 has been found INACTIVE. Deconfiguring PHB 4.
 30.37064|HWAS_I|I>hwasPlat.C: HostBoot is running so system is not at runtime.
 30.37065|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E000A, errlEid 0xFF12
 30.37068|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 30.37070|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E000A (i_deconfigRule 0)
 30.37074|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E000B, errlEid 0xFF12
 30.37076|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 30.37078|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E000B (i_deconfigRule 0)
 30.37082|ISTEPS_TRACE|PHB 5 on PEC 2 has been found INACTIVE. Deconfiguring PHB 5.
 30.37084|HWAS_I|I>hwasPlat.C: HostBoot is running so system is not at runtime.
 30.37086|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E000B, errlEid 0xFF12
 30.37088|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E000B (i_deconfigRule 0)
 30.37097|ISTEPS_TRACE|PROC 00050001 PEC0 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x01
 30.37099|ISTEPS_TRACE|PROC 00050001 PEC1 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x00
 30.37101|ISTEPS_TRACE|PROC 00050001 PEC2 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x04
 30.37103|ISTEPS_TRACE|Running p9_pcie_scominit HWP on target HUID 00050001
 30.37104|FAPI|p9_pcie_scominit.C: Start
 30.38200|  No valid PCIE IOP configuration found.  All PHBs on the processor will be disabled.
 30.38201|  ModuleId   0x0a MOD_COMPUTE_PCIE_CONFIG_ATTRS
 30.38202|  ReasonCode 0x170e RC_INVALID_CONFIGURATION
 30.38202|  UserData1  Target processor chip's HUID : 0x00050001002d0004
 30.38203|  UserData2  Target PEC HUID : 0xff00000000000000
 30.39333|------------------------------------------------
 30.39334|  Callout type             : Procedure Callout
 30.39334|  Procedure                : EPUB_PRC_HB_CODE
 30.39335|  Priority                 : SRCI_PRIORITY_HIGH
 30.39336|------------------------------------------------
 30.39336|  Hostboot Build ID:
 30.39337|================================================
 30.43183|ERRL|>>saveErrLogToPnor eid=90000008
 30.43185|ERRL|I>saveErrLogToPnor: 332 bytes flattened into 0x812c1000, slot 0
 30.43190|PNOR|>>PnorDD::writeFlash(i_address=0xa0000)>
 30.48238|PNOR|<<PnorDD::writeFlash(i_address=0xa0000)> io_buflen=00001200
 30.48240|ERRL|<<saveErrLogToPnor returning true
 30.48242|HWAS_I|I>hwasCallout.C: Procedure callout; proc 0x55 priority 0x6
 31.29717|FAPI|p9_pcie_scominit.C: End
 31.29718|ISTEPS_TRACE|SUCCESS :  proc_pcie_scominit HWP
 31.29839|ERRL|I>Got an error log Msg - Type: 0x00000039
 31.29841|ERRL|I>Flush message received
 31.29842|INITSVC|<<doIstep: step 8, substep 17
```

Please help us check it. Thanks! :)

HankChang, 2017-03-03:
BTW, for Romulus bringup phase, we added a workaround in MRW here: 
(just simply created a slot target to connect with 2nd CPU's phb2 x8 pcie)
https://github.com/HankChang/romulus-xml/commit/7949b95585e631b6c9f2bcb13bf4923d9815c8ba

HankChang, 2017-03-03:
The versions we're using:
```
	op-build-v1.14-61-gfa83b09-dirty
	buildroot-2016.11-4-g6b9fd80
	skiboot-8c51fda-p393d527
	hostboot-c3170c0
	linux-4.10.1-p1d7e95b
	petitboot-v1.4.1-p23c5d8b
	romulus-xml-e843e16-p5a6c11a
	occ-4141b5f
	hostboot-binaries-446c2ea
	capp-ucode-1bb7503

	- Buildroot Commit : "6b9fd80"
	- Git Branch: "master-next"
```

dcrowell77, 2017-03-03:
This feels like a bug in the ServerWiz2 logic.  We should still get valid (if disabled) values for disconnected busses.  I want @ploetzma to weigh in on the error.

ploetzma, 2017-03-06:
@HankChang If I understand correctly, proc1pec1phb1_x8 is connected to some device, but proc1pec1phb2 is not? Under the current HB code that is still an invalid configuration (as it expects pec1 to have all phb's used). If no phb's are used under a pec we will deconfigure both phb's and the resultant pec. I have not come across a partially used pec1 before and am not sure if this is expected used.

ploetzma, 2017-03-06:
I just talked with Rick Mata and he states that using the PEC and disabling one of the PHB's **is** allowed. I will have to updated the PEC1 table in host_proc_pcie_scominit.C

HankChang, 2017-03-07:
@ploetzma Yes, correct, Romulus 2nd CPU's proc1pec1phb1_x8 is connected to a physical x8 PCIE slot, but proc1pec1phb2 is not connected to anything.

HankChang, 2017-03-07:
Thank you for the clarification.

ploetzma, 2017-03-08:
@HankChang Here is the patch, if you could test this on Romulus that would be great.

https://github.com/ploetzma/op-build/commit/a42959aaed4e078f9f09d04757d928e4c7b5e6ba#diff-7ba3b48e13e0b7ebbb425bd6350ef5b8

HankChang, 2017-03-09:
Thank you @ploetzma , I've tested your patch on Romulus, and everything looks good!

With your patch on hostboot-c3170c0, there is no error message pop out during istep 8.17 and the 2nd CPU's PHB2 (proc1pec1phb2) can be de-configured individually & successfully.

```
 32.92484|INITSVC|>>doIstep: step 8, substep 17, task proc_pcie_scominit
 32.92485|ISTEP  8.17
 32.92645|INITSVC|I>Progress Code 8.17 Sent
 32.92651|MBOX|W>MSGSEND - mailboxsp is disabled. Message dropped! msgQ=0x80000008 type=0x40000011
 32.93860|ISTEPS_TRACE|PROC 00050000 PEC0 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x01
 32.93862|ISTEPS_TRACE|PROC 00050000 PEC1 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x03
 32.93865|ISTEPS_TRACE|PROC 00050000 PEC2 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x07
 32.93867|ISTEPS_TRACE|Running p9_pcie_scominit HWP on target HUID 00050000
 32.93868|FAPI|p9_pcie_scominit.C: Start
 32.98991|FAPI|p9_pcie_scominit.C: End
 32.98992|ISTEPS_TRACE|SUCCESS :  proc_pcie_scominit HWP
 32.99041|ISTEPS_TRACE|PHB 2 on PEC 1 has been found INACTIVE. Deconfiguring PHB 2.
 33.01182|HWAS_I|I>hwasPlat.C: HostBoot is running so system is not at runtime.
 33.01184|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E0008, errlEid 0xFF12
 33.01187|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 33.03333|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E0008 (i_deconfigRule 0)
 33.03339|ISTEPS_TRACE|PHB 4 on PEC 2 has been found INACTIVE. Deconfiguring PHB 4.
 33.03341|HWAS_I|I>hwasPlat.C: HostBoot is running so system is not at runtime.
 33.03342|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E000A, errlEid 0xFF12
 33.03345|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 33.03347|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E000A (i_deconfigRule 0)
 33.03351|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E000B, errlEid 0xFF12
 33.03353|HWAS_I|I>deconfigGard.C: Setting Target HWAS_STATE: functional=0, deconfiguredByEid=0xFF12
 33.03355|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E000B (i_deconfigRule 0)
 33.03360|ISTEPS_TRACE|PHB 5 on PEC 2 has been found INACTIVE. Deconfiguring PHB 5.
 33.03361|HWAS_I|I>hwasPlat.C: HostBoot is running so system is not at runtime.
 33.03363|HWAS_I|I>deconfigGard.C: Deconfiguring Target 002E000B, errlEid 0xFF12
 33.03365|HWAS_I|I>deconfigGard.C: _deconfigureByAssoc for 002E000B (i_deconfigRule 0)
 33.03374|ISTEPS_TRACE|PROC 00050001 PEC0 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x01
 33.03377|ISTEPS_TRACE|PROC 00050001 PEC1 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x02
 33.03379|ISTEPS_TRACE|PROC 00050001 PEC2 -> ATTR_PROC_PCIE_IOVALID_ENABLE: 0x04
 33.03381|ISTEPS_TRACE|Running p9_pcie_scominit HWP on target HUID 00050001
 33.03382|FAPI|p9_pcie_scominit.C: Start
 34.32371|FAPI|p9_pcie_scominit.C: End
 34.32372|ISTEPS_TRACE|SUCCESS :  proc_pcie_scominit HWP
 34.32451|ERRL|I>Got an error log Msg - Type: 0x00000039
 34.32453|ERRL|I>Flush message received
 34.32454|INITSVC|<<doIstep: step 8, substep 17
```

Also, I've checked all active PHBs on the 2nd CPU were functional while in Petitboot:
```
/ # lspci
0000:00:00.0 PCI bridge: IBM Device 04c1
0001:00:00.0 PCI bridge: IBM Device 04c1
0002:00:00.0 PCI bridge: IBM Device 04c1
0002:01:00.0 Mass storage controller: LSI Logic / Symbios Logic SAS3008 PCI-Express Fusion-MPT SAS-3 (rev 02)
0003:00:00.0 PCI bridge: IBM Device 04c1
0003:01:00.0 USB controller: Texas Instruments TUSB73x0 SuperSpeed USB 3.0 xHCI Host Controller (rev 02)
0004:00:00.0 PCI bridge: IBM Device 04c1
0004:01:00.0 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0005:00:00.0 PCI bridge: IBM Device 04c1
0005:01:00.0 PCI bridge: ASPEED Technology, Inc. AST1150 PCI-to-PCI Bridge (rev 04)
0005:02:00.0 VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 41)
0030:00:00.0 PCI bridge: IBM Device 04c1
0030:01:00.0 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0030:01:00.1 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0030:01:00.2 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0030:01:00.3 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0031:00:00.0 PCI bridge: IBM Device 04c1
0031:01:00.0 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0031:01:00.1 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0031:01:00.2 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0031:01:00.3 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0033:00:00.0 PCI bridge: IBM Device 04c1
0033:01:00.0 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0033:01:00.1 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0033:01:00.2 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
0033:01:00.3 Ethernet controller: Broadcom Limited NetXtreme BCM5719 Gigabit Ethernet PCIe (rev 01)
/ # 
```

HankChang, 2017-03-09:
So once the patch is merged to upstream, then this issue can be closed. 
Thanks!

ploetzma, 2017-04-19:
@HankChang this was merged 3/16 (https://github.com/open-power/hostboot/commit/ac0ff7b373da5e6bc16365d9c1421e701bb1e613)

If you could verify and close, that would be great.

ploetzma, 2017-06-20:
@HankChang - can you close this?

closed 2018-07-20 by wghoffa