open-power/hostboot #94: Host sending incomplete Inventory byte data [Witherspoon]

gkeishin, 2017-03-16:
```
root@witherspoon:~# ls /tmp/core_ipmid.852
/tmp/core_ipmid.852
root@witherspoon:~#

Feb 22 07:12:45 witherspoon ipmid[852]: Error updating inventory
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Serial Number] : Type Code=[ASCII+Latin] : Len = [16]terminate called after throwing an instance of 'std::logic_error'
```
`Driver Info`

```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/630/
- [x] VERSION_ID : v1.99.3-93-gf0a463a
- [x] Kernel     : Linux witherspoon 4.7.10-c72ced5a19ceae19c7d792ae3e982bee5607dab4 #1 Wed Mar 15 22:22:06 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```
```
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Serial Number] : Type Code=[ASCII+Latin] : Len = [16]terminate called after throwing an instance of 'std::logic_error'
Feb 22 07:16:21 witherspoon ipmid[852]: what(): basic_string::_M_construct null not valid
Feb 22 07:16:21 witherspoon ipmid[852]: : Val = [000000000000]
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Part Number] : Type Code=[ASCII+Latin] : Len = [16] : Val = [0000000]
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [FRU File ID] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 1] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 2] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 3] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 4] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 5] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 6] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 7] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: _append_to_dict: VPD Key = [Custom Field 8] : Type Code = [BINARY] : Len = [0] : Val = []
Feb 22 07:16:21 witherspoon ipmid[852]: fru data size:[56], area offset:[8], area_size:[48]
Feb 22 07:16:21 witherspoon ipmid[852]: Successfully verified area checksum. offset:[8]
Feb 22 07:16:21 witherspoon ipmid[852]: SUCCESS: Populated FRU areas for:[/tmp/ipmifru04]

                        Core happened during parsing of fruid=04
                        Field Name = Serial Number, Expected Length = 16,Actual Data = 12 bytes long.
                        Field Name = PartNumber , Exptceted Length = 16,Actual Data = 7 bytes long.
                        This difference is causing the core.

```

Logs are available in this defect: https://github.com/openbmc/openbmc/issues/1326

dcrowell77, 2017-03-16:
What part has this issue - the processor module, the planar or a dimm?

Why are you making an assumption about the number of bytes you get for these fields?  Is the problem with the metadata you are getting from the ipmi message or the actual size of the data itself?

ratagupt, 2017-03-16:
fruid 04 is for MemBuff.
it is a data inside the ipmi message which is incomplete.

wghoffa, 2017-03-16:
Could use a little bit more detail than that. Are you stating we send an IPMI message stating a size of 12 and then send you less bytes than that? Or are you for whatever reason assuming we will always send you 12 bytes?

wghoffa, 2017-03-16:
Also, what is "MemBuff" in this context? Planar VPD? There is no standalone memory card on this system.

ratagupt, 2017-03-17:
What I mean to say that host boot send an IPMI message stating 16bytes for serial number but send only 12 bytes.
Targeting type Membuff
TARGETING::TYPE_MEMBUF
https://github.com/open-power/hostboot/blob/master/src/usr/ipmi/ipmifruinv.C.

gkeishin, 2018-08-20:
`On GA1:`
```
"/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm3": {
        "BuildDate": "",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "0xce80",
        "Model": "M393A4K40BB2-CTD    ",
        "PartNumber": "",
        "Present": 1,
        "PrettyName": "0x0c",
        "SerialNumber": "0x3447b1ae",
        "Version": "0x00"
    },

DIMM 3 is the FRU ID 4 mapped and we can see the serial number correct `34 47 b1 ae`.
	
root@witherspoon:~# hexdump -C /tmp/ipmifru04
00000000  01 00 00 00 01 00 00 fe  01 06 00 02 ce 80 01 0c  |................|
00000010  d4 4d 33 39 33 41 34 4b  34 30 42 42 32 2d 43 54  |.M393A4K40BB2-CT|
00000020  44 20 20 20 20 01 00 04  34 47 b1 ae 00 00 00 c1  |D    ...4G......|
00000030  00 00 00 00 00 00 00 da                           |........|
00000038
root@witherspoon:~# 
```

`On GA2:`
```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm3": {
        "BuildDate": "",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "0x2c80",
        "Model": "36ASF4G72PZ-2G6D1   ",
        "PartNumber": "",
        "Present": 1,
        "PrettyName": "0x0c",
        "SerialNumber": "0x15d279fa",
        "Version": "0x31"
    },

	
DIMM 3 is the FRU ID 4 mapped and we can see the serial number correct `15 d2 79 fa`.
	
root@witherspoon:~# hexdump -C /tmp/ipmifru04
00000000  01 00 00 00 01 00 00 fe  01 06 00 02 2c 80 01 0c  |............,...|
00000010  d4 33 36 41 53 46 34 47  37 32 50 5a 2d 32 47 36  |.36ASF4G72PZ-2G6|
00000020  44 31 20 20 20 01 31 04  15 d2 79 fa 00 00 00 c1  |D1   .1...y.....|
00000030  00 00 00 00 00 00 00 97                           |........|
00000038
root@witherspoon:~# 
```

We can close this now. Sorry not following up on this one.

closed 2018-08-20 by dcrowell77