open-power/hostboot #95: IPL Performance Issue due to IPMI Polling

wghoffa, 2017-03-16:
@ozbenh has reported that the IPL performance is hampered due to the IPMI Interrupt polling mechanism in place in HB today. A sizeable performance increase could be achieved if HB configured/used LPC interrupts for IPMI communication.

rlippert, 2017-06-02:
FWIW I've noticed that boot performance actually speeds up a bit when decreasing the IPMI poll interval from default 100ms to 10ms.

Most of the time in the HB IPMI driver seems to be spent sleeping when the data is already available.

ozbenh, 2017-06-02:
Right. I tried to make the polling faster while a request is in progress, but the structure of the driver is a mess, it's hard to do without races and without more work than just adding 3 lines :-) Definitely worth looking into though.

rlippert, 2017-06-02:
Yea I also noticed that the BMC-side ipmid would crash a lot when I sped up the polling and hostboot tried to send a bunch of errl data to it (upon encountering some HW errors) which is why I didn't send the poll rate change out for review...

It would be great to have a unified driver framework between hostboot and skiboot :)

ozbenh, 2017-06-02:
Hahaha, as long as it's not Hostboot's :-)

There are two long term options I've been meaning to tackle eventually.

One is to create some kind of interpreted/JITed (eBPF based ?) mean of adding "methods" to device-tree nodes, which can be run both at boot time, but also at runtime in a sandbox in linux. A kind of equivalent of ACPI methods for device-tree. The idea being that a "node" can provide methods that can be either in that form, or natively implemented by the env (an OS for example could "replace" the FW provided bytecode driver for  a given peice of HW with a native one).

Another thing I've been thinking of looking at more closely is coreboot. However, I can't really make skiboot GPL at this point.

As for the BMC side ipmid, I assume you've been filing bug reports ? :-)