open-power/hostboot #96: EC property is

ratagupt, 2017-03-22:
EC property on processor VPD is not consistent.
EC property sometimes comes as CustomProperty2 or CustomProperty3.
Could be the difference is some time host boot sends the ECID and sometimes not.

First Instance
===========
_append_to_dict: VPD Key = [Manufacturer] : Type Code=[ASCII+Latin] : Len = [3] : Val = [IBM]
_append_to_dict: VPD Key = [Name] : Type Code=[ASCII+Latin] : Len = [16] : Val = [PROCESSOR MODULE]
_append_to_dict: VPD Key = [Serial Number] : Type Code=[ASCII+Latin] : Len = [12] : Val = [YA3932647569]
_append_to_dict: VPD Key = [Part Number] : Type Code=[ASCII+Latin] : Len = [7] : Val = [00UL864]
_append_to_dict: VPD Key = [FRU File ID] : Type Code = [BINARY] : Len = [2] : Val = [0x2020]
_append_to_dict: VPD Key = [Custom Field 1] : Type Code = [BINARY] : Len = [0] : Val = []
## # **__append_to_dict: VPD Key = [Custom Field 2] : Type Code=[ASCII+Latin] : Len = [5] : Val = [EC:20]_**
_append_to_dict: VPD Key = [Custom Field 3] : Type Code = [BINARY] : Len = [0] : Val = []

Another Instance
==============
  _append_to_dict: VPD Key = [Manufacturer] : Type Code=[ASCII+Latin] : Len = [3] : Val = [IBM]
_append_to_dict: VPD Key = [Name] : Type Code=[ASCII+Latin] : Len = [16] : Val = [PROCESSOR MODULE]
_append_to_dict: VPD Key = [Serial Number] : Type Code=[ASCII+Latin] : Len = [12] : Val = [YA3932647569]
_append_to_dict: VPD Key = [Part Number] : Type Code=[ASCII+Latin] : Len = [7] : Val = [00UL864]
_append_to_dict: VPD Key = [FRU File ID] : Type Code = [BINARY] : Len = [2] : Val = [0x2020]
_append_to_dict: VPD Key = [Custom Field 1] : Type Code = [BINARY] : Len = [0] : Val = []
_append_to_dict: VPD Key = [Custom Field 2] : Type Code=[ASCII+Latin] : Len = [37] : Val = [ECID:0196183141201596090D000000A42C87]
 ## **_append_to_dict: VPD Key = [Custom Field 3] : Type Code=[ASCII+Latin] : Len = [5] : Val = [EC:20]**
_append_to_dict: VPD Key = [Custom Field 4] : Type Code = [BINARY] : Len = [0] : Val = []
_append_to_dict: VPD Key = [Custom Field 5] : Type Code = [BINARY] : Len = [0] : Val = []
_append_to_dict: VPD Key = [Custom Field 6] : Type Code = [BINARY] : Len = [0] : Val = []
_append_to_dict: VPD Key = [Custom Field 7] : Type Code = [BINARY] : Len = [0] : Val = []
_append_to_dict: VPD Key = [Custom Field 8] : Type Code = [BINARY] : Len = [0] : Val = []

dcrowell77, 2017-03-22:
Can you provide some context here?  Where is all of that code you pasted coming from?  It isn't Hostboot so I'm guessing maybe OpenBMC?  Or some external tool that is processing the FRU inventory?

> Could be the difference is some time host boot sends the ECID and sometimes not.
This is exactly the case.  Early in the boot we do not have the ECID information for the processor yet.  That cannot be read from the chip until midway through boot.  Once we read that, we resend the inventory with that additional info.

ratagupt, 2017-03-24:
@dcrowell77 :- you are correct that these are the traces on the openbmc.
what is the custom field for EC?

dcrowell77, 2017-03-24:
First of all, EC != ECID, which are you referring to?  

For ECID there is a custom property that starts with the ASCII string "ECID:".  As mentioned above, that isn't filled in during the initial data population since the data is not available until later in the IPL.

For EC there is a custom property that starts with the ASCII string "EC:".  This should be present for all FRUs that contain chips.

ratagupt, 2017-03-31:
@dcrowell77 : I was more interested in which custom property ,host boot sends the EC?

ratagupt, 2017-03-31:
I understand that EC and ECID is not same,I am more interested to know in which custom property host boot send the EC data?

dcrowell77, 2017-03-31:
I'm still confused about what you are asking.  The custom property is literally "EC: 10".  If you are expecting that the value is always in the same order, that is not a good assumption.  We add the custom fields as we find them so the orders are not consistent.

Maybe @wghoffa can help translate the mess that is IPMI FRU Inventory...

pragupta, 2018-08-17:
Working as designed.

closed 2018-08-17 by pragupta