open-power/hostboot #97: Hostboot master-p8 built on ppc64 doesn't boot on Habanero

stewart-ibm, 2017-03-27:
op-build 1b366c2de9d38cbcc9a25635a20961c95f976687
hostboot-a4b7bb77b3c152af89a57f904f323228ad24373d

Does not boot on habanero:

```
  3.64097|Ignoring boot flags, incorrect version 0x0
  3.69225|ISTEP  6. 3
  4.05370|ISTEP  6. 4
  4.05426|ISTEP  6. 5
  8.84973|HWAS|PRESENT> DIMM[03]=AAAA0000AAAA0000
  8.84974|HWAS|PRESENT> Membuf[04]=CC00000000000000
  8.84974|HWAS|PRESENT> Proc[05]=8000000000000000
  8.99222|ISTEP  6. 6
  9.24862|ISTEP  6. 7
  9.38946|System shutting down with error status 0x90000289
  9.41836|================================================
  9.41837|Error reported by initservice (0x0500)
  9.41837|  Initialization Service launched a function and the task returned an error.
  9.41837|  ModuleId   0x01 BASE_INITSVC_MOD_ID
  9.41838|  ReasonCode 0x0506 WAIT_FN_FAILED
  9.41838|  UserData1  task id or task return code : 0x000000000000008d
  9.41838|  UserData2  returned status from task : 0x0000000000000001
  9.41838|User Data Section 0, type UD
  9.41839|  Subsection type 0x06
  9.41839|  ComponentId errl (0x0100)
  9.41839|  CALLOUT
  9.41839|  PROCEDURE ERROR
  9.41839|  Procedure: 85
  9.41840|User Data Section 1, type UD
  9.41840|  Subsection type 0x07
  9.41840|  ComponentId errl (0x0100)
  9.41841|User Data Section 2, type UD
  9.41841|  Subsection type 0x01
  9.41842|  ComponentId errl (0x0100)
  9.41842|  STRING
  9.41842|  host_cancontinue_clear
  9.41842|User Data Section 3, type UD
  9.41843|  Subsection type 0x03
  9.41843|  ComponentId errl (0x0100)
  9.41843|User Data Section 4, type UD
  9.41844|  Subsection type 0x01
  9.41844|  ComponentId errl (0x0100)
  9.41844|  STRING
  9.41845|  Hostboot Build ID: hostboot-a4b7bb7/hbicore.bin
  9.41845|================================================
```

This is after a full power off of the machine, fresh flash and boot attempt.

This particular Habanero has a TPM.

Built on Fedora 25 ppc64 on a (large, 224 threads) POWER7 system.

Host GCC is gcc (GCC) 6.3.1 20161221 (Red Hat 6.3.1-1)

williamspatrick, 2017-03-29:
This is the same symptom as #93 too isn't it?

Hostboot team needs their printk buffer to debug this much because it just indicates a process in Hostboot crashed.

williamspatrick, 2017-03-29:
The real esel might have the printk buffer too.

stewart-ibm, 2017-03-30:
I think #93 may be a misnomer, and perhaps it's just a product of this same bug (that was built on the same machine and it looks *awfully* similar).

Ironically, in a world of reproducible builds, we'd be able to spot this without even attempting the boot.

I'll try and bisect down what change made this start occurring.