open-power/hostboot #110: Hardware Error Detected on boot - Callout Physical:/Sys0/Node0/Proc0

gkeishin, 2017-08-19:
On boot on a DD2 system, we saw this error logged. The host did boot to OS

`Driver Info`

```
/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/686a9761": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.8-292-g92b1559"
    },
    "/xyz/openbmc_project/software/fd2cbe15": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.18-74-g53f4235-dirty,buildroot-2017.05-163-g8035745,skiboot-v5.7-92-g0818df625e35,hostboot-be278ac,linux-4.12.5-openpower1-pc06e0a6,petitboot-v1.5.1-p641e963,machine-xml-9a28007-p5e6f272,occ-71b5f68,hostboot-binaries-cf51511,capp-ucode-9c73e9f,sbe-bcb079b",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.18_1.32"
    }
}
```

`Errorlog snippet`
```
 | Reference Code           : BC10E504                                          |
 | Hex Words 2 - 5          : 000000E0 00000B00 00000000 00200000               |
 | Hex Words 6 - 9          : 00050000 00000303 9F050034 00000000               |
 |                                                                              |
 | ModuleId                 : 0x0B                                              |
 | Reason Code              : 0xE504                                            |
 | Code Location            : 0x0303                                            |
 |                                                                              |
 | PRD SRC Type             : PRD Detected Hardware Indication                  |
 | PRD SRC Class            : Hardware Error Detected                           |
 |                                                                              |
 | PRD Signature            : 0x00050000 0x9F050034                             |
 | Signature Description    : pu(n0p0) (INTCQFIR[52:54]) INT_CQ_FIR_PC_RECOV_ERR|
 |                          : OR_0_2:                                           |
 |------------------------------------------------------------------------------|
 |                              User Defined Data                               |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 6                                                 |
 | Created by               : errl                                              |
 | Callout type             : Hardware Callout                                  |
 | Target                   : Physical:/Sys0/Node0/Proc0                        |
 | CPU id                   : 0xFFFFFFFF                                        |
 | Deconfig State           : NO_DECONFIG                                       |
 | GARD Error Type          : GARD_NULL                                         |
 | Priority                 : SRCI_PRIORITY_MED                                 |
 |------------------------------------------------------------------------------|
```

```

dcrowell77, 2017-08-20:
Does this module have any history?  This could just be a hardware issue.  Is it consistent on every boot?

Was this logged at runtime or during ipl?  If you post the complete error log that should tell me.

gkeishin, 2017-08-20:
We see this consistently every boot on DD2 systems we got. I have DM you the logs. Let me know if you need more data..

dcrowell77, 2017-08-21:
>  | String data              : Hostboot Build ID: hostboot-be278ac/hbirt.bin     |
That tells me that the logs happen at runtime.  I think that the logs point to an issue in the interrupt logic.  It is possible something in opal and/or linux is causing the issue.  Sending over to @stewart-ibm to take a look.

apopple, 2017-08-21:
I believe this is caused by opal/linux - it has already been reported and tracked on our internal bugtrackers as BZ153951/SW387716

Last comment there from @stewart-ibm was "Lowering priority as not crash/checkstop. We do need to fix before GA though."

wghoffa, 2018-07-20:
Tracking the OPAL internal fix it was fixed awhile ago:

==== State: Open by: cde00 on 12 September 2017 00:22:39 ==== 46

------- Comment From sesmith@au1.ibm.com 2017-09-12 01:19:32 EDT-------
Ben has a fix for an issue he found.

In this series https://patchwork.ozlabs.org/project/skiboot/list/?series=2373

closed 2018-07-20 by wghoffa