open-power/hostboot #111: Commit 3da96b leaves op-build tree dirty

wghoffa, 2017-08-19:
Commit 3da96b958d3b248737febeb85951a0e446787d07 is causing a temporary file to be created which is leaving the op-build tree in a dirty state. At a minimum we need to move the temporary file into an appropriate build/staging directory as right now it is falling in op-build/buildroot/.

sgupta2m, 2017-08-20:
@rajadas2 can you please fix it.

rajadas2, 2017-08-21:
@wghoffa after building the op-build,
"op-build witherspoon-sequoia_defconfig && op-build hostboot-rebuild openpower-pnor-rebuild"

 i don't see anything in "git status"
-bash-4.1$ pwd
/esw/san5/rajadas2/open-power/witherspoon/op-build
-bash-4.1$ git status
# On branch master
# Changes not staged for commit:
#   (use "git add <file>..." to update what will be committed)
#   (use "git checkout -- <file>..." to discard changes in working directory)
#   (commit or discard the untracked or modified content in submodules)
#
#	modified:   buildroot (untracked content)
#
no changes added to commit (use "git add" and/or "git commit -a")

Is this the file you are referring to, when you mention a temporary file created is leaving the op-build tree in a dirty state
"buildroot/merged_target_types.xml"

wghoffa, 2017-08-21:
@rajadas2 - Is this before or after your fix?

Yes, this was the temporary file we saw: "buildroot/merged_target_types.xml"

rajadas2, 2017-08-22:
@wghoffa i have pushed a patch to remove the temp file
http://ralgit01.raleigh.ibm.com/gerrit1/44942

stewart-ibm, 2017-09-18:
Fixed in  b80767ab3dbff04b481cd2855d8af3a0c323f6c5

closed 2017-10-19 by stewart-ibm