open-power/hostboot #112: Host is flodding the BMC with the following message.

ratagupt, 2017-08-31:
IPMI Incoming: Seq 0x50, NetFn 0x0c, CMD:0x02
0x000000: 01 00 00 00

It generates unnecessary traffic on BMC.and it is filling the journal.

BMC keeps responding with unsupported parameter.

ratagupt, 2017-08-31:
I will update the parameter type for the get channel command which keeps coming down.

dcrowell77, 2017-08-31:
Hmm, that looks like our 'no command' value, which should never happen.  If you return an error there should be logs from us, do you have them?  Do you have the SOL?

What system are you running on?  What level of code are you using?

ratagupt, 2017-09-01:
It is xCAT witherspoon system,I have asked victor hu to update this defect..

whowutwut, 2017-09-01:
@dcrowell77  We will need to run some more experiments here...  xCAT has a genesis kernel that we run to do discovery and in testing the VLAN, I left it running long term for whatever reason .... 

It looks like inside there, we must be some inband call, do you know from those messages what it is?  to help us track it? When I turn off the compute side, the messages stop.  So I think it's caused by some command that is run on the genesis kernel when doing discovery.   You can either close this for now as I think it's caused by some command issues in the genesis loop....  Can you confirm @zet809?

dcrowell77, 2017-09-01:
Oh, the messages aren't from Hostboot?  I misunderstood the initial description.  If the messages are showing up at runtime then this is not a hostboot issue, it is something else (xcat evidently).

closed 2017-09-01 by dcrowell77