open-power/hostboot #114: Perl generators broken by new perl

wak-google, 2017-09-21:
```
Can't locate fapi_utils.pl in @INC (@INC contains: /usr/local/google/home/wak/google-src/zaius/output/zaius/host/lib/perl /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.24.1 /usr/local/share/perl/5.24.1 /usr/lib/x86_64-linux-gnu/perl5/5.24 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.24 /usr/share/perl/5.24 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base ../../xmltohb) at ../common/xmltohb/create_ekb_targattr.pl line 49.
```

Upstream change removes "." from @INC
http://search.cpan.org/dist/perl-5.26.0/pod/perldelta.pod#Removal_of_the_current_directory_(".")_from_@INC

We rely on this functionality in some of the xmltohb scripts for
require "fapi_utils.pl"; lines

Ideally this would be fixed by adding the proper path to fapi_utils in the @INC for each of the files. You can hack around this issue by adding the CWD to PERL5LIB for the entire build.

ozbenh, 2018-01-19:
Can somebody fix that one of these days ? I still can't build op-build master on any of my build systems here

madscientist159, 2018-01-19:
@ozbenh Cherry pick this into your tree:

https://git.raptorcs.com/git/talos-op-build/commit/?id=33b54e61ca6fcd1acaf0981ea891980ab4a84eeb

ozbenh, 2018-01-19:
@madscientist159 Ah thanks, what's the actual git url for these trees ?

madscientist159, 2018-01-19:
@ozbenh https://scm.raptorcs.com/scm/git/talos-op-build , in this case

ozbenh, 2018-01-19:
Alternatively, just doing "PERL_USE_UNSAFE_INC=1 op-build" worked, thanks

madscientist159, 2018-01-19:
@ozbenh Yep, that's basically what the linked commit does.  This was the first roadblock we ran into but have been tracking master (more or less) on Talos for a while now.

dcrowell77, 2018-01-19:
I don't see any issue with making this a permanent fix.  If someone wants to push up a pull request to op-build I'm sure @stewart-ibm would merge it

commit 9925aa35 (2018-02-05): Remove @INC with current directory from create_ekb_targattr.pl

Change-Id: I1e1342fd9444d6936d8c8fe030c50cc675296c3e
Resolves: #114
Reviewed-on: http://ralgit01.raleigh.ibm.com/gerrit1/46622
Tested-by: Jenkins Server <pfd-jenkins+hostboot@us.ibm.com>
Tested-by: Jenkins OP Build CI <op-jenkins+hostboot@us.ibm.com>
Tested-by: Jenkins OP HW <op-hw-jenkins+hostboot@us.ibm.com>
Reviewed-by: Prachi Gupta <pragupta@us.ibm.com>
Tested-by: FSP CI Jenkins <fsp-CI-jenkins+hostboot@us.ibm.com>
Reviewed-by: Christian R. Geddes <crgeddes@us.ibm.com>
Reviewed-by: William G. Hoffa <wghoffa@us.ibm.com>
closed 2018-02-05 by wghoffa