open-power/hostboot #117: How to output debug message in dimmBadDqBitmapAccessHwp()

ryanwyluo, 2017-10-13:
Hi All,
I try to output some debug message in routine dimmBadDqBitmapAccessHwp(), which is located in "src\usr\hwpf\hwp\dimmBadDqBitmapAccessHwp.C".
I used "CONSOLE::displayf()" to output message, but I got an compile error as below,
>>>>
exception caught: Could not find symbol _ZN7CONSOLE8displayfEPKcS1_z
exception caught: Could not find symbol _ZN7CONSOLE8displayfEPKcS1_z
<<<

Below is the sample code
>>>
#include <console/consoleif.H>
...
fapi::ReturnCode dimmBadDqBitmapAccessHwp()
{
    if (i_get)
    {
        FAPI_INF(">>dimmBadDqBitmapAccessHwp: Getting bitmap");
    }
    else
    {
        FAPI_INF(">>dimmBadDqBitmapAccessHwp: Setting bitmap");
    }

    fapi::ReturnCode l_rc;

    // Note the use of heap based arrays to avoid large stack allocations

CONSOLE::displayf(NULL, "dimmBadDqBitmapAccessHwp");
...
}
<<<

Is there any other ways to do this job or how can I fix the compile issue?
PS: with Trace lite debug it is ok, but it takes too long to debug issue.
Thanks in advance~
Ryan

dcrowell77, 2017-10-13:
If you look closely at your output I suspect that this is happening for the hbirt.bin image.  That is the runtime image, there is no console there.  Add #ifndef __HOSTBOOT_RUNTIME around your CONSOLE calls.

ryanwyluo, 2017-10-14:
Hi dcrowell77, 

That's exactly what you said.
After adding  "#ifndef __HOSTBOOT_RUNTIME ", I can output debug message at this routine,
Thanks a lot~

The modified code is as below:
<<<

Below is the sample code

#include <console/consoleif.H>
...
fapi::ReturnCode dimmBadDqBitmapAccessHwp()
{
if (i_get)
{
FAPI_INF(">>dimmBadDqBitmapAccessHwp: Getting bitmap");
}
else
{
FAPI_INF(">>dimmBadDqBitmapAccessHwp: Setting bitmap");
}

fapi::ReturnCode l_rc;

// Note the use of heap based arrays to avoid large stack allocations
**#ifndef __HOSTBOOT_RUNTIME**
CONSOLE::displayf(NULL, "dimmBadDqBitmapAccessHwp");
**#endif**
...
}
<<<

closed 2017-10-14 by ryanwyluo