open-power/hostboot #123: Fix enum constant in boolean context warnings

shenki, 2017-11-14:
target.H contains some code that checks AttributeTraits, which are enums. With GCC 7 testing an enum cases a warning:
```
../../../../../src/include/usr/targeting/common/target.H:668:5: warning: enum constant in boolean context [-Wint-in-bool-context]
```

```C++
template<const ATTRIBUTE_ID A>
bool Target::tryGetAttr(typename AttributeTraits<A>::Type& o_attrValue) const
{
    if(AttributeTraits<A>::readable) { } 
    if(AttributeTraits<A>::notHbMutex) { }
    if(AttributeTraits<A>::notFspMutex) { }
    return _tryGetAttr(A,sizeof(o_attrValue),&o_attrValue);
}
```

dcrowell77, 2017-11-14:
I believe this code is actually only there to force build failures when the enums don't exist.  I'm not quite sure how to fix it, I guess change it to this maybe?
    if(AttributeTraits<A>::readable == AttributeTraits<A>::readable) { } 
Or would the compiler optimize that out?

Maybe @bofferdn has some thoughts, think he added this way back when.

bofferdn, 2018-05-23:
the idea was to force a compile fail if you tried to write code that wrote an attribute when it's not writeable.  The compiler optimizes all that stuff away.