open-power/hostboot #127: The console display of dimm present do not match the hardware.

MengLiMarkham, 2018-02-27:
P9 hostboot code will get and display the dimms which are present in the system as below:
```
        TARGETING::TYPE l_type = target->getAttr<TARGETING::ATTR_TYPE>();
        TARGETING::ATTR_FAPI_POS_type l_pos = 0;
        if( target->tryGetAttr<TARGETING::ATTR_FAPI_POS>(l_pos) )
        {
            l_presData[l_type] |= (0x8000000000000000 >> l_pos);
        }
```
In P8 we use `ATTR_POSITION`, that's ok for P8. But in P9 hostboot `ATTR_FAPI_POS` will over 64, and when shift over 64, overlapping happened, then these dimms can not be displayed in the console log as preset. 
https://github.com/open-power/hostboot/blob/master/src/usr/isteps/istep06/host_discover_targets.C#L501

dcrowell77, 2018-02-27:
This issue has been reported elsewhere too...  The problem is that we don't really have a good position value to use that is consistent across all system types and all the target types that we want to display.

The best answer would be if we actually settled on a universal numbering scheme controlled via the MRW that actually matched the documentation for a particular system.

stewart-ibm, 2018-03-01:
Dan <notifications@github.com> writes:
> The best answer would be if we actually settled on a universal
> numbering scheme controlled via the MRW that actually matched the
> documentation for a particular system.

I'd be in favour of that if it helped us get something that matched what
as silkscreened on the board.

Mind you, it would require somebody to actually verify that it
matches... which our experience with firestone and PCI slots is that
somebody notices at least 12-18 months after shipping :/

-- 
Stewart Smith
OPAL Architect, IBM.