open-power/hostboot #128: Romulus / Talos does not boot on DD2.2

madscientist159, 2018-03-14:
Not sure if this is a hostboot or SBE issue, but on the latest op-build and DD2.2 the SBE hangs on ISTEP 5.  Nothing is printed to console.  IBM Austin has reproduced this issue on Romulus; we see it on Talos.

madscientist159, 2018-03-14:
Further investigation shows this is an SBE problem.  Bug moved to https://github.com/open-power/sbe/issues/10

madscientist159, 2018-03-15:
The SBE team sent this back here.  Apparently it's an issue with the HBBL.

@dcrowell77 can you look into this?

madscientist159, 2018-03-15:
Hostboot fed203b290c12b7b634c04eeec2b95fc496a726f appears to still work, as a starting point.

dcrowell77, 2018-03-15:
fed203b works, what is the first failing level you've seen?

I got a bit lost in open-power/sbe#10, can you tell me what happens after you force the update via Cronus?  It boots up into the sbe update step and then doesn't come back at all?  Or it reboots okay but off of the backup seeprom side?

pragupta, 2018-08-17:
D2.2 support is in HCODE now. Please reopen if it recreates.

closed 2018-08-17 by pragupta