open-power/hostboot #130: DD2.2 second processor I2C arbitration failure

madscientist159, 2018-03-25:
Noting this issue on a DD2.2 8-core Talos II system (Sforza).  This is a placeholder bug until proper debug can be performed.  Second CPU package is not available to the system after this fault as the MVPD cannot be read.

```
  7.68465|================================================
  7.68466|Error reported by i2c (0x0700) PLID 0x9000005A
  7.68467|  Bus Arbitration Lost Error was found in the I2C status register.
  7.68468|  ModuleId   0x07 I2C_CHECK_FOR_ERRORS
  7.68469|  ReasonCode 0x070c I2C_ARBITRATION_LOST_ONLY_FOUND
  7.68470|  UserData1  Status Register Value : 0x090e0c0000050001
  7.68471|  UserData2  Master Target : 0x0000ffff03e8001c
  7.68472|------------------------------------------------
  7.68472|  Callout type             : Hardware Callout
  7.68473|  CPU id                   : 5
  7.68474|  Target                   : Physical:/Sys0/Node0/Proc1
  7.68475|  Deconfig State           : NO_DECONFIG
  7.68476|  GARD Error Type          : GARD_NULL
  7.68476|  Priority                 : SRCI_PRIORITY_HIGH
  7.68477|------------------------------------------------
  7.68478|  Callout type             : Procedure Callout
  7.68479|  Procedure                : EPUB_PRC_HB_CODE
  7.68480|  Priority                 : SRCI_PRIORITY_LOW
  7.68480|------------------------------------------------
  7.68483|  Hostboot Build ID:
  7.68484|================================================
```

madscientist159, 2018-03-25:
Only seeing this when the secure mode jumpers are set to secure mode.

dcrowell77, 2018-03-26:
@bofferdn @mabaiocchi

dcrowell77, 2018-03-26:
Are you running with any patches?  One of the side effects of security is some restrictions on the i2c bus out to the sbe seeprom before the SMP is alive.  If you attempt that operation, it might show up like this.

madscientist159, 2018-03-26:
No patches to hostboot or the SBE.  Our work mostly centers on skiboot / skiroot / BMC at this point.

bofferdn, 2018-05-23:
This typically happens when master chip is in non-secure mode, but other chip is in secure mode.  Please check the chip security enablement is synchronized in either polarity.  

But .. we just integrated a change via HB commit d0eaecced89b9a8467cd40e76169f6c88db0e56d which stops issuing diagnostic mode i2c resets over FSI to the non-master chips.   Can you please verify either strategy fixes the problem?

wghoffa, 2018-07-20:
@madscientist159 -- Can you verify if either strategy fixes this problem?

pragupta, 2018-08-17:
Please reopen if master shows the same error. We pushed in a bunch of fixes related to this.

closed 2018-08-17 by pragupta