open-power/hostboot #132: Some DIMM SPD fields must be parsed before issuing IPMI Write FRU command

AlexanderAmelkin, 2018-04-10:
Currently the `Manufacturer Name` and `Product Name` fields of DIMM FRUs fed to the BMC contain raw binary data from `Manufacturer ID Code` (bytes 320-321) and `DRAM Device Type` (byte 2) fields of SPD.

This yields incorrect data presentation in BMC (see https://github.com/openbmc/openbmc/issues/1169).

I believe that parsing should be done on the OpenPOWER Firmware side so that the data written to FRU via IPMI is human-readable (e.g. "Samsung" instead of binary 0xCE80, and something like "DDR4 UDIMM, 32GB, 2400MHz, 64-bit, ECC" instead of binary 0x0C) because on the BMC side, strictly speaking, parsing those fields is not covered by any specification and parsing them according to JEDEC SPD spec would require building in some superfluous knowledge regarding hostboot operation into BMC firmware.

Please also note that for Product Name it is desirable to parse not just byte 2 of SPD, but also bytes 3 (for module type), 4,6,12,13 (for total capacity, bus width and ECC presence), and 18 (for speed).

dcrowell77, 2018-04-10:
> parsing them according to JEDEC SPD spec would require building in some superfluous knowledge regarding hostboot operation into BMC firmware.

I'm not sure I agree with this.  The JEDEC spec is universal so there is nothing Hostboot-specific about using it to parse things.  The only platform-specific piece in here is the format of the data in the FRU Inventory records itself.  There is no real standard there so it is all just legacy from the initial POWER8 OpenPOWER implementation.  The IPMI standard allows for binary or ascii data to be included in these fields.

However, I am not discounting the potential usefulness of such a change.  We're happy to take a look at a Hostboot commit to support your request. :-)

AlexanderAmelkin, 2018-04-22:
@dcrowell77, we most probably will provide such a patch some day. I've created this issues just to keep track of it. However, if anyone creates a patch before we do, that will be nice. :)

As for SPD and FRU standards, what I meant when I wrote "superfluous knowledge" is that when you read an SPD (in hostboot) you definitely know you're dealing with JEDEC-compliant data. When on the other hand you're parsing FRU on the BMC side, you generally have no idea what data is contained in the the custom data fields of this particular FRU record. The standardized FRU fields in any of the standardized areas do not allow for storing binary JEDEC SPD data, nor does the FRU specification refer to JEDEC anywhere at all. The BMC generally has no knowledge that binary 0xCE80 in the `Manufacturer Name` field of a Board Area must be treated according to some JEDEC spec, and 0x0C in the `Product Name` field is simply insufficient to decode the actual product type.

AlexanderAmelkin, 2018-09-12:
FYI: This is resolved for P8 in https://github.com/YADRO-KNS/op-build/pull/2
Would you like us to prepare a pull request to master-p8? I'm not sure if it cleanly applies to master.

dcrowell77, 2018-09-12:
Yes, that would be great!