open-power/hostboot #136: Spurious CPU2 guard entries created

madscientist159, 2018-05-01:
Under as of yet indeterminate conditions (mainly in the field, at customer locations) CPU2 will guard itself out on our Talos platforms.  Clearing the guard partition always restores the system to perfect working order.

We have seen this happen once or twice in our labs as well.  It is random, but sometimes hostboot will spew FSI errors, go down for reboot, and the second CPU will be guarded out for no reason.

Until this is fixed, is there a way we can disable the CPU guard mechanism?

dcrowell77, 2018-05-01:
There is never 'no' reason.  Any time there is a GUARD record there will be an error log.  Do you have the esel binary or at least the console output of the error log?

As far as disabling GARD completely, there is 1 big hammer in src/usr/hwas/HBconfig.
```
config NO_GARD_SUPPORT
    default n
    help
        Skip guarding when set
```
That will completely turn off the code that saves the record.

madscientist159, 2018-05-01:
Will watch for it.  Back when we had seen it on our side it was due to the BMC OCC hwmon drivers trying to access the OCC over FSI at the same time hostboot was trying to; the system would spit out FSI errors and guard out the CPU.  We're pretty careful to shut down the OCC hwmon drivers during IPL at this point, but it remains possible that they could be active during and after an abnormal shutdown event.

nashimus, 2018-05-06:
I just experienced this when rebooting Debian buster on 4.16 kernel.

Console and esel.out:
[console_output.txt](https://github.com/open-power/hostboot/files/1977282/console_output.txt)
[esel.out.txt](https://github.com/open-power/hostboot/files/1977283/esel.out.txt)

dcrowell77, 2018-05-06:
17.06743|Error reported by fsi (0x0A00) PLID 0x9000014D
 17.06859|  FsiDD::handleOpbErrors> Error during FSI access
 17.06860|  ModuleId   0x04 FSI::MOD_FSIDD_HANDLEOPBERRORS
 17.06861|  ReasonCode 0x0a04 FSI::RC_OPB_ERROR
 17.06862|  UserData1  Relative FSI Address : 0x000028fc001028fc
 17.06863|  UserData2  Absolute FSI Address : 0x0002008000050000
 | userdata1[00:31]         : Relative FSI Address                              |
 | userdata1[32:63]         : Absolute FSI Address                              |
 | userdata2[00:31]         : OPB Status Register                               |
 | userdata2[32:63]         : FSI Master HUID                                   |

 00000017.035507908|  181|FAPI            | 133|p9_setup_sbe_config.C: p9_setup_sbe_config::  Entering ...
 00000017.045377730|  181|FAPI_I          | 468|plat_hw_access.C: platGetCfamRegister: deviceRead returns error!
 00000017.045380164|  181|FAPI_I          | 478|plat_hw_access.C: fapiGetCfamRegister failed - Target pu:k0:n0:s0:p01, Addr 0000283F

Typically FSI errors like this are caused by some kind of hardware error out to the end device, e.g. full power or clocks.  The failure is happening in the step where we start up the second proc, which is basically the first time we interact with that chip.  I'd suggest looking at what the BMC might be doing in the reboot.

madscientist159, 2018-05-06:
@dcrowell77 This is consistent with the way the bug was first experienced, where the BMC was attempting comcurrent FSI access via the OCC hwmon drivers and causing the bus to lock up.  We are now completely unbinding and disabling the OCC drivers on the BMC during the IPL / reboot process, which has reduced the frequency of this type of failure, but as you can see it is still present.

My current understanding is that hostboot itself kicks off the SBE on CPU1, that this access at least is succeeding, but then trying to get SBE status at a later point is returning an error.  Assuming this is not BMC bus contention, what are the most likely candidates for this type of error, or is there any way to get additional information out of the "dead" CPU via hostboot?  Attaching Cronus is a no-go here due to the rarity of the failure; we have yet to reproduce this in-house.

On a slightly different topic, do you mind sharing the steps you used to dump the log above from the ERRL data?

Thanks!

dcrowell77, 2018-05-07:
Re : error log data -- There is a errl binary that decodes our error logs.  It isn't externally buildable (yet, that is being actively worked) but I believe it is available through one of the openpower support sites.  I don't have the details though...  Once you have that you can use https://github.com/open-power/hostboot/blob/master/src/build/debug/eSEL.pl to pull our data out of the BMC's logs.

> We are now completely unbinding and disabling the OCC drivers on the BMC during the IPL / reboot process

When are you doing this relative to the boot flow?  I'd think you'd want to do something like this:
1-reboot requested
2-unbind, stop all FSI access
3-drop power
4-apply power
5-kick off SBE on proc0
6-wait for OCC active sensor to turn on
7-restart occ hwmon

> what are the most likely candidates for this type of error

As mentioned above, in my experience it has been a problem with the processor not getting powered/reset correctly.  Looking a bit closer I guess the specific reg we failed on (283F) is not the first access we make so this does seem like it might be contention versus a boot issue.

> is there any way to get additional information out of the "dead" CPU via hostboot?

FSI is the only path we have, so if that isn't working we're out of luck.  We do grab some error data from proc0, but it doesn't really tell us much most of the time.

One more bit of info from the error log to narrow things down.  You can decode this backtrace using the hbicore.list.bz2 file that is generated by your hostboot build.
 | Backtrace                : # 0 0000000000050D14                              |
 |                          : # 1 000000000004C79C                              |
 |                          : # 2 000000004055337C                              |
 |                          : # 3 0000000040551BB0                              |
 |                          : # 4 0000000040552A5C                              |
 |                          : # 5 0000000040552DE4                              |
 |                          : # 6 0000000040553990                              |
 |                          : # 7 0000000000059D48                              |
 |                          : # 8 00000000000596B0                              |
 |                          : # 9 000000004047006C                              |
 |                          : #10 0000000040091CEC                              |
 |                          : #11 000000004008ECD0                              |
 |                          : #12 0000000000002580                              |

madscientist159, 2018-05-07:
> When are you doing this relative to the boot flow? I'd think you'd want to do something like this:
> 1-reboot requested
> 2-unbind, stop all FSI access
> 3-drop power
> 4-apply power
> 5-kick off SBE on proc0
> 6-wait for OCC active sensor to turn on
> 7-restart occ hwmon 

Yep, that's exactly what we're doing, to the point of even disabling fast reset because it was causing FSI bus issues with the OCC hwmon drivers.

The only lead we have at the moment is that apparently the VRM configuration for both CPU0 and CPU1 is subject to a race condition, and there's at least some chance that CPU1 might randomly not be brought up to full voltage before IPL is kicked off.  This is being actively worked now.

madscientist159, 2018-05-13:
Further investigation has led to this likely being the result of spurious scanFSI commands generated on the BMC due to BMC-side systemd-related race conditions encountered only on host-initiated reboot.  We have shored up the dependency sequencing in conjunction with a community firmware developer, and the resultant BMC images do not appear to trigger this fault.

closed 2018-05-13 by madscientist159